kizniche/Mycodo #654: Adding second TH10

4eyegit, 2019-05-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.4.3 
 
#### Problem Description
Please list: 
Can't get temperature reading on a second Sonoff TH10 with a DS18B20 temperature sensor.

Having setup a first Sonoff Th10 with a AM2301 Temp/Humidity sensor I just can't get the second one with the waterproof temp sensor to print the value on Mycodo. 

If I change the value "['AM2301']" on line #111 of th1x.py to "['DS18B20']" it prints the temp from the second sonoff, but loses the reading from the first one, of course.

With my very, very limited knowledge of coding I just can't seem to find an answer, correct or suggest a fix for this limitation on my on.

kizniche, 2019-05-03:
It looks like this device supports 4 different sensors. The Am2315 is hard coded in this module. Because of the way the measurement system is set up, it would be best to merely add new input modules (opposed to adding a drop down option in this module to select the sensor type).

To successfully create a TH1x Input module with the DS18B20, you'll have to test for me because I don't have this device. A few questions:

1. Have you successfully received a sensor measurement using any code or app with this device?
2. If so, can you provide the json text returned from the web address measurement query (like the current module uses for the Am2315)?
3. Are you receiving any errors in the daemon log with the I put module you modified?
4. Did you restart the daemon (backend) after saving your edits to the input module?

4eyegit, 2019-05-03:
Thank you for the response, Kyle.

Have you successfully received a sensor measurement using any code or app with this device?
Haven't tried any other MQTT connection with the Tasmota firmware installed. 

If so, can you provide the json text returned from the web address measurement query (like the current module uses for the AM2301)? 
You mean this one, right? http://{ip}/cm?cmnd=status%2010

Yes, we get: {"StatusSNS":{"Time":"2019-05-03T16:41:48","DS18B20":{"Temperature":26.2},"TempUnit":"C"}}

Are you receiving any errors in the daemon log with the I put module you modified?
No errors if I have only the CH0: Temperature selected
But if I select any more channels, nothing is printed.

And the Daemon log prints:
2019-05-03 17:46:28,323 - mycodo.th16_38d0ce9d - ERROR - InputModule raised an exception when taking a reading: 'AM2301'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/th1x.py", line 108, in get_measurement
    return_dict[1]['value'] = dict_data['StatusSNS']['AM2301']['Humidity']
KeyError: 'AM2301'

Did you restart the daemon (backend) after saving your edits to the input module?
Yes, depending on the modification I get error 502 Bad Gateway.

SAM26K, 2019-05-05:
One thing you have to watch is that the hole in the side of the Sonoff device sometimes blocks the sensor connector from seating all the way.  You can use a soldering iron to open the hole a little.
Also, have you tried just typing the ip address for the Sonoff in a browser window? That should also give you the temperature data and a simple GUI.

4eyegit, 2019-05-06:
Thank you Sam, I'm aware of those.

> One thing you have to watch is that the hole in the side of the Sonoff device sometimes blocks the sensor connector from seating all the way. You can use a soldering iron to open the hole a little. [...]

kizniche, 2019-05-06:
I just released 7.5.0 that has the new DS18B20 TH1x Input. Give it a try and let me know if it works. Thanks.

4eyegit, 2019-05-06:
Hi Kyle, thank you for the inputs. 
Was just testing it before your last reply. The readings work for both sensor/sonoff, but as soon as I enable "Dew point" and/or "VPD" calculations to the AM2301 input it stops printing the readings for that sonoff. 

Daemon error is: 

2019-05-06 15:55:57,514 - mycodo.input_27aabc73 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-05-06 15:55:58,477 - mycodo.th16_am2301_27aabc73 - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object is not subscriptable

kizniche, 2019-05-06:
Thanks for the feedback and log. I'll look into it.

4eyegit, 2019-05-06:
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 114, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 123, in get_measurement
    self.get_value(0), self.get_value(1))
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 128, in get_value
    return self._measurements[channel]['value']

kizniche, 2019-05-06:
Oh, this is a pretty bad bug. I'm going to work on a fix and get a new release out. Stand by.

kizniche, 2019-05-06:
Just released 7.5.1. Let me know if that fixed the issue.

4eyegit, 2019-05-06:
Thank you Kyle. That did it. 

My last request. 
Can you point me to a place I can learn how to use Mycodo to control the Sonoffs?
Not sure if I should be using "linux command" or "python command" to build an output for them. 

Thank you very much

closed 2019-05-06 by 4eyegit

kizniche, 2019-05-06:
I'm not sure what you're asking. What are you trying to accomplish?

4eyegit, 2019-05-06:
Sorry. I'm trying to turn on and off the switch to control humidity and air exhaust on the grow room. 
I actually want to control the VPD of the box. 
I still have to learn how to toggle the relay on and off using Mycodo's output controls.

kizniche, 2019-05-08:
You probably want to set up a conditional. The code for this will be Python. You can set your sensor as the measurement input, and execute a number of different actions based on the measurements and code.

SAM26K, 2019-05-28:
Know this is dated but to toggle on/off, I created linux command output controls and then created Function:conditional controls to turn them on and off depending on Temp Humidity and/or VPD.  Working great for several months.   On specific commands lookup the Tasmota project on Github.
Simple linux on/off commands:
```
curl http://192.168.1.34/cm?cmnd=Power%20on
curl http://192.168.1.34/cm?cmnd=Power%20off.
```