kizniche/Mycodo #655: Can't upgrade

SAM26K, 2019-05-05:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.03
 
#### Problem Description
Please list:
Won't upgrade from 7.3. Before attempting to upgrade it stopped communicating with devices.  All desktop widgets not getting updated data.  Thought an uprev might fix the problem.
### Errors
Error: Could not determine local mycodo version or online release versions: <urlopen error _ssl.c:704: The handshake operation timed out>
No upgrade is available. You are running the latest release, version 7.3.0.

Daemon Logs:
`2019-05-05 14:43:43,746 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:52,577 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:52,730 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:52,786 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:53,568 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:53,851 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:54,035 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:54,225 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:54,428 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:54,607 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:54,909 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:55,012 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:55,195 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:55,377 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:55,521 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:56,933 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:57,113 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:58,068 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:58,235 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-05-05 14:43:58,597 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 962, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'`

### Steps to Reproduce the issue:
How can this issue be reproduced?
click gear icon > upgrade.

kizniche, 2019-05-05:
It's working for me, so it looks like an issue on your end. Here's my upgrade page.

```
Installed version: 7.4.2
Latest version: 7.4.3
You are 1 release behind the latest version. 
```

You can always just move the current ~/Mycodo directory, install the latest version, and transfer your database. Here's how I would do that:

```
sudo service mycodo stop && sudo service mycodoflask stop
cd ~
mv Mycodo Mycodo-old
git clone https://github.com/kizniche/Mycodo
cd Mycodo
cp ~/Mycodo-old/databases/mycodo.db ./databases/
sudo /bin/bash ./install/setup.sh
```

kizniche, 2019-05-05:
If you have any custom Inputs, photos, or notes, this may pose an issue and you will have to copy those manually, but for most simple setups a simple database transfer will suffice.

SAM26K, 2019-05-05:
Trying to reinstall but getting this error:

> pi@raspberrypi3_2:~ $ git clone https://github.com/kizniche/Mycodo
Cloning into 'Mycodo'...
fatal: unable to access 'https://github.com/kizniche/Mycodo/': gnutls_handshake() failed: Error in the pull function.
pi@raspberrypi3_2:~ $

kizniche, 2019-05-05:
That may be what's causing your initial issue, as ```git``` is crucial to determining the version information. It looks like you may have corrupted data on your SD card causing issues.

kizniche, 2019-05-05:
It could also indicate a networking issue, if not an issue with software.

SAM26K, 2019-05-06:
Do you think it could have caused the  problem with no I/O between the devices and mycodo?
Is there a limit to the number of widgets you can have on the desktop?  The original problem happened soon after adding a couple of controls to the desktop.

kizniche, 2019-05-06:
```git``` and Mycodo are separate and one wouldn't influence the other, so anything happening in Mycodo wouldn't cause what you're seeing with ```git```.

One last thing to check, is your free space at 0%?

SAM26K, 2019-05-08:
Turns out my internet bandwidth took a dive for some reason.  Download speed is only 1Mbps.  Have a new cable modem on the way.  In the mean time is it   possible to restore Mycodo-old back to Mycodo?  Tried just cp -r  copy but daemon won't start.
Thanks.

kizniche, 2019-05-08:
Here's how to get the old version working:

```
sudo service mycodo stop
sudo service mycodoflask stop
cd ~
mv Mycodo Mycodo-older
mv Mycodo-old Mycodo
sudo /bin/bash ./Mycodo/mycodo/scripts/upgrade_post.sh
```

SAM26K, 2019-05-19:
Replaced Comcast router and no help. The problem seems to be  between my Linksys EA7500 wifi router and the comcast xfinity wifi router.  If I logon to  the comcast xfi wifi the bandwidth is ok.  If I logon to the Linksys router wifi the bandwidth to the internet is only 1Mbps but the bandwidth to the local devices is fine.  It looks as if the Comcast router is choking the internet bandwidth to the Linksys router for some reason.  Plan on replacing the comcast cable router with a 3rd  party compatible cable router.
Anyway the original problem seems to be fixed by using a better bandwidth when upgrading from the internet.  Got latest Mycodo rev working on the master rpi.
Thanks for the help.

SAM26K, 2019-05-21:
Trying to update second rpi to latest rev using your instructions above for preserving the database but now getting this error:
`Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1244, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 552, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: input.log_level_debug

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1058, in page_live
    input_dev = Input.query.all()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 3161, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 3317, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 3342, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 988, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/sql/elements.py", line 287, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1107, in _execute_clauseelement
    distilled_params,
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1248, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1466, in _handle_dbapi_exception
    util.raise_from_cause(sqlalchemy_exception, exc_info)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 383, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 128, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1244, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 552, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: input.log_level_debug
[SQL: SELECT input.id AS input_id, input.unique_id AS input_unique_id, input.name AS input_name, input.is_activated AS input_is_activated, input.log_level_debug AS input_log_level_debug, input.is_preset AS input_is_preset, input.preset_name AS input_preset_name, input.device AS input_device, input.interface AS input_interface, input.period AS input_period, input.power_output_id AS input_power_output_id, input.resolution AS input_resolution, input.resolution_2 AS input_resolution_2, input.sensitivity AS input_sensitivity, input.thermocouple_type AS input_thermocouple_type, input.ref_ohm AS input_ref_ohm, input.calibrate_sensor_measure AS input_calibrate_sensor_measure, input.location AS input_location, input.gpio_location AS input_gpio_location, input.i2c_location AS input_i2c_location, input.i2c_bus AS input_i2c_bus, input.ftdi_location AS input_ftdi_location, input.uart_location AS input_uart_location, input.baud_rate AS input_baud_rate, input.pin_clock AS input_pin_clock, input.pin_cs AS input_pin_cs, input.pin_mosi AS input_pin_mosi, input.pin_miso AS input_pin_miso, input.bt_adapter AS input_bt_adapter, input.switch_edge AS input_switch_edge, input.switch_bouncetime AS input_switch_bouncetime, input.switch_reset_period AS input_switch_reset_period, input.pre_output_id AS input_pre_output_id, input.pre_output_duration AS input_pre_output_duration, input.pre_output_during_measure AS input_pre_output_during_measure, input [... cut ...] ution AS input_adc_resolution, input.adc_sample_speed AS input_adc_sample_speed, input.cmd_command AS input_cmd_command, input.weighting AS input_weighting, input.rpm_pulses_per_rev AS input_rpm_pulses_per_rev, input.sample_time AS input_sample_time, input.port AS input_port, input.times_check AS input_times_check, input.deadline AS input_deadline, input.datetime AS input_datetime, input.custom_options AS input_custom_options 
FROM input]
(Background on this error at: http://sqlalche.me/e/e3q8)`

SAM26K, 2019-05-21:
Upgrades  seemed to finish successfully using the broader band comcast wifi.

SAM26K, 2019-05-21:
Update: Everything is working now with 7.5.3.  Installed 2nd rpi mycodo from scratch without trying to copy the db files.  Restored    previous version with working database and t'hen upgraded to  current  mycodo version using the better wifi connection
Everything seems to be working now   and up to date.

SAM26K, 2019-05-22:
One thing I noticed when trying to restore the database manually; It seemed to finish successfully but when I accessed the ip address with a browser it was asking for a current user id and password.  The one I used should have worked but that's when the 800 error  occurred.
Anyway things are working now.

SAM26K, 2019-05-23:
Since the original problem was caused by a bad wifi connection and not a Mycodo problem, I'm cllosing thisi ssue.
Thanks very much for all the help!

closed 2019-05-23 by SAM26K

RynFlutsch, 2019-05-24:
I am on 7.3.0 too:
`Last 60 lines of /var/log/mycodo/mycodobackup.log:
[2019-05-24 19:02:51] 
[2019-05-24 19:02:51] #### Create backup initiated 2019-05-24_19-02-51 ####
[2019-05-24 19:02:51] Backing up current Mycodo from /root/Mycodo to /var/tmp/Mycodo-backup-2019-05-24_19-02-51-7.3.0...Failed: Error while trying to back up current Mycodo install from /root/Mycodo to /var/Mycodo-backups/Mycodo-backup-2019-05-24_19-02-51-7.3.0.
[2019-05-24 19:02:51] Fr 24. Mai 19:02:51 CEST 2019
[2019-05-24 19:02:51] 
[2019-05-24 19:02:51] #### ERROR ####
[2019-05-24 19:02:51] There was an error detected while creating the backup. Please review the log at /var/log/`

kizniche, 2019-05-24:
I can't think of any reason why a backup would fail other than you are out of free disks pace.

SAM26K, 2019-05-25:
Just for the record, SD  card space is not a problem for me.  Still have
3.8GB free.  Frequent backups are a  lifesaver and its a good idea to  look
at the backup log to confirm its successful.

SAM26K, 2019-05-26:
Update on the wifi problem: For some reason enabling  "media prioritization" on the Linksys EA7500 reduces download speed drastically (~1Mbs bits per second not bytes).  Disabled  "media prioritization" in router's diagnostics section and now its working fine.

RynFlutsch, 2019-05-26:
Hey,
disk space is not the problem, Database is incorrect.
I will try the steps above.

SAM26K, 2019-05-27:
> Hey,
> disk space is not the problem, Database is incorrect.
> I will try the steps above.

Let me know how it goes.  I hit the 500 error when I tried it but that's was not my original issue.