kizniche/Mycodo #657: Conversion Addition: Convert to psi

Magnum-Pl, 2019-05-14:
Hey Kyle,

Can you add the option to convert to (psi)? I set up a ADS1115 ADC and a pressure transducer that I'd like to display in psi. 

The script you gave me to raise/lower a PID set point works great. Is this something you can add as an action? I'd like to see it as an action to raise/lower a PID set point with the ability to choose how many units are raised by each action.

kizniche, 2019-05-14:
You can add units on the Configuration -> Measurements page. I added a lot of units as a base set (for the supported sensors) but there are far too many units to add them all. Perhaps we should compile a list of common units that are missing and I'll add them to the default set.

I like the idea of a raise/lower setpoint action.

kizniche, 2019-05-15:
Added psi and the function actions. They'll be available in the next release.

closed 2019-05-15 by kizniche

Magnum-Pl, 2019-05-15:
Thanks. I was able to add the psi conversion in the configuration menu. I didn’t know that was possible. 

The new setpoint action will be handy. 

Thanks for your help

kizniche, 2019-05-15:
After you upgrade, let me know if you experience issues with the units. I haven't really explored issues with user-entered units and then adding the same unit to the built-in set, so you may have issues unless you delete the unit before upgrading (which is a pain because you can't delete a unit that's in use somewhere in the Mycodo system).

kizniche, 2019-05-16:
I'll actually try to do some checks for duplicate units and see if I can prevent any issues that could arise from that.

Magnum-Pl, 2019-05-16:
I’ll update as soon as it’s available and let you know if there are any issues. 

I’m trying to delete a custom measurement but it doesn’t seem like there’s an option to do that. I can delete units and conversions but not measurements. Is this something you can add?