kizniche/Mycodo #658: Installation fails

Routout, 2019-05-17:
Rpi3
Fresh install fails.
Full install log: https://pastebin.com/8DLFNt4q

```
~ $ systemctl status mycodo.service
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-05-17 09:01:47 EEST; 28min ago
  Process: 7097 ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py (code=exited,
May 17 09:01:47 Myco systemd[1]: Starting Mycodo server...
May 17 09:01:47 Myco python[7097]: Traceback (most recent call last):
May 17 09:01:47 Myco python[7097]:   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 40, in <module>
May 17 09:01:47 Myco python[7097]:     import rpyc
May 17 09:01:47 Myco python[7097]: ImportError: No module named 'rpyc'
May 17 09:01:47 Myco systemd[1]: mycodo.service: Control process exited, code=exited status=1
May 17 09:01:47 Myco systemd[1]: Failed to start Mycodo server.
May 17 09:01:47 Myco systemd[1]: mycodo.service: Unit entered failed state.
May 17 09:01:47 Myco systemd[1]: mycodo.service: Failed with result 'exit-code'.
```

And:

```
~ $ journalctl -xe
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit mycodo.service has begun starting up.
May 17 09:01:47 Myco python[7097]: Traceback (most recent call last):
May 17 09:01:47 Myco python[7097]:   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 40, in <module>
May 17 09:01:47 Myco python[7097]:     import rpyc
May 17 09:01:47 Myco python[7097]: ImportError: No module named 'rpyc'
May 17 09:01:47 Myco systemd[1]: mycodo.service: Control process exited, code=exited status=1
May 17 09:01:47 Myco systemd[1]: Failed to start Mycodo server.
-- Subject: Unit mycodo.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit mycodo.service has failed.
--
-- The result is failed.
May 17 09:01:47 Myco systemd[1]: mycodo.service: Unit entered failed state.
May 17 09:01:47 Myco systemd[1]: mycodo.service: Failed with result 'exit-code'.
May 17 09:01:47 Myco sudo[3174]: pam_unix(sudo:session): session closed for user root
May 17 09:17:01 Myco CRON[7122]: pam_unix(cron:session): session opened for user root by (uid=0)
May 17 09:17:01 Myco CRON[7126]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
May 17 09:17:01 Myco CRON[7122]: pam_unix(cron:session): session closed for user root
May 17 09:25:33 Myco influxd[6596]: ts=2019-05-17T06:25:33.970341Z lvl=info msg="Retention policy deletion check
May 17 09:25:33 Myco influxd[6596]: ts=2019-05-17T06:25:33.970705Z lvl=info msg="Retention policy deletion check
lines 1807-1831/1831 (END)
```

kizniche, 2019-05-17:
This is the same old issue with building bcrypt. You can fix this manually by lowering the version number of bcrypt in requirements.txt and reinstalling. I'll fix this in the next release.

Routout, 2019-05-17:
Yes, 3.1.4 version fixed this.

kizniche, 2019-05-17:
Jus released v7.5.3 that should fix this issue. Thanks for the bug report.

closed 2019-05-17 by kizniche

bungnewbie, 2019-06-24:
i use v7.x in raspberry 3, but i have same problem