kizniche/Mycodo #660: 2x PID: single output / combined min. off duration

floriske, 2019-05-28:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.5.3
 
#### Problem Description
Please list:

I converted a fridge to a meat curing chamber and have 3 relay outputs:

- Fridge
- Heater
- Humidifier

The fridge is used for both cooling and de-humidifying for which I have 2 PID controllers.

In order to be able to switch te fridge with both of them I had te create 2 seperate "Fridge" outputs using the same GPIO since it's not possible to use the same output with multiple PID controllers.

If during the de-humidifying process the temperature drops too much the heater will turn on simultanously to keep the temperature ok.

This all works but the problem is that the 2 Fridge outputs/PID's don't take into account eachother's "min. off duration". So when de cooling PID is in the "min. off duration" period the de-humidifying PID can still switch on the fridge which is unwanted behaviour because a Fridge has to "rest" some time after being switched off in order to avoid compressor failure.
 
Is there any way to overcome this problem?

kizniche, 2019-05-28:
Forum Reference: https://kylegabriel.com/forum/pid-tuning/relay-pid-headaches/

I've been meaning to write a reply to this, but I just haven't had the time. I'll try to make a thoughtful response very soon.

kizniche, 2019-05-28:
I will be responding in the forum with my thoughts on alternate configurations to achieve your intended goal, but something I noticed that is applicable as an issue is the inability of PIDs to recognize if an output should remain off. I will investigate adding a backend setting "off_until" that will store a datetime object that indicates when the output is able to be turned on after. This will allow a PID to set ```off_until``` from one PID and all other PIDs will be required to respect that (as the output controller is ultimately in control of outputs turning on and off.

kizniche, 2019-05-29:
I was able to successfully implement the new feature that allows two PIDs to operate with the same output and them both respect the ```Min Off Duration``` either of them sets. As an example, you can see the graph, below, that shows one output and two test PIDs with slightly different settings but each having a ```Min Off Duration``` of 120 seconds. The effect can be seen that there are periods of PID output yet no output turning on and also when there is an output it is from each PID at different periods of time, showing whichever PID that gets to turn the output on first is what ```Min Off Duration``` will be used.

[image]

kizniche, 2019-05-29:
And here is an excerpt of the daemon log with debugging on, showing the new ```off_until``` option working from within the output controller:

```
2019-05-28 21:22:12,191 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 8.258748 more seconds.
2019-05-28 21:24:02,174 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 79.350983 more seconds.
2019-05-28 21:24:46,557 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 34.96392 more seconds.
2019-05-28 21:26:16,486 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 101.189186 more seconds.
2019-05-28 21:27:42,233 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 15.4684 more seconds.
2019-05-28 21:27:46,519 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 11.178732 more seconds.
2019-05-28 21:29:32,250 - ERROR - mycodo.controller_output - Output 3 (On/Off Test) instructed to turn on, however the output has been instructed to stay off for 109.522523 more seconds.
```

floriske, 2019-05-29:
Man you are fast! ;-)

Great work! looking forward to thrying it

kizniche, 2019-05-29:
v7.5.4 Released. Feel free to reopen this if there are still issues related to this.

closed 2019-05-29 by kizniche