kizniche/Mycodo #661: RPM Control PID: Sinlge PWM output for both raise and lower

floriske, 2019-05-28:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.5.3
 
#### Problem Description

I have 2 PWM controlled fans of which I'm monitoring- and want to control the RPM. 

If I want to do this with a PID controller I have to create 2 PWM outputs in order to be able to do this because the raise and lower outputs aren't allowed to be the same.

Is it possible to make it possbile to use the same output in a future version?

kizniche, 2019-05-28:
I initially made this sanity check to prevent issues with PIDs, but your use case makes a good argument for removing this and allowing users to make that decision. I'll test removing this check and will see about implementing it in the next release.

floriske, 2019-05-29:
Thank you very much!

kizniche, 2019-05-29:
v7.5.4 Released. Feel free to reopen this if there are still issues related to this.

kizniche, 2019-05-29:
Also, for historical reference, this is related to #660

closed 2019-05-29 by kizniche