kizniche/Mycodo #662: Error 500: Internal Server Error setup page

EddNess, 2019-06-06:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

Hi
I was trying to add a mh-z19 sensor, connected like this: https://pypi.org/project/mh-z19/
I went to the setup setup data page and got this error log:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1752, in page_data
    choices_measurements_units = utils_general.choices_measurements_units(measurement, unit)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 209, in choices_measurements_units
    unit_name=dict_units[each_unit]['name'])
KeyError: ''

What have I done wrong? I'm a complete Noob to the pi

kizniche, 2019-06-07:
Can you list what you did in steps I can follow to reproduce this issue?

EddNess, 2019-06-08:
Sorry - can't remember clearly enough, trying to re-create.
This might take a while, I'll get an action list to you when done.

EddNess, 2019-06-08:
Hi

Well I've started from scratch to get a clean install.
It seems to work now. I'm going to post the action list anyway in case it helps someone else in the future.
Thank you for looking at this.

A list of install actions follows:
On PC (Win10)
Format SD card using SD Card Formatter v5.
Write 2019-04-08-raspbian-stretch.img to card using Win32 Disk Imager.
Raspberry Pi 3 Model B+ Plus 1.4GHz Quad Core 64Bit 1GB RAM (2018 Model) connected to monitor via HDMI/DVI cable, dell wireless keyboard & mouse, using a power source that has the following output: 5V DC 2500mA.
Sensors connected: MH-Z19 to GPIO 5v power & ground, RX to GP1014 Tx to GP1015. AM2315 to GPIO 5v power & ground, SDA to GP102 (2W 10Kohm resistor upload to power) SCL to GP103 (2W 10Kohm resistor upload to power).
SD card safely removed from PC, inserted into Pi and power-on.
Raspbian resized root system automatically and rebooted.
Time zone etc set to U.K., British, London (local). Password Blanked off. Screen left as-is. Wi-Fi found & password entered. Software update accepted & installed.
Note made of Pi's IP address.
Restart.

Check AM2315 sensor:
Command window:
sudo apt-get update
sudo apt-get upgrade
sudo reboot
sudo apt-get install i2ctools
sudo apt-get install python-dev
sudo apt-get install libi2c-dev
sudo raspi-config
In interfacing options turn on I2C
In advanced options expand filesystem, reboot.
Command window:
sudo i2cdetect -y 1 (twice AM2315 found at address 5c)

Check MH-Z19 sensor:
sudo apt-get install git
git clone https://github.com/UedaTakeyuki/mh-z19.git
cd mh-z19
./setup.sh (y option selected)
Reboot
sudo raspi-config
In interfacing options turn on ssh and serial
Reboot (ignore warning)
Command window:
sudo python -m mh_z19 (null? - check wiring – OK – think I killed it. This worked before. Power down, disconnect & order another).

Install Mycodo:
Power up
Get fed up with warning and:
Command Window:
sudo raspi-config (enter a password)
Check config has been done for Mycodo from the readme (on PC, not Pi) – yes
Command window:
Curl -L https://raw.githubusercontent.com/kizniche/Mycodo/master/install/install | bash
Agree to license
After install reboot
On PC open Firefox & navigate to Pi’s IP address
Ignore security warning, agree and set-up admin user. Login
Go to set up data page & it works!?
Could the dodgy sensor have upset Mycodo?

closed 2019-06-08 by EddNess

kizniche, 2019-06-08:
I was more wondering the steps you took from within the Mycodo web interface to cause the error, but thank you for the detailed steps. I don't see anything that should cause that error. The error disappearing from a reinstall suggests it was an issue with your first install that caused it. In any case, I'm glad it was resolved with a reinstall.