kizniche/Mycodo #663: Upgrade error

SAM26K, 2019-06-13:
## Mycodo Issue Report:
  - Specific Mycodo Version:
Mycodo Version: 7.5.5 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 2c3b61fdb239
Daemon Status: Running 
Daemon Process ID: 926 
Daemon RAM Usage: 92.396 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 87.428 MB 
Frontend Virtualenv: Yes
 
#### Problem Description
Get this error when trying to upgrade from V7.5.5:
×
Error: Could not determine local mycodo version or online release versions: <urlopen error getrandom() initialization failed. (_ssl.c:720)>

### Errors
2019-06-12 08:03:33,681 - ERROR - mycodo.daemon - Could not determine local mycodo version or online release versions. Upgrade checks can be disabled in the Mycodo configuration.

### Steps to Reproduce the issue:
Select "settings > upgrade"
 
### Additional Notes
Internet access ok.

kizniche, 2019-06-13:
I've been able to upgrade several of my systems from 7.5.5 and I can see several other systems could as well. I think this is prrhaps a networking issue or a file corruption issue.

kizniche, 2019-06-13:
It seems this could be the fault of the RPi hardware. [Here](https://github.com/crossbario/autobahn-python/issues/1086) is the same issue appearing with the RPi3.

kizniche, 2019-06-13:
The resolution seems to be:

```
sudo apt-get install rng-tools
```

Can you try that and see if it resolves the error?

kizniche, 2019-06-13:
[This](https://stackoverflow.com/a/50590564/4731724) also suggests installing ```rng-tools``` is the solution.

SAM26K, 2019-06-13:
Thanks very much for quick reply.  rmg-tools resolved the upgrade issue (am using RPi3's).  Now having   problems with the TH16 data inputs.  Some are not working now but working through that. 
Thanks again!

SAM26K, 2019-06-13:
Upgraded to 7.5.7 but now none of the data  inputs are returning data.
Daemon errors:
2019-06-13 04:36:38,685 - ERROR - mycodo.controller_input_f7d399a3 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-06-13 04:36:40,211 - ERROR - mycodo.inputs.linux_command_066f2eb0 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1717498992
2019-06-13 04:36:40,213 - ERROR - mycodo.controller_input_066f2eb0 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-06-13 04:36:41,374 - ERROR - mycodo.inputs.linux_command_6d345eb7 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1675535472
2019-06-13 04:36:41,383 - ERROR - mycodo.controller_input_6d345eb7 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-06-13 04:36:42,351 - ERROR - mycodo.inputs.ads1256_64a5f8ec - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1918891120
2019-06-13 04:36:42,364 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - InputModule raised an exception when taking a reading: value_set() got an unexpected keyword argument 'timestamp'
2019-06-13 04:36:43,393 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - InputModule raised an exception when taking a reading: value_set() got an unexpected keyword argument 'timestamp'
2019-06-13 04:36:43,630 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - InputModule raised an exception when taking a reading: value_set() got an unexpected keyword argument 'timestamp'
2019-06-13 04:36:43,637 - ERROR - mycodo.controller_input_352ca283 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-06-13 04:36:44,471 - ERROR - mycodo.inputs.linux_command_aaf664a6 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1683928176
2019-06-13 04:36:44,474 - ERROR - mycodo.controller_input_aaf664a6 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-06-13 04:36:45,604 - ERROR - mycodo.inputs.linux_command_f8c56ec5 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1834996848
2019-06-13 04:36:46,627 - ERROR - mycodo.inputs.linux_command_827843a8 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1818211440
2019-06-13 04:36:47,567 - ERROR - mycodo.inputs.linux_command_7e3dc86d - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1801426032
2019-06-13 04:36:49,482 - ERROR - mycodo.inputs.linux_command_2f48b990 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-06-13 04:36:49,483 - ERROR - mycodo.controller_input_2f48b990 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-06-13 04:36:50,009 - ERROR - mycodo.inputs.linux_command_b1ca893c - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1784640624
2019-06-13 04:36:50,705 - ERROR - mycodo.inputs.linux_command_adff79c3 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1759462512
2019-06-13 04:36:51,680 - ERROR - mycodo.inputs.linux_command_07360015 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1776247920
2019-06-13 04:36:53,043 - ERROR - mycodo.inputs.linux_command_977ca0cc - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1751069808
2019-06-13 04:36:53,666 - ERROR - mycodo.inputs.linux_command_f7d399a3 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1734284400
[... cut ...]
2019-06-13 04:36:58,950 - ERROR - mycodo.inputs.linux_command_aaf664a6 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1683928176
2019-06-13 04:37:00,624 - ERROR - mycodo.inputs.linux_command_f8c56ec5 - InputModule raised an exception when taking a reading: (sqlite3.ProgrammingError) SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 1996441152 and this is thread id 1834996848

kizniche, 2019-06-13:
Sorry about that. I just released v7.5.8 that should fix the remaining issues with Inputs after all the refactoring I did to fix the Ruuvitag/SHT31-SG issues.

closed 2019-06-13 by kizniche