kizniche/Mycodo #664: Unable to detect dependency following install

jlb-qi, 2019-06-13:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.5.7
 
#### Problem Description

I'm trying to create a new input for sensor readings coming over MQTT. I have the readings appearing in the Pi from a remote device and now need to get them into Mycodo. Having previously attempted the Linux command option, I realised I would have to dive a bit deeper and fully add the input. While long term it would be good to be able to import a JSON object full of values, at the moment I'm just trying to get the single value from a mosquitto subscription.

This requires the paho-mqtt package which I have added as a dependency, while this has already been installed, Mycodo seems to have an issue with detecting it.
 
### Errors
The dependency logs show;

[2019-06-13 22:03:09] Dependency installation beginning. Installing: paho-mqtt

[2019-06-13 22:03:09] 
[2019-06-13 22:03:09] #### Installing/updating paho-mqtt (pip-pypi)
[2019-06-13 22:03:13] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2019-06-13 22:03:14] Requirement already up-to-date: paho-mqtt in ./env/lib/python3.5/site-packages (1.4.0)

[2019-06-13 22:03:17] End install of paho-mqtt

[2019-06-13 22:03:17] 
[2019-06-13 22:03:17] #### Setting permissions

[2019-06-13 22:03:18] #### All Dependencies have been installed.

And yet afterwards I still get returned to the dependencies required screen showing;

Error: Add Input: The MQTT device you're trying to add has unmet dependencies: paho-mqtt

The unmet dependencies are the following python modules:
-  paho-mqtt

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Add the paho-mqtt module to a new input
2. Attempt to add the new input to your setup in Mycodo
 
###  Additional Notes

I have tried deleting the .dependency file already but to no avail

kizniche, 2019-06-13:
Can you attach the input so I can review the code?

jlb-qi, 2019-06-14:
At the moment it's just a copy of the linux input but with the dependency added as well as some descriptive changes to the Input Information

jlb-qi, 2019-06-14:
"""
# coding=utf-8
from mycodo.inputs.base_input import AbstractInput
from mycodo.utils.system_pi import cmd_output
from mycodo.utils.system_pi import str_is_float

# Measurements
measurements_dict = {
    0: {
        'measurement': 'humidity',
        'unit': 'percent'
    }
}

# Input information
INPUT_INFORMATION = {
    'input_name_unique': 'MQTT',
    'input_manufacturer': 'ABC',
    'input_name': 'MQTT',
    'measurements_name': 'Input from MQTT subscription',
    'measurements_dict': measurements_dict,

    'options_enabled': [
        'measurements_select',
        'period',
        'cmd_command',
        'log_level_debug'
    ],
    'options_disabled': ['interface'],

    'dependencies_module': [
        ('pip-pypi', 'paho-mqtt', 'paho-mqtt'),
        ('pip-git', 'paho-mqtt','git://github.com/eclipse/paho.mqtt.python')
    ],
    'enable_channel_unit_select': True,
    'interfaces': ['pi'],
    'cmd_command': 'shuf -i 50-70 -n 1',
}

class InputModule(AbstractInput):
    """ A sensor support class that returns a value from a command """

    def __init__(self, input_dev, testing=False):
        super(InputModule, self).__init__(input_dev, testing=testing, name=__name__)

        if not testing:
            self.command = input_dev.cmd_command

    def get_measurement(self):
        """ Determine if the return value of the command is a number """
        self.return_dict = measurements_dict.copy()

        out, _, _ = cmd_output(self.command)

        if str_is_float(out):
            list_measurements = [float(out)]
        else:
            self.logger.error(
                "The command returned a non-numerical value. "
                "Ensure only one numerical value is returned "
                "by the command.")
            return

        for channel in self.device_measurements:
            if self.is_enabled(channel):
                self.return_dict[channel]['unit'] = self.device_measurements[channel].unit
                self.return_dict[channel]['measurement'] = self.device_measurements[channel].measurement
                self.return_dict[channel]['value'] = list_measurements[channel]

        return self.return_dict

jlb-qi, 2019-06-14:
I tried both the pypi and git options but neither seem to be detected

kizniche, 2019-06-14:
Your dependency line is the issue:

```
('pip-pypi', 'paho-mqtt', 'paho-mqtt')
```

It should read:

```
('pip-pypi', 'paho', 'paho-mqtt')
```

```paho``` is the actual module that's installed, while ```paho-mqtt``` is the name in pypi that is used to install it.

That is, you would typically install with:

```
pip install paho-mqtt
```

and use the module with:

```
import paho
```

kizniche, 2019-06-14:
I'd like to work with you developing a built-in MQTT Input that has a variable number of keys/values that can be returned and stored in Mycodo. Let me know how your code development goes now that you can inport and add the module and I'll help with the expanding the functionality, much like I did with the TTN Input (search Inputs for "TTN Integration" to see how I implemented an Input that can return a variable number of measurements).

jlb-qi, 2019-06-14:
Much appreciated Kyle! I will certainly let you know how the code develops and once it's in an MVP state I'll create a pull request for review. Thanks again and have a great weekend

jlb-qi, 2019-06-19:
@kizniche managed to get some time to work on this yesterday and  so far I've managed to get the input set up and values from the sensor (over MQTT) to show in the daemon logs. I can't quite figure out how to get the measurements into the database though and I think it might be something to do with the daemon controller not setting the ID properly. Here are the messages when going through the steps of creation, activation, deactivation and deletion.

The messages; 

**When creating;** 
    Success: MQTT Input with ID 3 (32c79c3d-ea3d-4049-b8c2-6023b5229402) successfully added
    Success: Add Input

**When activating;**
    Success: Input Activate (SQL)
    Success: Activate Input
    Error: Error: result expired

**When deactivating;**
    Success: Input Deactivate (SQL)
    Error: Input controller with ID 32c79c3d-ea3d-4049-b8c2-6023b5229402 not found

**When Deleting**
    Success: Input Deactivate (SQL)
    Success: Delete device_measurements with ID: 0372f9f3-4072-44a7-9850-9871de546d14
    Success: Delete input with ID: 32c79c3d-ea3d-4049-b8c2-6023b5229402
    Success: Delete Input

    Error: Input controller with ID 32c79c3d-ea3d-4049-b8c2-6023b5229402 not found

jlb-qi, 2019-06-19:
`# coding=utf-8
import datetime
import time

import requests
from flask_babel import lazy_gettext

import paho.mqtt.client as mqtt

from mycodo.config import SQL_DATABASE_MYCODO
from mycodo.databases.models import Conversion
from mycodo.databases.models import DeviceMeasurements
from mycodo.databases.models import Input
from mycodo.databases.utils import session_scope
from mycodo.inputs.base_input import AbstractInput
from mycodo.utils.database import db_retrieve_table_daemon
from mycodo.utils.influx import add_measurements_influxdb
from mycodo.utils.influx import parse_measurement

MYCODO_DB_PATH = 'sqlite:///' + SQL_DATABASE_MYCODO

# Measurements
measurements_dict = {
    0: {
        'measurement': 'humidity',
        'unit': 'percent'
    }
}

# Input information
INPUT_INFORMATION = {
    'input_name_unique': 'MQTT',
    'input_manufacturer': 'ABC_test',
    'input_name': 'MQTT Protocol',
    'measurements_name': 'Variable measurements',
    'measurements_dict': measurements_dict,
    'measurements_use_same_timestamp': False,

    'options_enabled': [
        'custom_options',
        'measurements_select',
        'period',
        'log_level_debug'
    ],
    'options_disabled': ['interface'],

    'interfaces': ['Mycodo'],

    'dependencies_module': [
        ('pip-pypi', 'paho', 'paho-mqtt')],

    'custom_options': [
        {
            'id': 'mqtt_hostname',
            'type': 'text',
            'default_value': 'raspberrypi',
            'required': True,
            'name': lazy_gettext('Application ID'),
            'phrase': lazy_gettext('The hostname of the MQTT server')
        },
        {
            'id': 'mqtt_channel',
            'type': 'text',
            'default_value': '/rsu/pi',
            'required': True,
            'name': lazy_gettext('MQTT channel'),
            'phrase': lazy_gettext('The MQTT channel to subscribe to')
        },
        {
            'id': 'mqtt_keepalive',
[... 149 lines cut ...]
        #client.publish("/rsu/pi","from chattop")
        """ Determine if the return value of the command is a number """

        out = client.on_message
        self.logger.info(out)

        if str_is_float(out):
            list_measurements = [float(out)]
            self.logger.info(list_measurements)
        else:
            self.logger.error(
                "The command returned a non-numerical value. "
                "Ensure only one numerical value is returned "
                "by the command.")
            return

        for channel in self.device_measurements:
            if self.is_enabled(channel):
                self.return_dict[channel]['unit'] = self.device_measurements[channel].unit
                self.return_dict[channel]['measurement'] = self.device_measurements[channel].measurement
                self.return_dict[channel]['value'] = list_measurements[channel]

        return self.return_dict
        '''
`

kizniche, 2019-06-21:
Just a quick update to say I got the MQTT Input functioning. Here's a log output of the message ```1234``` being received:

```
2019-06-20 20:13:40,733 - INFO - mycodo.controller_input_2a2ce3f8 - Activated in 323.0 ms
2019-06-20 20:13:40,733 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - Connecting MQTT callback functions
2019-06-20 20:13:40,734 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - MQTT callback functions connected
2019-06-20 20:13:40,741 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - Connected to: localhost as mycodo_mqtt_client
2019-06-20 20:13:40,742 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - Subscribed to 'topic/test'
2019-06-20 20:13:40,743 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - Connected to 'topic/test', rc: 0
2019-06-20 20:13:40,744 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - Subscribing to topic: topic/test, 1, (0,)
2019-06-20 20:13:56,011 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - TEST00
2019-06-20 20:13:56,012 - INFO - mycodo.inputs.mqtt_2a2ce3f8 - Message: 1234
```

I installed the mosquitto MQTT broker on the Pi Mycodo is installed on, with:

```
sudo apt install mosquitto
```

Then set the Input Hostname to ```localhost```, Client ID to ```mycodo_mqtt_client_01```, Channel ID to ```topic/test```, then activated the Input. I then saved the code below on the same Pi to ```~/Mycodo/test_mqtt_client.py```:

```python
import paho.mqtt.client as mqtt
client = mqtt.Client("mycodo_mqtt_client_02")
client.connect("localhost", port=1883)
client.publish("topic/test", '1234')
client.disconnect()
```

Then executed it using the Mycodo virtualenv:

```
~/Mycodo/env/bin/python ~/Mycodo/test_mqtt_client.py
```

The code is not currently committed to github so you cannot test, however I will try to make a commit tonight and give instructions how to start testing the latest code.

kizniche, 2019-06-21:
I just pushed all the MQTT code. You can get it by [Upgrading to the master branch](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

Once you've done that, add the MQTT Input, set your ```Hostname``` and ```Port``` for the MQTT broker, and your desired ```MQTT Client ID```. Set the number of channels with the option ```Number of Measurements```. After saving you will see in the expanded Input, under Measurement Settings, that number of measurements appear. Change the ```Name``` of each to the channel name you want to subscribe to. Make sure you set a ```Measurement Unit``` for each. Activate the Input and whenever a float value with any of those channels are published, it will be written to the Mycodo database and will appear on the Dashboard and Info pages and be able to be used everywhere else throughout Mycodo.

Let me know if you experience any issues. I'll wait to make a release until you can test and respond with whether it's working or not.

Enjoy!

jlb-qi, 2019-06-21:
got an error when navigating to the Setup > Data screen

jlb-qi, 2019-06-21:
`Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1855, in page_data
    devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 135, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 117, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data.html", line 3, in top-level template code
    {% set help_page = ["data", dict_translation['data']['title']] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 262, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data.html", line 48, in block "body"
    {% include 'pages/data_options/input.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/input.html", line 141, in top-level template code
    {% include 'pages/data_options/input_options/TEMPLATE_OPTIONS_Measurements_Convert.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/input_options/TEMPLATE_OPTIONS_Measurements_Convert.html", line 92, in top-level template code
    {% if 'measurements_select_measurement_unit' in dict_inputs[each_input.device]['options_enabled'] %}
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'MQTT'`

jlb-qi, 2019-06-21:
I will have a look to debug it myself if I can

jlb-qi, 2019-06-24:
Hey Kyle, I managed to clear a few hours to test the code and managed to update it to be able to parse incoming JSON object with multiple measurements. I tried to create a seperate mqtt branch in the repo but to no avail. I updated the gist tho https://gist.github.com/jlb-qi/332553e6cfb2793e2b15534f129ddc30 so have a wee look and let me know what you think

jlb-qi, 2019-06-24:
it still needs a bit of work but this should definitely get it up and running

jlb-qi, 2019-06-24:
very satisfying 👍

[image]

kizniche, 2019-06-24:
I merged your edits with the module currently in the master branch. I have a few questions about what you did so I can remove extraneous code and make this more simple for anyone trying to understand what's going on. I added "TODO" notes for what I think can be removed.

1. Can you provide steps for how you set up your MQTT broker and published measurements (code/commands)?

2. Why is there an extra subscription in addition to the measurement channels? I removed this Input Option because the measurements that are added are what are subscribed to. Isn't only the channels specified in the measurements sufficient?

```python
        {
            'id': 'mqtt_channel',
            'type': 'text',
            'default_value': '/rsu/pi',
            'required': False,
            'name': lazy_gettext('Subscription Channel'),
            'phrase': lazy_gettext('The name of the thread with which to subscribe to')
        }

...

    def subscribe(self):
        """ Set up the subscriptions to the proper MQTT channels to listen to """
        try:
            for channel in self.channels_measurement:
                self.client.subscribe(self.mqtt_channel)
                #self.logger.debug("Subscribed to channel '{}'".format(self.mqtt_channel))
                self.client.subscribe(self.channels_measurement[channel].name)
                self.logger.debug("Subscribed to  measurement'{}'".format(self.channels_measurement[channel].name))
        except:
            self.logger.error("Could not subscribe to MQTT topic '{}'".format(
                self.mqtt_channel))
```

3. Isn't the payload channel irrelevant because the Mycodo channel is what's stored in the database and displayed to the user?

```python
                    payload_channel = payload_measurements.index(
                        self.channels_measurement[channel].name)
                    self.logger.debug(
                        "measurement: '{}' is mycodo channel: '{}' and "
                        "payload channel: '{}'".format(
                            self.channels_measurement[channel].name,
                            channel,
                            payload_channel))
```

jlb-qi, 2019-06-25:
In answer to your questions;

1. Certainly! I'll work on the documentation and send it over. I assume you prefer in a Markdown format? Otherwise just let me know.

2.  You're right, there's an extra line here I forgot to remove. Just need to take out the line.

```python
self.client.subscribe(self.channels_measurement[channel].name)
```

3. The payload channel is required for matching the measurement value in the incoming JSON with the right channel in the database. It'll probably make more sense when I get the documentation together. 

In addition, the JSON format that is required at the moment is a little unrefined and I'd like to extend it in future. This was enough to get it working tho. Again, will make more sense when I get the documentation together.

kizniche, 2019-06-25:
> You're right, there's an extra line here I forgot to remove. Just need to take out the line.

You have it backwards. This needs to stay:

```python
self.client.subscribe(self.channels_measurement[channel].name)
```

This needs to be removed:

```python
        {
            'id': 'mqtt_channel',
            'type': 'text',
            'default_value': '/rsu/pi',
            'required': False,
            'name': lazy_gettext('Subscription Channel'),
            'phrase': lazy_gettext('The name of the thread with which to subscribe to')
        }

...and...

self.client.subscribe(self.mqtt_channel)
```

```self.mqtt_channel``` is a single text string defined by the user. This means only one channel can be subscribed to with this method. However, ```self.channels_measurement``` holds the names of every measurement the user specifies (one or more measurements), which allows for several channels to be subscribed to. The use of ```self.mqtt_channel``` in addition to the variable number of measurements used as channels is redundant and will only confuse the user.

> The payload channel is required for matching the measurement value in the incoming JSON with the right channel in the database. 

I see this now, and would like to ask if this is a standard practice for MQTT transmissions. Using both a Mycodo channel and payload channel from a JSON string seems overly complex and confusing. Why not just subscribe to a channel that returns 1 value per channel?

kizniche, 2019-06-25:
I suppose what I'm asking is: What is the benefit of subscribing to one channel with a JSON payload rather than separate channels with a single value payload?

jlb-qi, 2019-06-26:
A very fari point. Tbh I think I had anchored my idea in the notion that a remote sensor unit (comprised of multiple sensors) would provide a single subscription. I did a bit of reading on MQTT and ESP8266 best practice and it think you're right about the seperate channel/topic per measurement. I tested your original code with a couple of sensor unit with a variety of sensors and they all work so I would say good to go with the original code you sent through.

jlb-qi, 2019-06-26:
The only thing I found during testing was that if you try to add a measurement but don't specify the units, you get an error message that **Error: Modify Measurement: list index out of range** rather than something like **Please select unit of measurement**. A minor detail but worth mentioning

kizniche, 2019-07-29:
How has the new Input been working for you?

jlb-qi, 2019-08-09:
Hi Kyle, apologies for the belated reponse, back at work now and time has been greatly limited. Yes the input has been working a treat! I have a bunch of ESP8266s with various sensor configurations in a variety of environments now all feeding into Mycodo. Real smart system : )

kizniche, 2019-08-17:
Glad to hear. I'll close this issue now that everything is working.

closed 2019-08-17 by kizniche