kizniche/Mycodo #665: high CPU rPi 3B

SAM26K, 2019-06-15:
## Mycodo Issue Report:
 Mycodo Version: 7.5.8 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 6333b0832b3d
Daemon Status: Running 
Daemon Process ID: 1108 
Daemon RAM Usage: 110.196 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 54.984 MB 
Frontend Virtualenv: Yes
#### Problem Description
High CPU usage slow response to commands
 
### Errors
  2019-06-13 21:22:07,639 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
2019-06-13 21:22:11,112 - ERROR - mycodo.daemon - Could not find Output Controller
2019-06-13 21:33:47,277 - ERROR - mycodo.daemon - Could not query output state: You must setup() the GPIO channel first
2019-06-13 21:33:47,714 - ERROR - mycodo.daemon - Could not query output state: You must setup() the GPIO channel first
2019-06-13 21:33:50,264 - ERROR - mycodo.daemon - Could not query output state: You must setup() the GPIO channel first
2019-06-13 21:57:35,184 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
2019-06-13 21:57:44,772 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
2019-06-13 21:57:45,880 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
2019-06-13 21:57:46,894 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
2019-06-13 21:57:53,029 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
2019-06-13 21:58:41,684 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state

htop screenshot:
[image]
Controls using high cpu are GPIO hardwired controls that seem to  physically work ok but generating daemon errors.

SAM26K, 2019-06-15:
better screen shots :
[image]

Output control:
[image]

kizniche, 2019-06-15:
Is this an excerpt from the startup sequence? If so, can you provide log lines from the full startup?

kizniche, 2019-06-15:
I just added a more thorough check to ensure the output controller has started before continuing with the daemon startup sequence. This should be much better than the mere 0.5 second wait time I previously had in its place.

kizniche, 2019-06-15:
If you want to check if this alleviates your issue, the master branch should be safe to upgrade to.

SAM26K, 2019-06-16:
Still having high cpu% with master changes.  The second to last screen shot above is htop command from ssh.  It's listed in order of CPU used.  It looks like a bunch of the output controls are generating daemon errors.  The status of these controls are constantly checked by the other rpi through the network.
Attached is the last 900 lines of the daemon error log right after a system restart.
[daemonerr1.txt](https://github.com/kizniche/Mycodo/files/3293760/daemonerr1.txt)

SAM26K, 2019-06-16:
Update:  If I disable the bash data controls on the other rpi, the CPU% comes way down.  Tried the following command on the other rpi:
`sshpass -p password ssh mycodo@192.168.1.51 mycodo-client --output_state f7809276-f6c3-4173-8ca4-14916b77ac87`

and get this error:
`Traceback (most recent call last):
  File "/usr/bin/mycodo-client", line 41, in <module>
    from mycodo.config import INFLUXDB_DATABASE
ImportError: No module named 'mycodo'
`
Those commands used to work ok. Had similar problem when trying to use new 'get measurement' command for ADS1256.  It like there is a      path    problem to mycodo.

SAM26K, 2019-06-16:
Confirmed that all the linux remote ssh commands no longer work. Not sure which version it changed but V7.3.1 worked. TIA!

kizniche, 2019-06-16:
> Confirmed that all the linux remote ssh commands no longer work. Not sure which version it changed but V7.3.1 worked. TIA!

Working on a fix for this.

SAM26K, 2019-06-16:
Wow that's fast thanks very much!  Safe enough to try yet?

kizniche, 2019-06-16:
What is executing ```mycodo-client --output_state```? It looks like you may have an infinite loop constantly executing again and again after the exception.

SAM26K, 2019-06-16:
There are some data controls that poll the state of the 8 channel relay pcb. No infinite loops that I know of.

SAM26K, 2019-06-16:
The controls continue to try to  poll the output  states.  Since the bash commands   fail they continue to generate the errors.

SAM26K, 2019-06-16:
Let me know when its safe to try the master.  Thanks very much!

kizniche, 2019-06-16:
You can try master now. Do you have a complete log that includes the startup sequence before the errors begin?

SAM26K, 2019-06-16:
Great news.  Remote bash commands all seem to be working again!
How do I create a complete start up log?
Thanks very much again ;and look forward  to   incorporating the new ADS1256 data access.
My ultimate goal is to have two raspberry pi's that work in  parallel so if there is a     problem with one of  the rpi's the  other can take over.

SAM26K, 2019-06-16:
Re-enabled all bash controls and CPU%  is under control.    Thanks again!

closed 2019-06-16 by SAM26K

kizniche, 2019-06-16:
I'm merely talking about the part of the log where the daemon starts up. You can induce a restart from the config menu (though doing so now won't produce the errors). If you show enough log history you will eventually find where the daemon starts. You may find it easier just to download the whole log to your computer and reverse searching for something like "Daemon started in"

kizniche, 2019-06-16:
Glad the two fixes worked and the daemon isn't having issues with the output controller.