kizniche/Mycodo #667: get-measurement; how to case text to decimal for display on dashboard

SAM26K, 2019-06-19:
## Mycodo Issue Report:
Mycodo Version: 7.5.10 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 6333b0832b3d
Daemon Status: Running 
Daemon Process ID: 18204 
Daemon RAM Usage: 92.132 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 63.18 MB 
Frontend Virtualenv: Yes
 
#### Problem Description
mycodo_client get-measure:
Unable to cast parsed value to decimal.  Everything produces daemon errors
### Errors
2019-06-18 01:01:23,389 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-06-18 01:01:52,529 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.

### Steps to Reproduce the issue:
Example of bash file that seems to return real values but still   produces errors above:

`#!/bin/bash
err=$((sshpass -p <password> ssh mycodo@192.168.1.17 /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 64a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0) 2>&1 >/dev/null)

#echo $err
out=$(sshpass -p <password> ssh mycodo@192.168.1.17 /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 64a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0)
#echo out
if [[ $out == *"SUCCESS;"* ]]; then
  #echo "1"
    echo ${out:8:8 | bc + 0}
 
else
 if [[ $out == *"ERROR;"* ]]; then
  echo "0"
 fi
fi`
 
### Additional Notes
No way to display values in measurement desktop readout.

SAM26K, 2019-06-19:
Not sure why pasting text deletes some of the spaces,  the code above is indented in the bash file.
Above code is located at /home/pi/moduser1/getA0.sh and produces the following correct output when executed at a terminal prompt:
```
pi@raspberrypi3_3:~ $ /home/pi/moduser1/getA0.sh
2.251506
```

SAM26K, 2019-06-19:
Screen shots of controls: First one is the rpi with the ADS1256 Waveshare shield PCB. The second one is the rpi with the 8 channel relay PCB.
[image]
[image]

SAM26K, 2019-06-20:
Tried several methods to cast to floating pt and same daemon errors (non-numeric returned).  Possibly a python data input control would solve the   problem?  Didn't see that one listed.

kizniche, 2019-06-21:
Edit ~/Mycodo/mycodo/inputs/linux_command.py and change lines 52-59 to look like this:

```python
        if str_is_float(out):
            list_measurements = [float(out)]
        else:
            self.logger.error(
                "The command returned a non-numerical value. "
                "Ensure only one numerical value is returned "
                "by the command. Value returned: '{}'".format(out))
            return
```

Save, then restart the daemon. You will be shown what's being returned in the error message now. It may be that there's extra characters like a return attached to the string and you may need to strip them before returning the value.

SAM26K, 2019-06-21:
Tried the simple edit above and now I can't open the web ui.  Restored the original linux_command.py file and same problem.   Used stretch desktop text editor to edit the file.  Is there some python permission that needs to be reset?  Should I use nano to edit? TIA.

kizniche, 2019-06-21:
You likely made a syntax error in the code. You can manually start the frontend to see what the error is by running:

```bash
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/lock/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

You should have the server crash and show you the line of code in the linux_command.py that's causing the issue. Fix the issue and restart the service with:

```
sudo service mycodoflask restart
```

SAM26K, 2019-06-21:
This is errors I'm getting.  Don't see linux_command.py:
`pi@raspberrypi3_3:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/lock/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2019-06-20 19:05:52 -0700] [7896] [INFO] Starting gunicorn 19.9.0
[2019-06-20 19:05:52 -0700] [7896] [INFO] Listening at: unix:/var/run/mycodoflask.sock (7896)
[2019-06-20 19:05:52 -0700] [7896] [INFO] Using worker: gthread
[2019-06-20 19:05:52 -0700] [7908] [INFO] Booting worker with pid: 7908
[2019-06-20 19:05:57 -0700] [7908] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/gthread.py", line 104, in init_process
    super(ThreadWorker, self).init_process()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py", line 129, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py", line 138, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py", line 52, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py", line 41, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/util.py", line 350, in import_app
    __import__(module)
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 10, in <module>
    from mycodo.mycodo_flask.app import create_app
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 35, in <module>
    from mycodo.mycodo_flask import routes_page
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 84, in <module>
    from mycodo.mycodo_flask.forms import forms_input
  File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_input.py", line 34, in <module>
    class InputAdd(FlaskForm):
  File "/home/pi/Mycodo/mycodo/mycodo_flask/forms/forms_input.py", line 36, in InputAdd
    dict_inputs = parse_input_information()
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 128, in parse_input_information
    input_custom = load_module_from_file(full_path)
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 69, in load_module_from_file
    input_custom = importlib.util.module_from_spec(spec)
  File "<frozen importlib._bootstrap>", line 574, in module_from_spec
AttributeError: 'NoneType' object has no attribute 'loader'
[2019-06-20 19:05:57 -0700] [7908] [INFO] Worker exiting (pid: 7908)
[2019-06-20 19:05:58 -0700] [7896] [INFO] Shutting down: Master
[2019-06-20 19:05:58 -0700] [7896] [INFO] Reason: Worker failed to boot.`

kizniche, 2019-06-21:
Make sure there weren't any extra files saved in that input directory, such as a temp editor file or something. Also, if that fails, delete the linux_command.py and download and save the original from https://raw.githubusercontent.com/kizniche/Mycodo/master/mycodo/inputs/linux_command.py

SAM26K, 2019-06-21:
Extra files was the wui loading  problem.  Going back to trying your edit again.

SAM26K, 2019-06-21:
Edit seems to be working now.  Got this error:

`2019-06-20 19:40:35,198 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''`

kizniche, 2019-06-21:
It appears nothing is being returned.

SAM26K, 2019-06-21:
When I execute the bash file (getA0.sh) at a command prompt it returns a good number and no error.

kizniche, 2019-06-21:
Try adding these debug lines around the command being executed. Make sure you have ```Log Level: Debug``` enabled for the Input to see these log lines.

```python
        self.logger.debug("Command being executed: {}".format(self.command))
        out, err, status = cmd_output(self.command)
        self.logger.debug("Command returned: Status: {}, Error: {}".format(err, status))
```

SAM26K, 2019-06-21:
Not sure how to arrange those code lines.  This is the  bash file first attempt:

> `#!/bin/bash
> self.logger.debug("Command being executed: {}".format(self.command))
> out=$(sshpass -p password ssh mycodo@192.168.1.17 /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 64a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0)
> #echo out
> if [[ $out == *"SUCCESS;"* ]]; then
>   #echo "1"
>   VOLT=$(printf "%.6f" ${out:8:8 | bc})
>   echo $VOLT | sed "s/ //g"
>   #echo $VOLT
> fi
> out, err, status = cmd_output(self.command)
> self.logger.debug("Command returned: Status: {}, Error: {}".format(err, status))

Get these errors from a terminal execution (no error before adding debug lines):

pi@raspberrypi3_3:~ $ /home/pi/moduser1/getA0.sh
/home/pi/moduser1/getA0.sh: line 2: syntax error near unexpected token `"Command being executed: {}".format'
/home/pi/moduser1/getA0.sh: line 2: `self.logger.debug("Command being executed: {}".format(self.command))'`

kizniche, 2019-06-21:
It's supposed to go in linux_command.py around line 70. Look for this to replace:

```python
        out, _, _ = cmd_output(self.command)
```

SAM26K, 2019-06-21:
Debug seems to be working now.  Daemon debug log

> 2019-06-20 21:23:19,322 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-20 21:23:19,746 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-20 21:26:19,144 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-20 21:26:19,440 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-20 21:29:19,143 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-20 21:29:19,428 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0

SAM26K, 2019-06-21:
When I execute getA0.sh it takes about  8 seconds to return a value.  Set  polling interval to 180 secs.

SAM26K, 2019-06-21:
What does the 'b' mean in the error output Value returned: 'b'''?

SAM26K, 2019-06-21:
Could this be a systemic problem with returning a decimal number from bash to python?

SAM26K, 2019-06-21:
Also tried this code.  Works ok running in a terminal session but still nothing returned to linux_command.py:

`   VOLT=${out:8:8 | bc}
  #echo $VOLT | sed "s/ //g"
  #echo $VOLT
  printf '%5.5f\n' $VOLT
`

kizniche, 2019-06-21:
'b' indicates it's a byte object, however it's empty, so that indicates there's no returned value. By chance are you not executing /bin/bash in your command? That is: ```/bin/bash /home/pi/moduser1/getA0.sh```

kizniche, 2019-06-21:
Does a script with this code in it work for your Bash Command Input?

test_output.sh

```bash
#!/bin/bash
echo "7.2"
```

kizniche, 2019-06-21:
Additionally, we should verify the script is even executing at all from the Mycodo Input. The easiest way to do this is to add the following line as the first line to execute in your script:

```bash
touch /home/pi/test_file
```

Have the Mycodo Input attempt to execute the script at least once, then check for the presence of the file at ```/home/pi/test_file```

SAM26K, 2019-06-21:
We're getting closer.  With only the following in the bash file I get good numbers on the live screen and no apparent errors (and test_file shows up at the path).:

``#!/bin/bash
touch /home/pi/test_file
echo "7.2"

The following doesn't work even though I replaced the output value with a constant 7.2;

 #!/bin/bash
out=$(sshpass -p password ssh mycodo@192.168.1.17 /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 64a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0)
if [[ $out == *"SUCCESS;"* ]]; then
         VOLT=${out:8:8 | bc -l}
          echo "7.2"
  else
    print '%5.5f\n' "1.0" | sed "s/ //g"
fi

If I execute the above in a terminal session it works but takes about 14 seconds to return a value.  If used in a data control it produces errors and no numbers,  I think what is happening is that linux_command,py is not allowing enough time for the value to return before it gives up and generates the error.

kizniche, 2019-06-21:
I'll test some more tonight. I can confirm the output command timeout period is 360 seconds by default, so it's not killing the command for taking too long. How long after activating the Input does it error with the return value? Right away or after how long it would normally take to return a value?

SAM26K, 2019-06-22:
Activated data input control at 17:43 with pc clock synced with raspberry pi and got these errors.  It looks like the errors are generated almost right away.

```
2019-06-21 17:43:10,339 - INFO - mycodo.controller_input_b7b32422 - Activated in 2432.3 ms
2019-06-21 17:43:10,340 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-21 17:43:11,252 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-21 17:43:11,253 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-06-21 17:46:09,930 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-21 17:46:10,389 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-21 17:46:10,390 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
```

SAM26K, 2019-06-22:
Doing more testing but it looks like linux_command.py is not waiting long enough for the ssh command to complete.  Seems to pass over that command and continue execution after the ssh line.

kizniche, 2019-06-22:
I think I know what the issue is... You're not using the full path to ```sshpass```

SAM26K, 2019-06-22:
Googling now but what should it be?

SAM26K, 2019-06-22:
The path to the actual ssh command?

kizniche, 2019-06-22:
Yes, like ```/usr/bin/sshpass```

You can find it with ```whereis sshpass```

SAM26K, 2019-06-22:
9pi@raspberrypi3_3:~ $ whereis sshpass
sshpass: /usr/bin/sshpass /usr/share/man/man1/sshpass.1.gz

Thats the  return. Should I use everything before sshpass.1.gz or the whole path? TIA!

SAM26K, 2019-06-22:
Looks like /usr/bin/sshpass is valid.  Evaluating that change now.

SAM26K, 2019-06-22:
Didn't seem to help.  It appears the ssh  command line is passed over before it has time to complete.

```
2019-06-21 21:17:33,606 - INFO - mycodo.controller_input_b7b32422 - Activated in 387.0 ms
2019-06-21 21:17:33,607 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-21 21:17:33,946 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-21 21:17:33,947 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-06-21 21:18:33,487 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-21 21:18:33,790 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-21 21:18:33,791 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-06-21 21:19:33,489 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command being executed: /home/pi/moduser1/getA0.sh
2019-06-21 21:19:33,808 - DEBUG - mycodo.inputs.linux_command_b7b32422 - Command returned: Status: None, Error: 0
2019-06-21 21:19:33,809 - ERROR - mycodo.inputs.linux_command_b7b32422 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-06-21 21:19:33,809 - ERROR - mycodo.controller_input_b7b32422 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
```

SAM26K, 2019-06-22:
This is the code that seems to be skipped over before it returns a value and yet works in a terminal session:

`out=$(/usr/bin/sshpass -p password ssh mycodo@192.168.1.17 /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 64a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0)
`

kizniche, 2019-06-22:
Perhaps try using the full path to ```aah``` as well.

SAM26K, 2019-06-22:
Not sure what you mean.  I thought I was   using the full path.?

kizniche, 2019-06-22:
you changed ```sshpass```, not ```ssh``` later in the command.

SAM26K, 2019-06-22:
oh ssh? will try that

kizniche, 2019-06-22:
Perhaps [attempting with Python's paramiko](https://stackoverflow.com/a/18905987/4731724) may be an easier solution that trying to get ssh to play nicely without a keyboard.

kizniche, 2019-06-22:
I didn't have a chance to test anything. I'll try to in the next day or so.

SAM26K, 2019-06-22:
Tried adding /usr/bin to ssh path. same result.  doesn't wait  for  return value.

SAM26K, 2019-06-22:
Thanks for the help.

SAM26K, 2019-06-22:
let me know if you want remote access and will set that up  again.

SAM26K, 2019-06-23:
Looking into python paramiko ssh alternative.  Is it ok to use a python script file with a linux_command data input (.py instead   of .sh) in the controls command path?

kizniche, 2019-06-23:
It's okay, though I would recommend using the Mycodo virtualenv. So, if you wanted to install a pip package to the virtualenv, you would use this command:

```
~/Mycodo/env/bin/pip install paramiko
```

And to execute a script that uses that library from the Bash Command Input, use this as the command:

```
~/Mycodo/env/bin/python ~/path/to/script.py
```

SAM26K, 2019-06-24:
Going the para;miko route. Major learning curve  but more secure.  Definitely Linux ssh  command is not executed  by Python.

SAM26K, 2019-06-25:
One thing to note is that my boolean bash inputs that check  GPIO status on a remote    pi still works using similair Linux ssh commands.  It's  only the new get_measurement command for the ADS1256 analog A/D converter that doesn't work.

When you get the time can you look at it again on  your end to see why ssh doesn't work for ADS1256 but works ok for GPIO status?  Still learning about paramiko and you may solve the problem before I get anything working.

Thanks Again!

SAM26K, 2019-06-29:
Hi Kyle,
Making  progress to get paramiko working.  Following code seems to open a session but keep running into syntax errors.  Using  Python 3 raspbian stretch ide.  Any ideas?

[image]

kizniche, 2019-06-29:
Try this:

```
cmd = "/home/pi/Mycodo/env/bin/python /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement ID UNIT CHANNEL"
stdin, stdout, stderr = ssh.exec_command(cmd)
```

SAM26K, 2019-07-01:
Here's the latest try:
`import paramiko  
from select import select
import os
import paramiko
from pathlib import Path
from paramiko.ssh_exception import BadHostKeyException, AuthenticationException, SSHException
mid_cli = paramiko.SSHClient()
ssh = paramiko.SSHClient()
mid_cli.set_missing_host_key_policy(paramiko.AutoAddPolicy())
mid_cli.connect(hostname='192.168.1.17', username='pi', password='password')
io_tupple = mid_cli.exec_command('nc 192.168.1.51 22')
cmd = "/home/pi/Mycodo/env/bin/python /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 4a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0"
stdin, stdout, stderr = ssh.exec_command(cmd)`

The last line throws this error:
`Traceback (most recent call last):
  File "/home/pi/moduser1/para3.py", line 13, in <module>
    stdin, stdout, stderr = ssh.exec_command(cmd)
  File "/usr/local/lib/python3.5/dist-packages/paramiko/client.py", line 508, in exec_command
    chan = self._transport.open_session(timeout=timeout)
AttributeError: 'NoneType' object has no attribute 'open_session'`

The rpi with the ADS1256 I'm trying to get data from is 192.168.17.  The rpi I'm using to remotely access the data is 192.168.1.51.  Any ideas? Thanks in advance.

SAM26K, 2019-07-01:
> import paramiko  
from select import select
import os
import paramiko
from pathlib import Path
from paramiko.ssh_exception import BadHostKeyException, AuthenticationException, SSHException
mid_cli = paramiko.SSHClient()
ssh = paramiko.SSHClient()
mid_cli.set_missing_host_key_policy(paramiko.AutoAddPolicy())
mid_cli.connect(hostname='192.168.1.17', username='pi', password='TARA9CAT')
io_tupple = mid_cli.exec_command('nc 192.168.1.51 22')
cmd = "/home/pi/Mycodo/env/bin/python /home/pi/Mycodo/mycodo/mycodo_client.py --get_measurement 4a5f8ec-251a-4146-bd39-e17ae4e88ec2 V 0"
stdin, stdout, stderr = ssh.exec_command(cmd)

>2nd try

SAM26K, 2019-07-01:
tried setting cmd to  "ls" to simplify command and same error.   Seems to be missing a timeout  para,meter?

kizniche, 2019-07-01:
I just tested the following code to work. Read what I wrote in the comments to add your host to your client's list of known hosts.

```python
# coding=utf-8
import os
import paramiko

hostname = '10.48.44.141'
username = "pi"
password = 'password'
port = 22
command = 'ls /etc/'
client = paramiko.SSHClient()

try:
    client.load_system_host_keys()
    client.set_missing_host_key_policy(paramiko.WarningPolicy)

    # For the next line to work, you must at least once have connected from the client to the host
    # to add the host to the list of known hosts.
    # To do this, run the following command from the client computer:
    # ssh user@hostIP
    # and when prompted "Are you sure you want to continue connecting?", choose "yes"
    client.load_host_keys(os.path.expanduser('~/.ssh/known_hosts'))

    client.connect(hostname, port=port, username=username, password=password)
    stdin, stdout, stderr = client.exec_command(command)
    print(stdout.read().decode())
finally:
    client.close()
```

kizniche, 2019-07-01:
You can alternatively crate a known_hosts file that you can use to distribute with your script, in order to run on different clients, by doing the following (from https://stackoverflow.com/a/42177236/4731724):

Change this:

```
client.load_host_keys(os.path.expanduser('~/.ssh/known_hosts'))
```

to this:

```
client.load_host_keys(os.path.join(os.path.dirname(__file__), 'known_hosts'))
```

Then, run this command to create the ```known_hosts``` file. Make sure this file is in the same directory as the script.

```
ssh -o GlobalKnownHostsFile=/dev/null -o UserKnownHostsFile=./known_hosts user@host
```

kizniche, 2019-07-01:
The reason I mention is because ```set_missing_host_key_policy(paramiko.AutoAddPolicy())``` is insecure. Although I realize you're probably not caring too much about security and man-in-the-middle attacks.

SAM26K, 2019-07-02:
Great news.  Finally got a success return with your code guidance above.  Now all I have to do is parse the data which should be simple enough.
Not too worried about security yet but may try to improve it.
Thanks very much for the help.

SAM26K, 2019-07-02:
Update: Unfortunately I'm back to the original error with null data   returned  and mycodo  daemon error for the   input     control..  Same error as Linux  bash   version.. Will research further  and reply with more specifics soon   but    initial tests seem  to be  the same as    using  Linux   bash code.

SAM26K, 2019-07-02:
Same errors  as  with the Linux version.  The error is returned in less than a second.

> 2019-07-02 00:21:48,803 - INFO - mycodo.controller_input_80d83ca4 - Deactivated in 0.4 ms
2019-07-02 00:24:05,102 - INFO - mycodo.controller_input_80d83ca4 - Activated in 7414.3 ms
2019-07-02 00:24:05,104 - DEBUG - mycodo.inputs.linux_command_80d83ca4 - Command being executed: /home/pi/moduser1/KRUN1py
2019-07-02 00:24:05,504 - DEBUG - mycodo.inputs.linux_command_80d83ca4 - Command returned: b'', Status: None, Error: 127
2019-07-02 00:24:05,505 - ERROR - mycodo.inputs.linux_command_80d83ca4 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''

SAM26K, 2019-07-03:
Tried a simple print("1.4") as a  python command and it returns null and throws non-numeric error (works ok if run in python 3  IDE).
echo("1,4")  in Linux works.

[42 more replies]

closed 2019-07-17 by SAM26K