kizniche/Mycodo #669: Mycodo DHT22 humidity readings

ralphknoops, 2019-07-03:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.5.10
 
#### Problem Description
Please list:
I'm using an DHT22 but i'm getting weird humidity data readings and errors.
It worked correct in previous versions, before 7.x.

The humidity constantly gives a 0% or 1% reading. Sometimes it gives a higher reading.

After I still had this issue, I did an upgrade from 7.5.3 to 7.5.10 today. But no luck yet.
 
### Errors
2019-07-03 19:09:49,403 - ERROR - mycodo.controller_input_b90028c8 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-03 19:09:51,689 - ERROR - mycodo.controller_input_5e076805 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-03 19:10:49,437 - ERROR - mycodo.controller_input_b90028c8 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

### Steps to Reproduce the issue:
Watch the Dashboard, Live or logs.

### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2019-07-03:
Can you confirm if pigpiod is running?

ralphknoops, 2019-07-03:
The temperature on the same DHT22 is working properly, this means pigpiod is running right?

kizniche, 2019-07-03:
Yes, probably. Do you have a new DHT22 you can use to test on a minimal circuit with short wires to the Pi?

ralphknoops, 2019-07-04:
If i'm correct I have a spare one. Need some time for this, hopefully today or tomorrow.

ralphknoops, 2019-07-05:
I tried a different DHT22 and it's giving back proper reading. So hardware related..
Thanks for the quick response

closed 2019-07-07 by kizniche