kizniche/Mycodo #670: TH10 w/ AM2301 reporting C as F

toposwope, 2019-07-09:
## Mycodo Issue Report:
  - Specific Mycodo Version:

Mycodo Version: 7.5.10 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 6333b0832b3d
Daemon Status: Running 
Daemon Process ID: 759 
Daemon RAM Usage: 68.756 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 53.084 MB 
Frontend Virtualenv: Yes
 
#### Problem Description

TH10 + AM2301 with Tasmota 6.6.0. Mycodo reports humidity correctly but displays incorrect data for Temperature. When set to read C and Do Not Convert it lists the C temp much lower than actual. When Input set to Convert to Fahrenheit, it lists the correct temperature in Celcius but displays it as Fahrenheit.

 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Add TH10 w/ AM2301 w/ Tasmota as input
 2. Live data lists incorrect C temperature
 3. Set input to Convert to Fahrenheit
4. Live data displays correct C but displayed as F
 
### Additional Notes
Images attached. Thanks for the great product!
[image]
[image]
[image]
[image]

kizniche, 2019-07-10:
This was an easy fix. Thanks for the bug report. The fix will be live in the next release.

toposwope, 2019-07-10:
Awesome thanks!

toposwope, 2019-07-10:
Hi, sorry but after upgrade no data at all being collected from the sensor...
[image]

kizniche, 2019-07-10:
I don't own this devoce/sensor so I cannot test myself. So please help me out with this. Line 95 of the input file is:

```
dict_data['StatusSNS']['TempUnit']
```

Can you verify is the unit is contained within the returned dictionary? The location of the unit is likely different from this or non-existent.

In the meantime, to get your input working, you can change this variable to ```"C"```, including the quotes, save the file, and restart the backend.

toposwope, 2019-07-11:
Hi, yes sure happy to help you out. Could you please be a bit more specific as to what I need to do exactly? 

Which specific file is the input file and what exactly do I change to "C"?

I do not see AM2301 listed in /home/pi/Mycodo/mycodo/inputs/, only AM2315 & AM2320

kizniche, 2019-07-11:
You can place the following line before the quoted code, above, to have the dictionary of the returned data printed in the daemon log. That can be accessed from the config menu.

```
self.logger.info("TEST: {}".format(dict_data))
```

Im on mobile, so being technical isn't very easy. I can explain anything with more detail later when I have access to my computer.

kizniche, 2019-07-11:
Replace ```dict_data['StatusSNS']['TempUnit']``` with ```"C"```

toposwope, 2019-07-11:
Thanks. Here is the log. The input is working fine as I can connect to it via IP and see the temp and humidity changing and everything was working fine apart from the conversion on the previous release.

2019-07-11 10:52:39,734 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - InputModule raised an exception when taking a reading: unorderable types: NoneType() >= int()
2019-07-11 10:52:54,261 - ERROR - mycodo.inputs.sensorutils - Conversion not found for C to C
2019-07-11 10:52:54,265 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - InputModule raised an exception when taking a reading: unorderable types: NoneType() >= int()
2019-07-11 10:52:54,268 - ERROR - mycodo.controller_input_330a56d5 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2019-07-11:
The log line doesn't seem to be inserted properly. I just got to a computer. Here is the entirety of the get_measurement() finction you should test with:

```python
    def get_measurement(self):
        self.return_dict = measurements_dict.copy()

        url = "http://{ip}/cm?cmnd=status%2010".format(ip=self.ip_address)
        r = requests.get(url)
        str_json = r.text

        dict_data = json.loads(str_json)
        self.logger.error("TEST: {}".format(dict_data))

        # Convert string to datetime object
        datetime_timestmp = datetime.datetime.strptime(
            dict_data['StatusSNS']['Time'], '%Y-%m-%dT%H:%M:%S')

        # Convert temperature to SI unit Celsius
        if self.is_enabled(0):
            temp_c = dict_data['StatusSNS']['AM2301']['Temperature']
            # temp_c = convert_from_x_to_y_unit(
            #     dict_data['StatusSNS']['TempUnit'],
            #     'C',
            #     dict_data['StatusSNS']['AM2301']['Temperature'])
            self.value_set(0, temp_c, timestamp=datetime_timestmp)

        if self.is_enabled(1):
            humidity = dict_data['StatusSNS']['AM2301']['Humidity']
            self.value_set(1, humidity, timestamp=datetime_timestmp)

        if (self.is_enabled(2) and
                self.is_enabled(0) and
                self.is_enabled(1)):
            dewpoint = calculate_dewpoint(
                self.value_get(0), self.value_get(1))
            self.value_set(2, dewpoint, timestamp=datetime_timestmp)

        if (self.is_enabled(3) and
                self.is_enabled(0) and
                self.is_enabled(1)):
            vpd = calculate_vapor_pressure_deficit(
                self.value_get(0), self.value_get(1))
            self.value_set(3, vpd, timestamp=datetime_timestmp)

        return self.return_dict
```

toposwope, 2019-07-11:
Added above to file, saved file, restarted backend. 

Then if I try to access the live data I get error:

Error 500: Internal Server Error

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1099, in page_live
    list_devices_adc=list_analog_to_digital_converters(),
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 57, in list_analog_to_digital_converters
    dict_inputs = parse_input_information()
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 128, in parse_input_information
    input_custom = load_module_from_file(full_path)
  File "/home/pi/Mycodo/mycodo/utils/inputs.py", line 70, in load_module_from_file
    spec.loader.exec_module(input_custom)
  File "<frozen importlib._bootstrap_external>", line 669, in exec_module
  File "<frozen importlib._bootstrap_external>", line 775, in get_code
  File "<frozen importlib._bootstrap_external>", line 735, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 79
    def get_measurement(self):
                             ^
IndentationError: unindent does not match any outer indentation level

**Log:**

Last 30 lines of /var/log/mycodo/mycodo.log:
  File "<frozen importlib._bootstrap_external>", line 735, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 79
    def get_measurement(self):
                             ^
IndentationError: unindent does not match any outer indentation level
2019-07-11 13:35:17,448 - ERROR - mycodo.daemon - Could not activate Input controller with ID 330a56d5-898d-445e-b3c7-11ddde39c2db: unindent does not match any outer indentation level (th1x_am2301.py, line 79)
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 573, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 95, in __init__
    self.dict_inputs = parse_input_information()
  File "/var/mycodo-root/mycodo/utils/inputs.py", line 128, in parse_input_information
    input_custom = load_module_from_file(full_path)
  File "/var/mycodo-root/mycodo/utils/inputs.py", line 70, in load_module_from_file
    spec.loader.exec_module(input_custom)
  File "<frozen importlib._bootstrap_external>", line 669, in exec_module
  File "<frozen importlib._bootstrap_external>", line 775, in get_code
  File "<frozen importlib._bootstrap_external>", line 735, in source_to_code
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 79
    def get_measurement(self):
                             ^
IndentationError: unindent does not match any outer indentation level
2019-07-11 13:35:17,467 - INFO - mycodo.daemon - All activated Input controllers started
2019-07-11 13:35:17,470 - INFO - mycodo.daemon - All activated Math controllers started
2019-07-11 13:35:17,473 - INFO - mycodo.daemon - All activated PID controllers started
2019-07-11 13:35:17,475 - INFO - mycodo.daemon - All activated LCD controllers started
2019-07-11 13:35:17,979 - INFO - mycodo.daemon - Mycodo daemon started in 14.261 seconds
2019-07-11 13:35:17,992 - INFO - mycodo.daemon - 40.58 MB RAM in use

toposwope, 2019-07-11:
Created a graph on the dashboard and can see data but still showing the C incor
[image]
rectly

kizniche, 2019-07-11:
The code doesn't seem to have been copied with the cirrect indentation and the Input didn't start. Here is the entire input module: [th1x_am2301.txt](https://github.com/kizniche/Mycodo/files/3380286/th1x_am2301.txt)

toposwope, 2019-07-11:
Great thanks, it is now showing correct info:

[image]

Error log:

2019-07-11 15:11:18,414 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'AM2301': {'Temperature': 20.2, 'Humidity': 56.1}, 'TempUnit': 'C', 'Time': '2019-07-11T06:11:18'}}
2019-07-11 15:11:33,891 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'AM2301': {'Temperature': 20.3, 'Humidity': 56.2}, 'TempUnit': 'C', 'Time': '2019-07-11T06:11:33'}}
2019-07-11 15:11:47,721 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'AM2301': {'Temperature': 20.3, 'Humidity': 56.1}, 'TempUnit': 'C', 'Time': '2019-07-11T06:11:47'}}
2019-07-11 15:12:03,075 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'AM2301': {'Temperature': 20.3, 'Humidity': 55.8}, 'TempUnit': 'C', 'Time': '2019-07-11T06:12:02'}}
2019-07-11 15:12:18,066 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'AM2301': {'Temperature': 20.3, 'Humidity': 56.0}, 'TempUnit': 'C', 'Time': '2019-07-11T06:12:17'}}
2019-07-11 15:14:40,996 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:14:40', 'TempUnit': 'C', 'AM2301': {'Humidity': 56.1, 'Temperature': 20.3}}}
2019-07-11 15:14:55,194 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:14:55', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:15:10,394 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:15:10', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:15:25,324 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:15:25', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.2}}}
2019-07-11 15:15:40,255 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:15:40', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.9, 'Temperature': 20.3}}}
2019-07-11 15:15:55,312 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:15:55', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:16:09,953 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:16:09', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.9, 'Temperature': 20.3}}}
2019-07-11 15:16:25,051 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:16:25', 'TempUnit': 'C', 'AM2301': {'Humidity': 56.0, 'Temperature': 20.3}}}
2019-07-11 15:16:40,184 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:16:40', 'TempUnit': 'C', 'AM2301': {'Humidity': 56.0, 'Temperature': 20.3}}}
2019-07-11 15:16:55,207 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:16:55', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.9, 'Temperature': 20.2}}}
2019-07-11 15:17:10,176 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:17:10', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.2}}}
2019-07-11 15:17:25,631 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:17:25', 'TempUnit': 'C', 'AM2301': {'Humidity': 56.0, 'Temperature': 20.3}}}
2019-07-11 15:17:40,058 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:17:39', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:17:55,453 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:17:55', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:18:10,034 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:18:09', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:18:25,260 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:18:25', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.6, 'Temperature': 20.3}}}
2019-07-11 15:18:40,035 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:18:39', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.7, 'Temperature': 20.3}}}
2019-07-11 15:18:55,250 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:18:55', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.8, 'Temperature': 20.3}}}
2019-07-11 15:19:09,995 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:19:09', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.6, 'Temperature': 20.4}}}
2019-07-11 15:19:25,087 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:19:25', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.6, 'Temperature': 20.3}}}
2019-07-11 15:19:40,029 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:19:39', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.6, 'Temperature': 20.3}}}
2019-07-11 15:19:55,083 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:19:55', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.6, 'Temperature': 20.3}}}
2019-07-11 15:20:10,335 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:20:10', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.6, 'Temperature': 20.3}}}
2019-07-11 15:20:24,912 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:20:24', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.7, 'Temperature': 20.3}}}
2019-07-11 15:20:40,624 - ERROR - mycodo.inputs.th1x_am2301_330a56d5 - TEST: {'StatusSNS': {'Time': '2019-07-11T06:20:40', 'TempUnit': 'C', 'AM2301': {'Humidity': 55.7, 'Temperature': 20.3}}}

kizniche, 2019-07-11:
Thanks. The returned dictionary has the structure I thought it did. I think I know where the issue is. At least you now have it working. I'll push a commit with the fix and make a new release tomorrow, then you can upgrade/test when you have time.

toposwope, 2019-07-11:
Great, will do, thanks so much!

kizniche, 2019-07-11:
v7.6.1 Released

SAM26K, 2019-07-12:
Looks like the TH16's stopped working after rev 7.6.0.  Went from 7.6.0 to 7.6.2 and now no data is received by mycodo from the TH16's.

toposwope, 2019-07-12:
Same here, no data after upgrade. 

Error from log:

2019-07-12 13:09:50,689 - ERROR - mycodo.inputs.sensorutils - Units are the same (C to C). Not performing conversion.
2019-07-12 13:09:50,694 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - InputModule raised an exception when taking a reading: argument of type 'float' is not iterable

kizniche, 2019-07-12:
> 2019-07-12 13:09:50,694 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - InputModule raised an exception when taking a reading: argument of type 'float' is not iterable

Thanks for the log! This seems to be the issue. Now to track down exactly where this is occurring.

kizniche, 2019-07-12:
Can you [Upgrade to Master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and report back what appears in the Daemon Log?

toposwope, 2019-07-12:
2019-07-12 13:37:52,283 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 13:38:07,552 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST: {'StatusSNS': {'Time': '2019-07-12T04:38:07', 'TempUnit': 'C', 'AM2301': {'Temperature': 19.6, 'Humidity': 48.8}}}
2019-07-12 13:38:07,568 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - InputModule raised an exception when taking a reading: argument of type 'float' is not iterable
2019-07-12 13:38:21,986 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST: {'StatusSNS': {'Time': '2019-07-12T04:38:22', 'TempUnit': 'C', 'AM2301': {'Temperature': 19.6, 'Humidity': 48.8}}}
2019-07-12 13:38:22,006 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - InputModule raised an exception when taking a reading: argument of type 'float' is not iterable
2019-07-12 13:38:37,200 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST: {'StatusSNS': {'Time': '2019-07-12T04:38:37', 'TempUnit': 'C', 'AM2301': {'Temperature': 19.6, 'Humidity': 48.8}}}
2019-07-12 13:38:37,216 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - InputModule raised an exception when taking a reading: argument of type 'float' is not iterable

kizniche, 2019-07-12:
Your log lines indicate you did not successfully upgrade to master. Did you follow the instructions in the link I provided?

toposwope, 2019-07-12:
Yes but I did not do this as I thought it would just put me back to the latest release:

Navigate to the upgrade page (Config -> Upgrade)
You can now upgrade Mycodo to the latest commit on the Github master branch.

I guess I should have. Will do now

kizniche, 2019-07-12:
The master branch is ahead of the latest release. I pushed code to master just prior to requesting you upgrade to it.

SAM26K, 2019-07-12:
Upgraded to master and TH16's still not working.  Tried to replace data input control to make sure right;    probe is selected and still no data.

Daemon (Error) log:

> 
> `2019-07-11 21:39:00,636 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:39:02,670 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:39:02,674 - ERROR - mycodo.controller_input_352ca283 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:39:14,011 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:39:15,653 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:39:17,896 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:39:29,050 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:39:29,078 - ERROR - mycodo.controller_input_1ab8f7c4 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:39:31,560 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:39:31,564 - ERROR - mycodo.controller_input_dc446b55 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:39:32,651 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:39:44,046 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:39:45,642 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:39:47,669 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:39:47,673 - ERROR - mycodo.controller_input_352ca283 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:39:59,098 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:40:00,667 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:40:02,714 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:40:14,018 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:40:14,032 - ERROR - mycodo.controller_input_1ab8f7c4 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:40:15,643 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:40:15,647 - ERROR - mycodo.controller_input_dc446b55 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:40:17,685 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:40:29,038 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:40:30,670 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:40:32,686 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00
> 2019-07-11 21:40:32,690 - ERROR - mycodo.controller_input_352ca283 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-11 21:40:44,229 - ERROR - mycodo.inputs.th1x_am2301_1ab8f7c4 - TEST00
> 2019-07-11 21:40:45,634 - ERROR - mycodo.inputs.th1x_am2301_dc446b55 - TEST00
> 2019-07-11 21:40:47,703 - ERROR - mycodo.inputs.th1x_am2301_352ca283 - TEST00`

toposwope, 2019-07-12:
Sorry about that. Ok now its done. Here is error log:

2019-07-12 14:45:16,156 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:45:16,163 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:45:31,131 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:45:46,149 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:46:01,621 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:46:01,638 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:46:16,257 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:46:31,273 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:46:46,081 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:46:46,088 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:47:01,226 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:47:16,183 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:47:31,244 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:47:31,268 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:47:46,069 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:48:01,298 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:48:16,072 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:48:16,080 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:48:31,891 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:48:46,074 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:49:01,294 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:49:01,312 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:49:16,111 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:49:31,440 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:49:46,749 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:49:46,756 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-12 14:50:01,289 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:50:16,388 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:50:31,327 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST00
2019-07-12 14:50:31,351 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

SAM26K, 2019-07-12:
So you are also confirming it doesn't work yet?  It wasn't clear from your last post.

SAM26K, 2019-07-12:
Just want to mention that I have 3 TH16's Ice been using  for awhile.  All of them use   Tasmota V6.4.0 with AM2301 probes.  All are configured for deg F and the mycodo controls were set to either C or F conversion.  Did not experience this original issue problem.  Setting the temp to C in the tasmota settings seems to be the problem.  It works if set to deg F with earlier versions of Mycodo.

toposwope, 2019-07-12:
Yes, confirming after upgrade to master still no data

kizniche, 2019-07-12:
For some reason the debug line I added is not yielding the exception traceback, which it should. Make sure you both have "Log Level: Debug" enabled for your input and paste the resulting Daemon Log output.

kizniche, 2019-07-12:
I just pushed a potential fix. Can you upgrade to master again and see if your inputs work?

SAM26K, 2019-07-12:
Still no data from the 4 channels but the bash controls that test for on//off state do work.
Daemon (Error) Log:

> `Last 30 lines of :
> 2019-07-12 14:58:49,386 - ERROR - mycodo.controller_input_6380985b - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 14:58:53,319 - ERROR - mycodo.controller_conditional_62a9b9a7 - Error evaluating conditional statement. Code and Traceback below.
> 2019-07-12 14:59:04,317 - ERROR - mycodo.controller_conditional_7271a899 - Error evaluating conditional statement. Code and Traceback below.
> 2019-07-12 14:59:04,393 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 14:59:19,585 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 14:59:31,819 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST00
> 2019-07-12 14:59:32,017 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST00
> 2019-07-12 14:59:34,333 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 14:59:34,336 - ERROR - mycodo.controller_input_6380985b - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 14:59:46,085 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST00
> 2019-07-12 14:59:46,625 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST00
> 2019-07-12 14:59:49,371 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 15:00:01,346 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST00
> 2019-07-12 15:00:01,348 - ERROR - mycodo.controller_input_9aa5162c - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 15:00:01,522 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST00
> 2019-07-12 15:00:01,525 - ERROR - mycodo.controller_input_946017ab - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 15:00:04,403 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 15:00:04,714 - ERROR - mycodo.controller_conditional_7271a899 - Error evaluating conditional statement. Code and Traceback below.
> 2019-07-12 15:00:17,198 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST00
> 2019-07-12 15:00:17,450 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST00
> 2019-07-12 15:00:19,683 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 15:00:19,685 - ERROR - mycodo.controller_input_6380985b - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 15:00:31,217 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST00
> 2019-07-12 15:00:31,575 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST00
> 2019-07-12 15:00:34,376 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00
> 2019-07-12 15:00:46,145 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST00
> 2019-07-12 15:00:46,147 - ERROR - mycodo.controller_input_9aa5162c - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 15:00:46,465 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST00
> 2019-07-12 15:00:46,466 - ERROR - mycodo.controller_input_946017ab - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 15:00:49,327 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST00`

kizniche, 2019-07-12:
> Still no data from the 4 channels but the bash controls that test for on//off state do work.

What does 4 channels mean? What bash controls that test for on/off? I don't have this device/sensor, and this doesn't make sense about what I know this device to do. To my knowledge, it returns a temperature and a humidity.

SAM26K, 2019-07-12:
1. Temp 2. Humidity 3. Dew pt 4. VPD.

Example of bash code that returns the on/off state: (commands from Tasmota docs)
```

> #!/bin/bash
> out=$(curl -s http://192.168.1.36/cm?cmnd=Power1)
> #echo $out
> if [[ $out == *"ON"* ]]; then
>   echo "1"
> else
>  if [[ $out == *"OFF"* ]]; then
>   echo "0"
>  fi
> fi
```

kizniche, 2019-07-12:
I just pushed a change to master that should show the traceback in the Daemon Log. Please upgrade to master and see of the log shows this traceback.

kizniche, 2019-07-12:
I see what you mean now. I'm on mobile, so this isn't easy. Hopefully this last change will help show what the issue is.

SAM26K, 2019-07-12:
Still no data.  Feel free to logon remotely and experiment.  Only a humidifier and Fan are hooked up and can be toggled.  Water  pump and valve solenoids are not plugged in:

Last 30 lines of Daenon(error) log :
`

> 2019-07-12 16:07:32,994 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST02: Traceback (most recent call last):
> 2019-07-12 16:07:33,705 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST01
> 2019-07-12 16:07:33,709 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST02: Traceback (most recent call last):
> 2019-07-12 16:07:44,063 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST01
> 2019-07-12 16:07:44,068 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST02: Traceback (most recent call last):
> 2019-07-12 16:07:44,069 - ERROR - mycodo.controller_input_946017ab - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 16:07:47,760 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST01
> 2019-07-12 16:07:47,762 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST02: Traceback (most recent call last):
> 2019-07-12 16:07:48,452 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST01
> 2019-07-12 16:07:48,454 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST02: Traceback (most recent call last):
> 2019-07-12 16:07:59,044 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST01
> 2019-07-12 16:07:59,047 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:02,639 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST01
> 2019-07-12 16:08:02,643 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:02,644 - ERROR - mycodo.controller_input_6380985b - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 16:08:03,474 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST01
> 2019-07-12 16:08:03,476 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:03,477 - ERROR - mycodo.controller_input_9aa5162c - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 16:08:14,264 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST01
> 2019-07-12 16:08:14,284 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:18,763 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST01
> 2019-07-12 16:08:18,800 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:19,847 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST01
> 2019-07-12 16:08:19,849 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:24,150 - ERROR - mycodo.controller_conditional_7271a899 - Error evaluating conditional statement. Code and Traceback below.
> 2019-07-12 16:08:29,115 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST01
> 2019-07-12 16:08:29,123 - ERROR - mycodo.inputs.th1x_am2301_946017ab - TEST02: Traceback (most recent call last):
> 2019-07-12 16:08:29,125 - ERROR - mycodo.controller_input_946017ab - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 16:08:32,659 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST01
> 2019-07-12 16:08:32,662 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST02: Traceback (most recent call last):`

SAM26K, 2019-07-12:
Last 30 lines of Daemon(all) log:

> `2019-07-12 16:29:47,758 - ERROR - mycodo.inputs.th1x_am2301_6380985b - TEST02: Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 121, in get_measurement
>     self.value_get(0), self.value_get(1))
>   File "/var/mycodo-root/mycodo/inputs/sensorutils.py", line 114, in calculate_vapor_pressure_deficit
>     svp = calculate_saturated_vapor_pressure(temp_c)
>   File "/var/mycodo-root/mycodo/inputs/sensorutils.py", line 100, in calculate_saturated_vapor_pressure
>     if None in temp_c:
> TypeError: argument of type 'float' is not iterable
> 
> 2019-07-12 16:29:47,759 - ERROR - mycodo.controller_input_6380985b - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2019-07-12 16:29:48,506 - DEBUG - mycodo.inputs.th1x_am2301_9aa5162c - Returned Data: {'StatusSNS': {'Time': '2019-07-13T00:29:48', 'AM2301': {'Temperature': 78.8, 'Humidity': 54.3}, 'TempUnit': 'F'}}
> 2019-07-12 16:29:48,546 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST01
> Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 121, in get_measurement
>     self.value_get(0), self.value_get(1))
>   File "/var/mycodo-root/mycodo/inputs/sensorutils.py", line 114, in calculate_vapor_pressure_deficit
>     svp = calculate_saturated_vapor_pressure(temp_c)
>   File "/var/mycodo-root/mycodo/inputs/sensorutils.py", line 100, in calculate_saturated_vapor_pressure
>     if None in temp_c:
> TypeError: argument of type 'float' is not iterable
> 2019-07-12 16:29:48,548 - ERROR - mycodo.inputs.th1x_am2301_9aa5162c - TEST02: Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/inputs/th1x_am2301.py", line 121, in get_measurement
>     self.value_get(0), self.value_get(1))
>   File "/var/mycodo-root/mycodo/inputs/sensorutils.py", line 114, in calculate_vapor_pressure_deficit
>     svp = calculate_saturated_vapor_pressure(temp_c)
>   File "/var/mycodo-root/mycodo/inputs/sensorutils.py", line 100, in calculate_saturated_vapor_pressure
>     if None in temp_c:
> TypeError: argument of type 'float' is not iterable
> 
> 2019-07-12 16:29:48,549 - ERROR - mycodo.controller_input_9aa5162c - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.`

kizniche, 2019-07-12:
It's extremely weird the tracebacks aren't being shown. I'm not going to be able to diagnose this unless I have access to test, and I won't be back to a computer until Sunday.

SAM26K, 2019-07-12:
That's too bad.  This issue is messing up the changes you made for me in the other open issue.
The original code worked fine if tasmota firmware is set to deg F.  Can; you (or me) restore the original code for the TH16/10's until you can fix this minor fluke?

toposwope, 2019-07-12:
Upgraded to master, no data and error log:

2019-07-13 09:40:31,935 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:40:31,944 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:40:46,480 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:40:46,490 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:40:46,493 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-13 09:41:02,391 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:41:02,401 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:41:17,027 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:41:17,037 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:41:31,942 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:41:32,541 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:41:32,546 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-13 09:41:47,351 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:41:47,361 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:42:01,884 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:42:02,018 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:42:16,613 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:42:16,632 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:42:16,645 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-13 09:42:32,618 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:42:32,627 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:42:46,654 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:42:46,663 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:43:02,215 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:43:02,232 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:43:02,236 - ERROR - mycodo.controller_input_7f584517 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-07-13 09:43:16,458 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:43:16,466 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):
2019-07-13 09:43:32,123 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST01
2019-07-13 09:43:32,181 - ERROR - mycodo.inputs.th1x_am2301_7f584517 - TEST02: Traceback (most recent call last):

kizniche, 2019-07-12:
I actually see i overlooked the traceback in your most recent log. I can see what's wrong and will push a fix in a moment to test.

kizniche, 2019-07-12:
It should finally be fixed now. Upgrade to master and give it a test.

SAM26K, 2019-07-12:
Looks like the tracebacks don't show up with clicking Daemon(Error) button.

kizniche, 2019-07-12:
Ah. That is why they weren't being copied. You should always show the whole Daemon log. Any of the other Daemon log buttons only show the lines with that specific word in it ("ERROR", "DEBUG", etc.). That gives me good reason to remove those buttons. They certainly caused a lot of headache today.

SAM26K, 2019-07-13:
Now not getting data from anything (live    page everything says "...no data..."

Last 60 lines of /var/log/mycodo/mycodo.log:
> `
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:12,669 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:22,421 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:26,650 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:27,986 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:30,278 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:30,511 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:31,869 - ERROR - mycodo.daemon - Could not find Output Controller
> 2019-07-12 17:24:32,122 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:33,075 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:34,083 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'
> 2019-07-12 17:24:41,305 - ERROR - mycodo.daemon - Could not turn output off: 'NoneType' object has no attribute 'output_on_off'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 902, in output_off
>     return_status = self.controller['Output'].output_on_off(
> AttributeError: 'NoneType' object has no attribute 'output_on_off'
> 2019-07-12 17:24:42,824 - ERROR - mycodo.daemon - Could not query output state: 'NoneType' object has no attribute 'output_state'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 979, in output_state
>     return self.controller['Output'].output_state(output_id)
> AttributeError: 'NoneType' object has no attribute 'output_state'`

SAM26K, 2019-07-13:
Correction.  Guess I didn't give it enough time.  Getting data now including TH16's.

toposwope, 2019-07-13:
Working perfectly now for me as well after latest upgrade to master

SAM26K, 2019-07-17:
> Working perfectly now for me as well after latest upgrade to master

Are you ok to close this issue? Works for me with 7.6.3 release.
Reply with the 'close' button if ok.

SAM26K, 2019-07-17:
BTW I like the extra error buttons but now you know to look for the button label when troubleshooting.

closed 2019-07-17 by toposwope