kizniche/Mycodo #673: Issues with multiple MQTT inputs

sophanox, 2019-07-29:
Hey there,

I'm running version 7.6.3, but I'm having issues with multiple MQTT inputs. It appears that after a second MQTT input is enabled, something breaks. Both inputs work individually. 

I'm not sure if it's a bug in the feature, or something wrong with my configuration?

Here is a relevant log:

```2019-07-29 18:21:23,644 - INFO - mycodo.controller_input_77549b18 - Activated in 274.7 ms
2019-07-29 18:21:23,645 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Connecting MQTT callback functions
2019-07-29 18:21:23,646 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - MQTT callback functions connected
2019-07-29 18:21:23,649 - INFO - mycodo.inputs.mqtt_paho_77549b18 - Connected to localhost as mycodo_mqtt_client
2019-07-29 18:21:23,650 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Subscribed to MQTT channel 'air-temp-humid/Temperature-Humidity/Temperature'
2019-07-29 18:21:23,651 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Subscribed to MQTT channel 'air-temp-humid/Temperature-Humidity/Humidity'
2019-07-29 18:21:23,653 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Connected to 'None', rc: 0
2019-07-29 18:21:23,654 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Subscribing to mqtt topic: None, 1, (0,)
2019-07-29 18:21:23,654 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Subscribing to mqtt topic: None, 2, (0,)
2019-07-29 18:21:25,860 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Message received: Channel: air-temp-humid/Temperature-Humidity/Temperature, Value: 25.60
2019-07-29 18:21:25,868 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Adding measurement to influxdb: {0: {'timestamp_utc': datetime.datetime(2019, 7, 29, 17, 21, 25, 860356), 'unit': 'C', 'value': 25.6, 'measurement': 'temperature'}}
2019-07-29 18:21:25,966 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Message received: Channel: air-temp-humid/Temperature-Humidity/Humidity, Value: 49.31
2019-07-29 18:21:25,973 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Adding measurement to influxdb: {1: {'timestamp_utc': datetime.datetime(2019, 7, 29, 17, 21, 25, 966261), 'unit': 'percent', 'value': 49.31, 'measurement': 'humidity'}}
2019-07-29 18:21:26,317 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Client created with ID mycodo_mqtt_client
2019-07-29 18:21:26,320 - INFO - mycodo.controller_input_aac9beb8 - Activated in 284.1 ms
2019-07-29 18:21:26,321 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Connecting MQTT callback functions
2019-07-29 18:21:26,322 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - MQTT callback functions connected
2019-07-29 18:21:26,326 - INFO - mycodo.inputs.mqtt_paho_aac9beb8 - Connected to localhost as mycodo_mqtt_client
2019-07-29 18:21:26,326 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Disconnected result code 1
2019-07-29 18:21:26,328 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Subscribed to MQTT channel 'Substrate-moisture'
2019-07-29 18:21:26,329 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Connected to 'None', rc: 0
2019-07-29 18:21:26,330 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Subscribing to mqtt topic: None, 1, (0,)
2019-07-29 18:21:27,330 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Connected to 'None', rc: 0
2019-07-29 18:21:27,331 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Disconnected result code 1
2019-07-29 18:21:28,337 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Connected to 'None', rc: 0
2019-07-29 18:21:28,337 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Disconnected result code 1
2019-07-29 18:21:29,341 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Connected to 'None', rc: 0
2019-07-29 18:21:29,342 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Disconnected result code 1
2019-07-29 18:21:30,346 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Disconnected result code 1
2019-07-29 18:21:30,347 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Connected to 'None', rc: 0
2019-07-29 18:21:31,352 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Connected to 'None', rc: 0
2019-07-29 18:21:31,353 - DEBUG - mycodo.inputs.mqtt_paho_aac9beb8 - Disconnected result code 1
2019-07-29 18:21:32,363 - DEBUG - mycodo.inputs.mqtt_paho_77549b18 - Disconnected result code 1
```
Those last lines then just continue indefinitely. Only deactivating both inputs, then reactivating fixes the issue.

Here is a pic of the settings for what it's worth:

[image]

Thanks for looking!

kizniche, 2019-07-29:
Can you describe what you mean by "breaks"?

kizniche, 2019-07-29:
I was able to replicate your issue. I'll investigate a solution.

kizniche, 2019-07-29:
It appears you're using the same client ID. Change your client IDs to unique names and see if that works. I actually wasn't able to replicate your issue when using unique client IDs.

sophanox, 2019-07-30:
That was it! Thanks very much

closed 2019-07-30 by sophanox