kizniche/Mycodo #674: Remote Admin - Unable to add host

Ryobidelacroix, 2019-07-31:
## Mycodo Issue Report:
  - Specific Mycodo Version: **7.6.3**
 
#### Problem Description

Unable to add host in 'Remote Admin' Page. 

Attempting to add host under my current Admin account receives this error: 

> "Error: Unable to authenticate with user and password."

 This is the same error if the password has been entered incorrectly, however I have confirmed the password is correct. 

I have created a new 'Admin' user to check if this was just my normal admin account, however when attempting to add this new admin user as a host, a new error appears:
> "Error: Error: Remote Host Add: write() argument must be str, not None"

### Errors

Alerts, as above: 
> "Error: Unable to authenticate with user and password."
> "Error: Error: Remote Host Add: write() argument must be str, not None"

### Steps to Reproduce the issue:
* I have created a new 'Admin' user to check if this was just my normal admin account, however issue remains. 
* My domain address has been entered, Admin username, then password. Error messages show after hitting add host. 
* Pi rebooted
* Ensured port forwarding was functional
 
### Additional Notes
* WebAccess Log:

Full IP and usr/pssw info redacted 

`192.168.1.4 - - [31/Jul/2019:20:10:12 +0930] "GET /time HTTP/1.1" 200 14 "https://192.168.1.8/logview" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_5) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/12.1.1 Safari/605.1.15"
93.**.2*9.** - - [31/Jul/2019:20:10:19 +0930] "GET /newremote/?user=R***&passw=******* HTTP/1.1" 200 105 "-" "python-requests/2.22.0"`

* Web Error Log:

`2019/07/31 19:50:14 [alert] 531#531: worker process 532 exited on signal 9
2019/07/31 19:50:50 [crit] 533#533: *1 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /last/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/c0c2f920-b641-4f80-ae76-7881e07ec96d/120 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/last/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/c0c2f920-b641-4f80-ae76-7881e07ec96d/120", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"
2019/07/31 19:50:55 [crit] 535#535: *6 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /last/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/c0c2f920-b641-4f80-ae76-7881e07ec96d/120 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/last/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/c0c2f920-b641-4f80-ae76-7881e07ec96d/120", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"
2019/07/31 19:50:55 [crit] 534#534: *3 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /dashboard HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/dashboard", host: "192.168.1.8", referrer: "https://192.168.1.8/remote/setup"
2019/07/31 19:50:55 [crit] 536#536: *4 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /media/com_demo/graphics/loader.white.gif HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/media/com_demo/graphics/loader.white.gif", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"
2019/07/31 19:50:56 [crit] 534#534: *3 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /past/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/cf5d0551-6898-4da2-9a9f-64ced0c9b585/72000 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/past/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/cf5d0551-6898-4da2-9a9f-64ced0c9b585/72000", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"
2019/07/31 19:50:56 [crit] 536#536: *5 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /time HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/time", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"
2019/07/31 19:50:56 [crit] 536#536: *7 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /last/ac0512d5-e4e5-4cca-97fb-c84de57b5854/input/219576d3-5911-4186-af48-c56ce38a3a3e/3600 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/last/ac0512d5-e4e5-4cca-97fb-c84de57b5854/input/219576d3-5911-4186-af48-c56ce38a3a3e/3600", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"
2019/07/31 19:50:56 [crit] 536#536: *8 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.4, server: , request: "GET /past/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/c0c2f920-b641-4f80-ae76-7881e07ec96d/72000 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/past/ea4ce218-3aff-4848-bc34-5beeeb8f25c2/input/c0c2f920-b641-4f80-ae76-7881e07ec96d/72000", host: "192.168.1.8", referrer: "https://192.168.1.8/dashboard"`

:)

kizniche, 2019-08-17:
Sorry for the delayed response. The Remote Admin interface is currently experimental and no guaranteed to work. It's being worked on and is merely available as a demo. I have had other parts of the system needing attention, so I haven't addressed any issues with this part of the system in a while.

Ryobidelacroix, 2019-08-21:
Not a problem at all! Feel free to close this issue if you need.

closed 2019-08-25 by Ryobidelacroix