kizniche/Mycodo #675: MH-Z19 sensor not returning readings in Mycodo

EddNess, 2019-08-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.6.3
 
#### Problem Description

Updates were done including pip.
An MH-Z19 sensor was set up using this tek:
https://github.com/UedaTakeyuki/mh-z19
It gave readings in the pi console using python 2 and python 3 commands.
Its being run on a Pi 3 B+, so the UART device was changed to /dev/serial0, no readings. I've tried /dev/ttyS0, still nothing. I've gone back to /dev/ttyAMA0, but still no readings.
Console has been disabled with no change.

Any ideas?

If there is a better / easier CO2 sensor that you know of please let me know, and I'll quite happily switch. 

Many thanks in advance

Elliott

kizniche, 2019-08-03:
To use the MH-Z19 with Mycodo in its current state, you'll need to either use a Serial-to-USB board or add an Input module that has code that can work with the type of connection you made to the Pi. The solution that works with your current hardware setup is to add a custom Input using the code you have tested to work. I'm leaving for vacation today with very limited connectivity the next week, so I may not be of direct help with this. There are example custom Input modules, documentation in the manual, and the ability to copy the current MH-Z19 Input module and modify it to fit your needs with code that works. Let me know your skill level and what you need help with.

EddNess, 2019-08-03:
Hi
Thanks for the prompt reply.
Unfortunately I'm nearly a complete Noob, but I'll have a look at what you have suggested and give it a go!
This might take some time, so enjoy your vacation.
If I'm completely stuck I'll just have to wait a bit. Thing usually occur to me when I'm away from the immediate issue anyway.
Elliott

kizniche, 2019-08-04:
I'll see about creating a new input you can test with your current wired connection.

kizniche, 2019-08-04:
Create a new file named mh_z19_uart.py and add the following code as the contents:

```python
# coding=utf-8
import time

from flask_babel import lazy_gettext

from mycodo.inputs.base_input import AbstractInput
from mycodo.inputs.sensorutils import is_device

measurements_dict = {
    0: {
        'measurement': 'co2',
        'unit': 'ppm'
    }
}

INPUT_INFORMATION = {
    'input_name_unique': 'MH_Z19_UART',
    'input_manufacturer': 'Winsen',
    'input_name': 'MH-Z19 (UART)',
    'measurements_name': 'CO2',
    'measurements_dict': measurements_dict,

    'options_enabled': [
        'pre_output',
        'log_level_debug'
    ],
    'options_disabled': ['interface'],

    'dependencies_module': [
        ('pip-pypi', 'mh_z19', 'mh_z19')
    ],

    'interfaces': ['UART']
}

class InputModule(AbstractInput):
    """ A sensor support class that monitors the MH-Z19's CO2 concentration """

    def __init__(self, input_dev, testing=False):
        super(InputModule, self).__init__(input_dev, testing=testing, name=__name__)

        if not testing:
            import mh_z19
            self.sensor = mh_z19

    def get_measurement(self):
        """ Gets the MH-Z19's CO2 concentration in ppmv via UART"""
        self.return_dict = measurements_dict.copy()
        co2 = None
        self.value_set(0, self.sensor.read())
        return self.return_dict
```

Save the file. Go to Config -> Configure -> Inputs, then select the file you just created and click Import Input Module. If it says successfully added, go to the Setup -> Data page and search for the "MH-Z19 (UART)" input and add it. Activate it and see if it successfully receives CO2 measurements on the Live page.

EddNess, 2019-08-05:
Thanks a lot
Where is the old version stored? I'd like to add the code to turn off the baseline correction this sensor has. It should be a copy & paste after a bit of study.
Absolutely no rush as I'm just about to head off to a ten hour shift at work.
Elliott

kizniche, 2019-08-05:
I'm not sure what you mean by old version. I created that Input from the repository you provided. I on vacation and going out of cell reception for the next 6 days. Give the instructions I gave a shot.

kizniche, 2019-08-05:
Looking at the code of the repository you linked, there's no real difference from the built-in Input. It's likely you are no using the correct serial device. Can you verify which device is being created for your sensor?

EddNess, 2019-08-06:
Apologies for not replying earlier, long shifts at work.

I have given the code you kindly wrote a shot, but with no readings I'm afraid.

I was worried about the automatic baseline calibration this sensor has which is in the version included with Mycodo, but not in the test code that has just been tried. This sensor needs to be run outside for a bit to re-calibrate anyway as it took me ages to get it working in the pi's console.

I'm not sure what you mean by verifying the serial device being used, but it is connected by Ground to ground, Vcc to 5v+, RX to TX and TX to RX via the GPIO (in the corner furthest from the CPU) as shown in the link mentioned earlier, although I am using a breakout board. Maybe there is a command I'm not aware of?

I have looked at your earlier posts and ordered a serial to usb board as this might be the easiest solution. The one I feel best for the build has a long delivery time against it so no need to rush about this.

Please enjoy your vacation

Theoi-Meteoroi, 2019-08-15:
Looking at this one.  I got a couple of days to play with this stuff.  
I wouldn't worry too much about the ABC function until you get at least a good serial stream of data, reliably.  I had intentions of creating a github reference for this sensor since I invested a good deal of time chasing threads to figure out what Winsen was doing here.  I've exchanged emails with them, but never seem to get to anyone but sales.   The most interesting stuff was in Russian, IIRC.   This is a good sensor however it does seem to require frequent re-calibration due to the intended function ( human living space ventilation applications ) so I tend to use the MHz-19 model for lab experiments and long term deployments.  It is the same probe used in rather expensive industrial handheld probes since it has temperature and humidity compensation.

So, any luck with the USB adapter?  Those should be able to supply a regulated 5V of at least 100mA that can usually be enough for this sensor.  Not so much with the MHz-19 as it will pull about that much (or more) on every sample time.   I've had to tack large (~47uF) tantalum capacitors on the ones I've tried just to prevent voltage sags that reset the sensor.

EddNess, 2019-08-15:
Thanks for looking at this.
This particular one needs a bit of time outside with its ABC on as its giving readings over 1500 ppm, when my other equipment gives readings of around 800ppm. That's using the Ueda Takeyuki method of course. To get around this in the long-term, I intend to have a pi set-up for the gourmet mushrooms and a more portable one that can be used for outside baseline calibration. Not ideal, but an answer.
I'm still awaiting the USB adapter. I've ordered two as the one that fits the overall build best has to come half-way around the world. I've been given a delivery date of the 31st August, so its a patience game at the moment.
How have you been connecting the capacitors, across the 5v power supply?
I'll be sure to keep everyone updated with the results.

EddNess, 2019-08-22:
Hi Guys
Good news. The USB to serial adapter board arrived today. It has been connected and using /dev/ttyUSB0 Mycodo is reading it with the original coding.
It due to be dry tomorrow, so I'll run the sensor outside for the day to get the ABC set using the GPIO settings from Ueda Takeyuki in the hope of getting calibrated again.
Then its connect the relays and get my head around PID controllers ;)
Thank you for your help and patience.
Elliott

closed 2019-10-03 by kizniche