kizniche/Mycodo #676: SHT10 Not Working

berserktron3k, 2019-08-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.6.3
  - Hardware: Raspberry Pi 3 Model B Rev 1.2
#### Problem Description
Please list:
SHT10 does not obtain measurements.
When adding the SHT1x/7x, there are no error messages.
When specifying the 'Pin (GPIO)' (in my case, GPIO20), there is no error. 
When specifying 'Clock Pin' (in my case, GPIO21), there is no error.
When I save the Input device after making changes, there is no error.
 
### Errors
  - When trying to active the Input device, I get this as a pop-up: Error: Could not activate Input controller with ID 612853b7-06f1-4fae-993f-882fa25aef41: 'Input' object has no attribute 'clock_pin'. Also, this shows up in the Daemon log: ERROR - mycodo.daemon - Could not activate Input controller with ID 612853b7-06f1-4fae-993f-882fa25aef41: 'Input' object has no attribute 'clock_pin'
  - When I try to Deactivate the device, I get this as a pop-up: Error: Input controller with ID 612853b7-06f1-4fae-993f-882fa25aef41 not found.

berserktron3k, 2019-08-03:
After removing and readding, the same symptoms exist, and this is captured from the Daemon (ALL) log:
2019-08-03 17:14:42,536 - ERROR - mycodo.daemon - Could not activate Input controller with ID 835110b7-07be-4483-9a32-0ba409eb0be7: 'Input' object has no attribute 'clock_pin'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 573, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 182, in __init__
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/sht1x_7x.py", line 79, in __init__
    self.clock_pin = input_dev.clock_pin
AttributeError: 'Input' object has no attribute 'clock_pin'

kizniche, 2019-08-04:
I pushed a fix with commit https://github.com/kizniche/Mycodo/commit/c2ee99850bd2c233543ee6eac3583c270466b37c
If you want to manually make those changes, it should work. Let me know if you do.

berserktron3k, 2019-08-04:
> I pushed a fix with commit c2ee998

Thanks! I’ll tale a look when i get home.

berserktron3k, 2019-08-06:
That seems to have worked! Thanks for the quick fix!

Not sure if this is related, though, but definitely strange. I don't think this is a problem with the code, but hoping you can maybe point me in the right direction.

My SHT-10 started off reading normal. Right in the expected range of both temp and humidity, however within 24 hours, it's reading over double what it was. For example, it should be reading right around 27C, and it's currently at 64C. Additionally, humidity (which should be in the high 90's) is reading at 134%. Any ideas? Googling hasn't helped. Tried deactivating and reactivating the sensor, tried switching voltage and timing.

kizniche, 2019-08-17:
You may have a bad sensor. Do you have another you can replace with to test? I haven't looked at the [sht-sensor module](https://pypi.org/project/sht-sensor/) code, but I can say it's not the code in Mycodo doing it, as it merely takes the returned values from that module.

kizniche, 2019-08-17:
Whoops, it looks like I'm the maintainer of the [sht-sensor module](https://pypi.org/project/sht-sensor/). That's embarrassing :wink

I'll check it out and see if I can spot anything that might cause that.

kizniche, 2019-08-17:
Can you test [pi-sht1x](https://github.com/drohm/pi-sht1x) and see if you get the same results?

berserktron3k, 2019-08-17:
Well, now what's even stranger is it seems to have switched back to normal(ish). See attached screenshot. 
[image]
As you can see in the image, it was reading in the 140+ F and nearly the same for humidity. Then for some reason it dropped to the 70's F. I can assure you there was no such temperature extremes. Also, it read extremely high regardless of C or conversion to F.

Do you think a loose connection to the clock pin could cause this?

Switching voltages doesn't seem to make a difference either.

In trying pi-sht1x, it is reading normal, however since Mycodo is too, I'm not sure that's an accurate test. I'll keep my eye on it.

closed 2019-10-03 by kizniche