kizniche/Mycodo #677: USB Camera not working.

berserktron3k, 2019-08-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.6.3
  - Hardware version: Raspberry Pi 3 Model B Rev 1.2
 
#### Problem Description
Please list:
When trying to capture a photo, Mycodo acts as if it is taking a photo, but then does not display it. In browsing the file system, there are no photos saved. This is a known good (tested) USB camera, and taking a photo from the command prompt seems to work properly. Additionally, a piCamera has the same issue.
 
### Errors
No errors in the log.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Connect a USB camera
 2. Configure it in Mycodo
 3. Try to take a picture
 
### Additional Notes
This was occurring before and after a clean rebuild of both the OS and the Mycodo app.

kizniche, 2019-08-04:
Was this with the new Debian version, Raspbian Buster? What was the name of the software you used that was working?

berserktron3k, 2019-08-04:
It is now Buster, however it was the version immediately before as well.

I don’t recall the command I ran, but I believe it was the same as what
Mycodo calls, just from the command line. Also, the USB camera works fine
on a Windows PC.

kizniche, 2019-08-17:
I just got my USB camera back to test with, so I should have an update on this soon.

kizniche, 2019-10-07:
Got around to finding the cause of this issue. It will be live in the 7.8.0 release.

commit 2adc882a (2019-10-07): Fix inability to capture photos with USB camera (fixes #677)
closed 2019-10-07 by kizniche