kizniche/Mycodo #678: 7.6.3 Test Email Not Sending

wllyng, 2019-08-06:
Mycodo Issue Report: 7.6.3

Problem Description:
While using the test email alert system I noticed that none of the emails showed up in my inbox despite saying that the email was sent successfully. I have filled in all of the SMTP account information and have enabled the POP and IMAP stuff on the gmail account.

kizniche, 2019-08-07:
For Gmail, you must also enable "less secure apps", https://support.google.com/accounts/answer/6010255

wllyng, 2019-08-07:
Thanks it works now

closed 2019-08-07 by wllyng

robocode-LAB, 2023-11-08:
I know this is an old thread, but it looks like Gmail will not longer allow less secure apps since 2022. Is there a workaround for getting email notifications?

[image]

kizniche, 2023-11-08:
Gmail isn't the only email host.