kizniche/Mycodo #682: Backend shutdown after bad data from linux command input read

BobDDole, 2019-08-30:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 7.6.3
#### Problem Description
Please list:
Backend Daemon stops randomly due to "bad reads"

  - what were you trying to do
Linux command/Bash command from input(s)
curl -s 'http://10.10.0.88/cm?cmnd=%20Status%2011' | jq -r '.StatusSTS' | jq -r '.POWER1'| grep -c ON

AND/OR

curl -s 'http://10.10.0.88/cm?cmnd=%20Status%2011' | jq -r '.StatusSTS' | jq -r '.POWER2'| grep -c ON
  - specific setup details that are involved
Parsing data from a tasmota sonoff module to get it's power status 
### Errors
  - List any errors you encountered.
Backend just stops after getting no data from tasmota module or getting corrupt data from it 

  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

2019-08-30 15:45:07,288 - ERROR - mycodo.inputs.linux_command_c419a78e - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-08-30 15:45:07,292 - ERROR - mycodo.inputs.linux_command_f3048a36 - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'isalpha'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Create a linux/bash command input and use the command string or strings listed above
 2. Since you don't have a device on the network on that address nothing more needed just wait for the above error or errors and Daemon shut down.
(I'm polling same sonoff/tasmota  device but two power states from it)
  
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?
Seems like 7.6.3 wants to error and shut down on corrupt values from a linux/bash  command call whereas the older 7.4 would ignore errors and continue on it's way.
This is completely unusable asis since it kills the server on any "bad/missing" data.

kizniche, 2019-08-30:
Have you tried putting your commands in a bash script and executing the script with Mycodo?

kizniche, 2019-08-30:
> Backend Daemon stops randomly due to "bad reads"

The two exceptions you pasted shouldn't stop the daemon. In fact, those exceptions are handled by the Input module and are merely a warnings. Do you have log lines demonstrating a deamon malfunction?

kizniche, 2019-08-30:
I've added more concise exception-handling, but I'm still skeptical this is causing a daemon crash, or if the daemon is crashing, this is the cause. Also in the commit moments ago, I changed the log line from error to debug, so you'll need to set Log Level Debug in the Input settings to see them. That should reduce the clutter in the daemon log.

BobDDole, 2019-08-30:
Not sure what is happening but with those two linux commands deactivated I have had no daemon crashes.
Here is a cut from my log:

2019-08-30 15:44:28,925 - INFO - mycodo.daemon - Mycodo daemon v7.6.3 starting
2019-08-30 15:44:29,668 - INFO - mycodo.daemon - Anonymous statistics enabled
2019-08-30 15:44:29,711 - INFO - mycodo.daemon - Starting rpyc server
2019-08-30 15:44:30,698 - INFO - mycodo.controller_output - Output controller activated in 272.4 ms
2019-08-30 15:44:31,200 - INFO - mycodo.daemon - Output controller fully started
2019-08-30 15:44:31,201 - INFO - mycodo.daemon - All activated Conditional controllers started
2019-08-30 15:44:31,202 - INFO - mycodo.daemon - All activated Trigger controllers started
2019-08-30 15:44:32,274 - INFO - mycodo.controller_input_d9d5c81e - Activated in 890.1 ms
2019-08-30 15:44:33,629 - INFO - mycodo.controller_input_1b96b61f - Activated in 1168.2 ms
2019-08-30 15:44:34,524 - INFO - mycodo.controller_input_f3048a36 - Activated in 706.7 ms
2019-08-30 15:44:35,650 - INFO - mycodo.controller_input_c419a78e - Activated in 704.3 ms
2019-08-30 15:44:35,652 - INFO - mycodo.daemon - All activated Input controllers started
2019-08-30 15:44:35,653 - INFO - mycodo.daemon - All activated Math controllers started
2019-08-30 15:44:35,664 - INFO - mycodo.daemon - All activated PID controllers started
2019-08-30 15:44:35,666 - INFO - mycodo.daemon - All activated LCD controllers started
2019-08-30 15:44:36,168 - INFO - mycodo.daemon - Mycodo daemon started in 7.242 seconds
2019-08-30 15:44:36,173 - INFO - mycodo.daemon - 48.62 MB RAM in use
2019-08-30 15:45:07,288 - ERROR - mycodo.inputs.linux_command_c419a78e - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-08-30 15:45:07,292 - ERROR - mycodo.inputs.linux_command_f3048a36 - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'isalpha'
2019-08-30 15:45:07,294 - INFO - mycodo.daemon - All LCD controllers stopped
2019-08-30 15:45:07,295 - INFO - mycodo.daemon - All PID controllers stopped
2019-08-30 15:45:07,295 - INFO - mycodo.daemon - All Math controllers stopped
2019-08-30 15:45:07,342 - INFO - mycodo.controller_input_1b96b61f - Deactivated in 48.0 ms
2019-08-30 15:45:07,379 - INFO - mycodo.controller_input_d9d5c81e - Deactivated in 85.3 ms
2019-08-30 15:45:07,391 - INFO - mycodo.controller_input_c419a78e - Deactivated in 97.4 ms
2019-08-30 15:45:07,394 - INFO - mycodo.controller_input_f3048a36 - Deactivated in 101.1 ms
2019-08-30 15:45:07,396 - INFO - mycodo.daemon - All Input controllers stopped
2019-08-30 15:45:07,396 - INFO - mycodo.daemon - All Trigger controllers stopped
2019-08-30 15:45:07,397 - INFO - mycodo.daemon - All Conditional controllers stopped
[image]

This is how I have the command configured. If it helps any.

kizniche, 2019-08-30:
Have you tried moving the command into a bash script and making the command the execution of the script?

Try making the edits from this commit, restart the daemon, and see if it fixes the issue: https://github.com/kizniche/Mycodo/commit/2340fef177db998e351fc5690027f1d6fb6c32af

BobDDole, 2019-08-30:
Just tried putting it into a bash script and executing via script call. 
Same result.

kizniche, 2019-08-31:
Have you tried both recommendations?

BobDDole, 2019-08-31:
Being a noob to commit updates and such how does one pull the changes from github?

Sorry for the noobness.

BobDDole, 2019-08-31:
Yes, I made the commit changes and all I get is a 503 Bad Gateway error. I can't get the web interface up until I reverse the changes.

kizniche, 2019-09-03:
An error would indicate you made a typo in the manual edit of the files. Make sure you use only spaces (no tabs).

BobDDole, 2019-09-07:
Yes it was the "tabs" that did me in. 
I did the commits and now have no 503 error anymore but I have the same shutdown issues\
here's the latest log:

2019-09-07 04:33:59,274 - DEBUG - mycodo.inputs.linux_command_727d9bd8 - Command being executed: curl -s 'http://10.10.0.53/cm?cmnd=%20Status%2011' | jq -r '.StatusSTS' | jq -r '.POWER1'| grep -c ON
2019-09-07 04:34:59,318 - DEBUG - mycodo.inputs.linux_command_727d9bd8 - Command returned: None, Status: None, Error: None
2019-09-07 04:34:59,319 - DEBUG - mycodo.inputs.linux_command_727d9bd8 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'None'
2019-09-07 04:34:59,324 - INFO - mycodo.daemon - All LCD controllers stopped
2019-09-07 04:34:59,325 - INFO - mycodo.daemon - All PID controllers stopped
2019-09-07 04:34:59,325 - INFO - mycodo.daemon - All Math controllers stopped
2019-09-07 04:34:59,362 - INFO - mycodo.controller_input_4e679a49 - Deactivated in 37.9 ms
2019-09-07 04:34:59,370 - INFO - mycodo.controller_input_95510879 - Deactivated in 45.8 ms
2019-09-07 04:34:59,420 - INFO - mycodo.controller_input_727d9bd8 - Deactivated in 96.1 ms
2019-09-07 04:34:59,421 - INFO - mycodo.daemon - All Input controllers stopped
2019-09-07 04:34:59,422 - INFO - mycodo.daemon - All Trigger controllers stopped
2019-09-07 04:34:59,422 - INFO - mycodo.daemon - All Conditional controllers stopped
2019-09-07 04:34:59,450 - INFO - mycodo.controller_output - Output controller deactivated in 26.9 ms

BobDDole, 2019-09-15:
Ok so using the commits you made and modifying my curl call some seems to have resolved the problem:
curl -s -m 10 'http://10.10.0.53/cm?cmnd=%20Status%2011' | jq -r '.StatusSTS' | jq -r '.POWER1'| grep -c ON
My routine is ran every 15 seconds but the default timeout for curl is 30 seconds so I added the -m 10 to make it 10 second timeout thinking possibly the curl commands were overlapping each other and causing the 'NONE' value to return.
Seems this might be the case since making this change I have not had a single backend shutdown.

Thanks for the help, Appreciate it!

kizniche, 2019-09-15:
Good to hear! It was likely the timeout because all I did was prevent the exception from stopping the whole Input. It seems there's something else with curl that caused the whole daemon to stop.

BobDDole, 2019-09-24:
Seems so, going to convert these to a single python input now that you've implemented it (thanks for that as well) Now off to learn enough python to do that. LOL

closed 2019-09-24 by BobDDole

kizniche, 2019-09-24:
You'll probably want to check out the Python module "urllib"