kizniche/Mycodo #683: Daemon Stops Running When SPI Enabled

smatthews95, 2019-09-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 
 
#### Problem Description
I am trying to add a Waveshare  High Precision AD/DA (ADS1256) board for adding some analog sensors. After adding it in the output tab, an unmet dependencies warning popped up and I was prompted to to install the missing dependencies. I then ran "raspi-config" to enable SPI. When I then activated that input on the data, the daemon immediately stopped running. I tried rebooting and the daemon was still not running. I ran "raspi-config" again to disable SPI and the daemon started running again. When I repeated the process again, activating the ADS1256 on the data page caused the daemon to stop running.
 
### Errors
"Error: Error: invalid message type: None"

See Traceback below:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 911, in page_info
    ram_use_daemon = control.ram_use()
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 150, in ram_use
    return self.rpyc_client.root.ram_use()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 485, in root
    self._remote_root = self.sync_request(consts.HANDLE_GETROOT)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 455, in sync_request
    return self.async_request(handler, *args, timeout=timeout).value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/rpyc/core/async_.py", line 100, in value
    self.wait()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/rpyc/core/async_.py", line 47, in wait
    self._conn.serve(self._ttl)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 384, in serve
    self._dispatch(data)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 358, in _dispatch
    raise ValueError("invalid message type: %r" % (msg,))
ValueError: invalid message type: None

### Steps to Reproduce the issue:
1. Connect the WaveShare board (https://www.waveshare.com/High-Precision-AD-DA-Board.htm)
2. Go to the data tab and add an input, selecting the "Texas Instruments: ADS1256: Voltage (WaveShare, Analog to Digital Converter)(UART)" option.
3. Get dependencies flag
4. Click "Install Dependencies" button.
5. Wait for installation to complete.
6. Go back to the "Data" page.
7. Try to activate input source.
8. Get an error and realize that SPI is not enabled.
9. Run "raspi-config" and enable SPI.
10. Notice the daemon is not running.
11. Reboot the Pi. 
12. See 
 
### Additional Notes
TBD

kizniche, 2019-09-03:
Does this only affect the /info page? Are you sure the daemon has stopped (what indicates the daemon has stopped)? Can you paste any relevant log lines from the Daemon log?

kizniche, 2019-09-03:
Do you have a /dev/spi* device that exists (the ADC)?

smatthews95, 2019-09-04:
The indication that the daemon has stopped is a couple of things. First, the text in in the top left that lists the Mycodo version turns red and a message says that the daemon has stopped when you hover over that text. Second, the graphs and other dashboard elements stop working.

Here are the lines that look relevant from the daemon log:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpyc/utils/server.py", line 180, in _authenticate_and_serve_client
    self._serve_client(sock2, credentials)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpyc/utils/server.py", line 202, in _serve_client
    self._handle_connection(conn)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpyc/utils/server.py", line 208, in _handle_connection
    conn.serve_all()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 400, in serve_all
    self.serve(None)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 384, in serve
    self._dispatch(data)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/rpyc/core/protocol.py", line 348, in _dispatch
    msg, seq, args = brine.load(data)
TypeError: cannot unpack non-iterable NoneType object
2019-09-04 12:14:00,897 - ERROR - mycodo.inputs.ads1256_f1339e97 - ADC returned measurement of 0 (indicating something is wrong).
2019-09-04 12:14:00,898 - ERROR - mycodo.controller_input_f1339e97 - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-09-04 12:14:01,906 - ERROR - mycodo.inputs.ads1256_f1339e97 - ADC returned measurement of 0 (indicating something is wrong).

I'm not entirely sure what you mean by your last question. However, "spidev0.0" and "spidev0.1" both show up in "/dev". I've mostly used I2C before. Is there an equivalent to "i2cdetect" for spi?

kizniche, 2019-09-04:
Can you provide the daemon log lines immediately prior to what you pasted?

kizniche, 2019-09-04:
Also, can you paste the demon log lines where the daemon stopped and indicate in the timeline when that happened?

kizniche, 2019-09-04:
Also, please check if you have success returning measurements from your ADC with running the example code from https://github.com/ul-gh/PiPyADC

Have you tested any code that returns measurements from your board to confirm you have it connected and communicating properly?

smatthews95, 2019-09-04:
Yes, I've run "exmple.py" from PiPyADC. It runs fine, though the values are jumping all over the place.

I did a reboot and recreated the event. The complete daemon log from initialization through the present is attached in a text document.

[Full Daemon Log](https://github.com/kizniche/Mycodo/files/3576391/Full.Daemon.Log)

The event takes place at approximately 14:56:37.

kizniche, 2019-09-04:
I've never seen that RPyC error before. I'm not sure what to make of it. I'm hesitant to even try to start debugging RPyC because in the latest version (unreleased) RPyC has been completely replaced by Pyro4.

You can try a couple things:

1. It now looks like https://github.com/ul-gh/PiPyADC is compatible with Python 3 (not just Python 2). Because of this, and because there have been bug fixes since my fork, I've gone ahead and synced my repository to his, and you can now grab the latest code.

2. You can install the latest Mycodo code that's currently unreleased to try to alleviate the RPyC issue.

I would start with 1, as it's the easiest to try. If that doesn't work, you can try 2.

### 1. Upgrade the PiPyADC library

```bash
cd ~/Mycodo
./env/bin/pip install --upgrade -e git://github.com/kizniche/PiPyADC-py3.git#egg=pipyadc_py3
sudo service mycodo restart
```

### 2. install the latest Mycodo code

If you want to give the latest code a run, you can backup your current Mycodo install and install this unreleased version with the following commands:

```bash
sudo service mycodo stop
sudo service mycodoflask stop
cd ~
mv Mycodo Mycodo-old
git clone https://github.com/kizniche/Mycodo
sudo /bin/bash ~/Mycodo/install/setup.sh
```

kizniche, 2019-09-11:
7.7 has been released. See if upgrading to it fixes your issue.

closed 2019-09-11 by kizniche