kizniche/Mycodo #684: PWM output - Cound not connect to pigpiod

Baldius, 2019-09-08:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 ver. 7.6.3 on Raspberry Pi 3 B+, Raspian Buster Lite (Kernel version: 4.19)

Hello Kyle and everybody else!

#### Problem Description

My PWM outputs through hardware pins or any other pins do not work. After trying to set the duty cycle an error appears:

[image]

For trying to resolve the problem I made a clear install of Raspian Buster Lite (Version July 2019) and Mycodo (after expansion of the file system, setting up raspi-config like enabling SSH, reboot). The only change I made was adding a PWM (GPIO) output and installed the dependencies (pigpio) after I was prompted to do so, but the problem remains.

I also uninstalled pigpiod inside Mycodo (Configure --> Raspberry Pi --> Service options: Configure pigpiod --> Uninstall pigpiod) and reinstall it in that way after rebooting - still the same problem.

[image]

 
### Errors

The corresponding Daemon (ERROR) log:

> 2019-09-08` 04:56:50,894 - ERROR - mycodo.controller_output - PWM 91e77480-bee0-4b1a-875e-eab42b8931de was unable to be setup on pin 13: 'NoneType' object has no attribute 'sendall'

> 2019-09-08` 04:56:54,891 - ERROR - mycodo.daemon - Could not turn output on: 'NoneType' object has no attribute 'sendall'

> 2019-09-08 04:57:19,015 - ERROR - mycodo.controller_output - Cound not connect to pigpiod

### Steps to Reproduce the issue:

Making a clear installation of Raspian Buster Lite and Mycodo, adding a PWM Output.

 
### Additional Notes

I also did read #617 but it didn't solve my problem - after prompted to install the dependency, reboot - then adding GPIO, set up PWM and turn PWM % did not help.

kizniche, 2019-09-08:
Thanks for the bug report. I'll look into this and see if I can replicate the issue.

kizniche, 2019-09-08:
I was able to replicate the error and push a fix. It appears the location of pigpiod has changed from /usr/local/bin to /usr/bin/, which causes the systemd scripts to attempt to execute a non-existent file.

This fix won't be available until the next release, but you can make the changes found in commit 1e6d0dd38b0ca2ac29accb8056a53c4669b15a60 and then execute these commands to enable the service file:

```bash
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh disable-pigpiod
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh enable-pigpiod-low
```

kizniche, 2019-09-08:
Since this is a pretty big bug, I've gone ahead and released v7.7.0. So, you can upgrade normally rather than editing those files manually. You may still need to run those commands after upgrading, though.

kizniche, 2019-09-08:
Actually, found an issue. Fixing before making final release.

Baldius, 2019-09-08:
Thank you very much for that fast reply! I already upgraded before getting your last comment and now will wait for that final release before testing it.

kizniche, 2019-09-08:
I just released v7.7.1 which has bug fixes I found upon release of 7.7.0. Try an upgrade, then execute those two commands, above, and see if you can set a PWM.

Baldius, 2019-09-08:
After upgrading to 7.7.1, executing the commands above and rebooting no error appears anymore:

[image]

But unfortunately it seems like there is no PWM output. I have connected two servo motors and they remain still.

This is the output when I executed the two commands:

`pi@Viktor:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh disable-pigpiod`

#### Disabling installed pigpiod startup script
Removed /etc/systemd/system/multi-user.target.wants/pigpiod_low.service.
Removed /etc/systemd/system/pigpiod_low.service.
Removed /etc/systemd/system/pigpiod.service.
Failed to disable unit: Unit file pigpiod_low.service does not exist.
Failed to disable unit: Unit file pigpiod_high.service does not exist.

`pi@Viktor:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh enable-pigpiod-low`

#### Enabling pigpiod startup script (1 ms sample rate)
Created symlink /etc/systemd/system/pigpiod.service → /home/pi/Mycodo/install/pigpiod_low.service.
Created symlink /etc/systemd/system/multi-user.target.wants/pigpiod_low.service → /home/pi/Mycodo/install/pigpiod_low.service.
Created symlink /etc/systemd/system/pigpiod_low.service → /home/pi/Mycodo/install/pigpiod_low.service.

After trying to set the PWM output there are no errors is die Daemon log referring to the PWM output.

kizniche, 2019-09-08:
You might have to restart the daemon after executing those two commands. Does the output turn green with the duty cycle displayed after manually turning it on from the Outputs page?

kizniche, 2019-09-08:
Disregard my previous comment about the output. It does appear to be working. Do you have a multimeter or oscilloscope to test?

kizniche, 2019-09-08:
Make sure you're using the correct pin. The numbering is for BCM 12, so verify this is the pin you're using and not physical pin 12.

Baldius, 2019-09-08:
I just testet with an oscilloscope - you're right, it looks like PWM is working fine. I also used the right pin numbering (physical 32 for BCM 12), so I guess something's wrong with my motors or I messed up something else. But it's not PWM anymore, so thank you very much for fixing that problem!

kizniche, 2019-09-08:
Glad to help. Thanks for the bug report and helping get this fixed. I'll close this issue as it seems to now be working. Let me know if you have any other issues and good luck getting your servos spinning.

closed 2019-09-08 by kizniche