kizniche/Mycodo #687: SHT35 Sensor

chitters2004, 2019-09-16:
Having difficulty adding SHT35 sensor. Added dependencies for SHT31 and grove.py, had it working for a while but no longer.
[Mycodo log v1.txt](https://github.com/kizniche/Mycodo/files/3614537/Mycodo.log.v1.txt)
[Mycodo log - grove.py.txt](https://github.com/kizniche/Mycodo/files/3614538/Mycodo.log.-.grove.py.txt)
[sht35v1.txt](https://github.com/kizniche/Mycodo/files/3614547/sht35v1.txt)
[SHT35grove.txt](https://github.com/kizniche/Mycodo/files/3614548/SHT35grove.txt)

kizniche, 2019-09-16:
It looks like the custom input you made wasn't modified enough to use the grove.py library. For the custom input I provided, do you have a log of any output after activating that one? Make sure to select the correct I2C address, as I believe 0x44 will be selected by default, but 0x45 is what should be selected, if I'm reading correctly.

Have you been able to get any code to return a measurement, to verify you have it wired correctly and are using the correct I2C address?

kizniche, 2019-09-16:
For reference, this is related to https://kylegabriel.com/forum/general-discussion/custom-inputs-and-pid-for-grow-tent/

Also, can you verify this is the board/sensor you have? http://wiki.seeedstudio.com/Grove-I2C_High_Accuracy_Temp&Humi_Sensor-SHT35/

kizniche, 2019-09-16:
I see you mention it was working for a while, but not anymore. This is indeed strange. It indicates there may be a wiring issue. Is your sensor connected by a set of short wires and you're sure the connections are secure?

Also, just to make sure, you're adding the custom inputs to Mycodo through [Gear Icon] -> Configure -. Inputs?

chitters2004, 2019-09-16:
Sorry for late reply, the setup is offsite and have just managed to get up here. I have rewired the sensor and there  must have been a dodgy connection as am now getting readings again using both custom inputs. Using the one you kindly wrote for me I am now getting the following in the log:

2019-09-16 15:09:17,136 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.817196917677578, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.63523308155948, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.587727541146577, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 649.2740030537091, 'timestamp_utc': None}}
2019-09-16 15:09:22,146 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.80117494468604, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.63981078812849, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.573011884479794, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 648.4967307004886, 'timestamp_utc': None}}
2019-09-16 15:09:27,155 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.83054856183719, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.75577935454338, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.625154872805414, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 645.2688328007195, 'timestamp_utc': None}}
2019-09-16 15:09:32,163 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.80117494468604, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.67795834287023, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.58073223776555, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 647.0717137129249, 'timestamp_utc': None}}
2019-09-16 15:09:37,174 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.817196917677578, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.69474326695659, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.59977192191627, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 647.0488988253177, 'timestamp_utc': None}}
2019-09-16 15:09:42,183 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.817196917677578, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.71305409323263, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.60347635941797, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 646.364251370428, 'timestamp_utc': None}}
2019-09-16 15:09:47,192 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.817196917677578, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.67338063630122, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.59544917091811, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 647.8476541893556, 'timestamp_utc': None}}
2019-09-16 15:09:52,207 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.80117494468604, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.69016556038758, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.583202092519564, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 646.6157082769045, 'timestamp_utc': None}}
2019-09-16 15:09:57,114 - DEBUG - mycodo.controllers.controller_input_03a80cd6 - Adding measurements to InfluxDB with ID 03a80cd6-e70a-4a7a-90f7-9b3067a72988: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.78782330052644, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.70542458228428, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 24.573252305024145, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 645.5428934670488, 'timestamp_utc': None}}
[... cut ...]
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 122, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/custom_inputs/sht35.py", line 99, in get_measurement
    self.bus.write_i2c_block_data(self.i2c_address, 0x24, [0x00])
  File "/var/mycodo-root/env/lib/python3.7/site-packages/smbus2/smbus2.py", line 482, in write_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error

I am indeed using http://wiki.seeedstudio.com/Grove-I2C_High_Accuracy_Temp&Humi_Sensor-SHT35/ and using 0x45.

kizniche, 2019-09-16:
Are the measurements from both inputs returning similar values? If so, I will merely rename the SHT31 input to SHT31/35 as the Adafruit_SHT31 library appears to work with both. Looking at the datasheet, the SHT30 should also work, so it may be better to name it "SHT3x".

Now that we have this issue resolved, we can go back to the forum to discuss PID tuning and other setup considerations/configurations. I'll leave a longer reply on the forum later when I get home.

chitters2004, 2019-09-16:
They are indeed. Reverted to SHT31 Lib.

closed 2019-09-16 by chitters2004