kizniche/Mycodo #688: PID Controllers not issuing outputs

chitters2004, 2019-09-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.7.3
 
#### Problem Description
Please list:

Updated to latest. PID controllers not functioning
 
### Errors
 2019-09-18 00:07:24,781 - ERROR - mycodo.controllers.controller_pid_ad0b8bc6 - Exception executing check_pid() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 122, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 350, in check_pid
    self.manipulate_output()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 712, in manipulate_output
    min_off=self.raise_min_off_duration)
  File "/var/mycodo-root/mycodo/mycodo_client.py", line 183, in output_on
    return self.pyro_server.output_on(
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 90, in __getattr__
    self._pyroGetMetadata()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 356, in _pyroGetMetadata
    self.__pyroCreateConnection()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 329, in __pyroCreateConnection
    self.__check_owner()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 456, in __check_owner
    raise errors.PyroError("the calling thread is not the owner of this proxy, "
Pyro5.errors.PyroError: the calling thread is not the owner of this proxy, create a new proxy in this thread or transfer ownership.

kizniche, 2019-09-18:
It looks like I rushed in replacing Pyro4 with Pyro5. I'm pushing a fix that will correct PIDs turning outputs on and off, but there are a lot of other functions I need to modify for them to work. This is a small bandaid until I have enough time to fix it all. If you need a fully-operational system, I suggest you use the restore feature to restore to an earlier backed up version.

kizniche, 2019-09-18:
I figured out a much simpler solution than what I was planning to do. I just pushed that fix and released v7.7.4.

closed 2019-09-18 by kizniche

chitters2004, 2019-09-18:
Thank you, PID indeed working again. Still getting the odd Remote I/O error though.

2019-09-18 10:53:45,362 - ERROR - mycodo.inputs.sht31_6162b51f - InputModule.get_measurement() method raised IOError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 122, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/sht31.py", line 146, in get_measurement
    self.value_set(1, self.sensor.read_humidity())
  File "/home/pi/Mycodo/env/src/adafruit-sht31/Adafruit_SHT31.py", line 133, in read_humidity
    (temperature, humidity) = self.read_temperature_humidity()
  File "/home/pi/Mycodo/env/src/adafruit-sht31/Adafruit_SHT31.py", line 112, in read_temperature_humidity
    buffer = self._device.readList(0, 6)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 134, in readList
    results = self._bus.read_i2c_block_data(self._address, register, length)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 215, in read_i2c_block_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
OSError: [Errno 121] Remote I/O error

kizniche, 2019-09-18:
Try the custom Input I provided earlier and see if the same IO errors occur.

kizniche, 2019-09-20:
SHT31 Issue moved to #695

commit 0f8f8bd9 (2019-09-18): Release v7.7.4: Fix issue with Pyro5 proxy handling (closes #688)