kizniche/Mycodo #689: Raspberry Pi 4 Requires latest wiringpi needs 2.52

chitters2004, 2019-09-17:
Attempts to manually update before mycodo installation are reverted. 

Attempts post installation fail to update.

https://project-downloads.drogon.net/wiringpi-latest.deb

Theoi-Meteoroi, 2019-09-17:
I think it is also important to mention Gordon's post about WiringPi.

[The End of Everything as we know it](http://wiringpi.com/wiringpi-deprecated/)

I work as a support guy in my day job.   And to see folks that give support for no charge really seems a charity and act of kindness.    I enjoy helping where I can however I have no expectation of the author to figure it out for me, or to train me in the methods required to use the code.   

This code runs in a python virtualenv.    There is your first clue.  Now read the installation script.

Open Source is participatory, not a spectator occupation.

kizniche, 2019-09-18:
@chitters2004 What are you using wiringpi for? Are there any alternatives?

@Theoi-Meteoroi That was a depressing read, but thanks for sharing it. I've felt the same way at times. Fortunately I still find Mycodo worth the effort to keep it open and up to date. Besides knowing the software helps a lot of people, it's [the little things like this](https://github.com/kizniche/Mycodo/issues/671#issuecomment-530797782) that keep me motivated.

chitters2004, 2019-09-20:
When using rasPi 4 unable to get readout with  i2cdetect -y 1 . When I updated wiring pi it worked but then mycodo install reverted it with no obvious way to me to update again.

kizniche, 2019-09-20:
Are you using the ads1256 that you need this for, or something external from Mycodo?

kizniche, 2019-09-20:
Oh, I see i2cdetect. Let me see.

kizniche, 2019-09-20:
The only place wiringpi is installed is during the initial installation, and if the ads1256 Input is installed. Otherwise, only if the ads1256 is installed, it will upgrade the wiringpi Python library when the Mycodo upgrade happens. It doesn't seem like the upgrade caused the issue if you didn't have the ads1256 Input module installed.

chitters2004, 2019-09-20:
Oh ok, is ic2detect nothing to do with wiringpi? It may have been a coincidence that the command/output in system info only worked after updating wiringpi.

chitters2004, 2019-09-20:
oh damn, i meant gpio readall

chitters2004, 2019-09-20:
has a board error returned until I updated it to 2.52

chitters2004, 2019-09-20:
and was unable to configure GPIO outputs

kizniche, 2019-09-20:
I didn't think RPi.GPIO that's used to switch the GPIO pins depended on wiringpi. Let me look into that.

kizniche, 2019-09-20:
Or are you referring to switching GPIO pins outside of Mycodo, with the gpio executable?

kizniche, 2019-09-20:
Since wiringpi is being deprecated, I might just remove it entirely from Mycodo and let the user decide to install it. That will remove conflicts with user configurations. The only purpose of wiringpi in Mycodo at the moment is the as1256 Input, and I believe it's entirely handled by the wiringpi Python module that's installed inside the virtualenv, so the build/install from https://github.com/WiringPi/WiringPi-Python that's performed in the install/setup.sh during the initial installation is likely unnecessary.

Let me know if you were referring to GPIO manipulation outside of Mycodo. If so, I'll see about removing the winringpi install during the Mycodo install.

chitters2004, 2019-09-20:
All was working well, tried to get GPIO output to relay but nothing was happening. Found the error in mycodo system info. googled the error, it pointed to wiring pi, so tried to update that but it stayed at the same version. Did a fresh flash, tried gpio readall and got the same unidentified board (17 error), when i installed latest wiringpi it gave me the output.

chitters2004, 2019-09-20:
Will do a fresh install on rPi 4 later and see if i can replicate

kizniche, 2019-09-20:
Can you also paste the error you get?

For reference, this is also the only reference to gpio I can find in Mycodo:

https://github.com/kizniche/Mycodo/blob/ed2da82d4a12840ffbe7bce41b1e5fe0b3d84cf5/mycodo/mycodo_flask/routes_page.py#L854

chitters2004, 2019-09-20:
I believe it was 

`Oops - unable to determine board type... model: 17`

chitters2004, 2019-09-20:
People online were having problems when 3b came out too and solved by updating wiringpi

chitters2004, 2019-09-20:
The error appeared here:

[image]

kizniche, 2019-09-20:
I just updated the wiringpi install script section to install fro the deb instead of building from source. If you have time, can you fresh flash Raspbian and fresh install Mycodo to see if everything works out of the box? Here's how to install from master:

```
cd
apt install git
git clone https://github.com/kizniche/Mycodo
sudo /bin/bash ~/Mycodo/install/setup.sh
```

kizniche, 2019-09-20:
But it would be beneficial to know if the Mycodo Outputs were really affected rather than maybe you caught an update I made that fixed it between your testing installs. If you have the error from the log, that would be helpful.

chitters2004, 2019-09-20:
I have a raspi4 fresh install from last night, checking it now

chitters2004, 2019-09-20:
[image]

kizniche, 2019-09-20:
But do the Outputs work from within Mycodo?

kizniche, 2019-09-20:
If it's only the "gpio readall" command that's impacted, that's pretty minor and I can replace that on the I do page with software that's still maintained.

chitters2004, 2019-09-20:
log file looks ok when i set up dummy output with nothing connected. If it makes a difference to have an actual output connected i can check tomorrow.

chitters2004, 2019-09-20:
Shall I do that before testing master update?

kizniche, 2019-09-20:
If it turns green when you click ON its likely working, because the status indicator (red/green color) is determined girectly from RPi.GPIO reading the pin state. So it appears it's just "gpio readall" that's affected in Mycodo. But please do test with a multimeter or relay when you can. I don't yet have a Pi4 to test with.

kizniche, 2019-09-20:
A master upgrade won't affect the currently-installed wiringpi, so you can do that and still test later.

chitters2004, 2019-09-20:
sorry, I meant fresh install after testing the outputs on 7.7.6

kizniche, 2019-09-20:
If your Outputs turn on in Mycodo with 7.7.6, then it's not too important to do a fresh install, because I'll be looking for a replacement for wiringpi, as it's no longer maintained.

chitters2004, 2019-09-20:
Ok, will do a test tomorrow anyway and then keep my pi4 current with latest as test platform alongside 3b+

closed 2019-10-03 by kizniche