kizniche/Mycodo #690: "Error: Activate Conditional: Measurement must be set" since update 7.7.3 & 7.7.4

buzzfuzz2k, 2019-09-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.7.3 & 7.7.4
 
#### Problem Description
Since last 2 updates (7.7.3 & 7.7.4) my functions doesn't work anymore?
When I save the code, i get the green validation but when i want to activate a function i get this error code:
"Error: Activate Conditional: Measurement must be set"
https://www.cjoint.com/doc/19_09/IIsqhLUJzMO_Capture2.PNG

I also tried to recreate my function with new measure ID, but same....
here is a part of the code (i removed the second part, not usefull now)
https://www.cjoint.com/doc/19_09/IIsqhuXwXdO_Capture.PNG

thanks

 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2019-09-18:
> I also tried to recreate my function

Does this mean you added a completely new function, after upgrading to 7.7.4, with all the settings default, then set it up, and it still displayed this behavior?

I haven't been able to replicate this issue on 7.7.4, so that's strange.

buzzfuzz2k, 2019-09-18:
I did on 7.7.3

I'll try now

buzzfuzz2k, 2019-09-18:
I did on 7.7.4 and it seems to work

The other difference is that i have completely removed all old ID and conditionnal...

kizniche, 2019-09-18:
There are some configuration changes that rarely result in strange behavior if things are deleted that other things depend on. It's difficult to account for all case-use scenarios. Glad you resolved the issue.

buzzfuzz2k, 2019-09-18:
i may have find some new informations...
the problem is not with the version !

I did this function and it works:
`temperature_int = self.measure("{1b264478}")
if temperature_int is not None and temperature_int > 13.6:
    self.run_action("{b7eb48b4}")
if temperature_int is not None and temperature_int < 12.6:
    self.run_action("{b1130d8f}")`

I then add an "output state condition" to know if an output is already on or off. I use this code:
`temperature_int = self.measure('{1b264478}')
temps_compresseur = self.measure('{f75a5c35}')

if temperature_int is not None:
    if temperature_int > 13.6 and temps_compresseur == 'off':
        self.run_action("{b7eb48b4}") # allumer le compresseur
    elif temperature_int < 12.6 and temps_compresseur == 'on':
        self.run_action("{b1130d8f}") # eteindre le compresseur`

The "save" was ok (green), but when i want to activate, i get the message again!

I copy/paste the old code, same error but when i removed the conditional output state line, it worked!

as i said in this post 
https://kylegabriel.com/forum/general-discussion/conditional-wit-output-duration/

i would like to get this code to work
`temperature_int = self.measure('{1b264478}')
temps_compresseur = self.measure('{f75a5c35}')

if temperature_int is not None:
    if temperature_int > 13.6 and temps_compresseur == 'off':
        self.run_action("{b7eb48b4}") # allumer le compresseur
    elif temperature_int < 12.6 and temps_compresseur == 'on':
        self.run_action("{b1130d8f}") # eteindre le compresseur

if temperature_int is None or (temperature_int is not None and temperature_int > 16):
    output_on_duration = self.control.output_sec_currently_on('1b6ada50-1e69-403a-9fa6-ec748b16dc23')
    if output_on_duration > 600:
        self.run_action("{7d4fecd9}") # Turn output off
        self.run_action("{b9d216aa}") # email`

but the first part is not working

thank you for the help

kizniche, 2019-09-18:
Okah. That info should help me track down the issue.

kizniche, 2019-09-18:
Found the issues. Pushing a fix and releasing 7.7.5 shortly.

buzzfuzz2k, 2019-09-18:
great!

thanks

closed 2019-09-18 by kizniche