kizniche/Mycodo #692: Daemon/PID

chitters2004, 2019-09-19:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.7.5
 
#### Problem Description
Web interface indicates daemon not running.

pi@raspberrypi:~ $ ps aux | grep '/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py'
root       886  5.7  5.1 127028 46392 ?        Ssl  16:53   0:20 /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py
pi        2076  0.0  0.2   7616  2012 pts/0    S+   16:59   0:00 grep --color=auto /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py

### Errors
  2019-09-19 16:40:46,304 - ERROR - mycodo.controllers.controller_pid_f82dd4d9 - Exception executing check_pid() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 280, in connect_and_handshake
    sslContext=sslContext)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 286, in create_socket
    sock.connect(connect)
ConnectionRefusedError: [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/core.py", line 280, in locate_ns
    proxy._pyroBind()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 180, in _pyroBind
    return self.__pyroCreateConnection(True)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 340, in __pyroCreateConnection
    connect_and_handshake(conn)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 302, in connect_and_handshake
    raise errors.CommunicationError(err) from x
Pyro5.errors.CommunicationError: cannot connect to ('localhost', 9090): [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 122, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 352, in check_pid
    self.manipulate_output()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 805, in manipulate_output
    min_off=self.lower_min_off_duration)
  File "/var/mycodo-root/mycodo/mycodo_client.py", line 170, in output_on
    return self.proxy().output_on(
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 90, in __getattr__
    self._pyroGetMetadata()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 356, in _pyroGetMetadata
    self.__pyroCreateConnection()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 332, in __pyroCreateConnection
    uri = core.resolve(self._pyroUri)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/core.py", line 187, in resolve
    with locate_ns(uri.host, uri.port) as nameserver:
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/core.py", line 284, in locate_ns
    raise errors.NamingError("Failed to locate the nameserver") from x
Pyro5.errors.NamingError: Failed to locate the nameserver

kizniche, 2019-09-19:
Try a fresh install of Mycodo on a newly-flashed Raspbian.

kizniche, 2019-09-19:
Is this an error that occurs every attempt to manipulate the output, or is it rare?

kizniche, 2019-09-19:
Does it go through all 3 attempts and fails all 3?

chitters2004, 2019-09-19:
Have just completed fresh install at 7.7.5 and seems to be running ok now. Though still getting those intermittent errors on sht35. I switched back to your input and get the following in log.

2019-09-19 18:12:45,066 - INFO - mycodo.controllers.controller_input_29e1d667 - Activated in 245.8 ms
2019-09-19 18:13:06,995 - ERROR - mycodo.inputs.sht35_29e1d667 - InputModule.get_measurement() method raised IOError: [Errno 121] Remote I/O error
2019-09-19 18:13:15,047 - ERROR - mycodo.inputs.sht35_29e1d667 - InputModule.get_measurement() method raised IOError: [Errno 121] Remote I/O error
2019-09-19 18:13:22,995 - ERROR - mycodo.inputs.sht35_29e1d667 - InputModule.get_measurement() method raised IOError: [Errno 121] Remote I/O error
2019-09-19 18:13:27,425 - INFO - mycodo.controllers.controller_input_29e1d667 - Deactivated in 60.8 ms
2019-09-19 18:13:41,562 - INFO - mycodo.controllers.controller_input_29e1d667 - Activated in 259.9 ms
2019-09-19 18:13:41,691 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.634393835355155, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 58.88456549935149, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.672876590617292, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1808.6249635132792, 'timestamp_utc': None}}
2019-09-19 18:13:43,561 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.647745479514768, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 58.92118715190356, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.6955459682273, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1808.3917255784727, 'timestamp_utc': None}}
2019-09-19 18:13:45,526 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.661097123674367, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 58.95780880445563, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.71821168384661, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1808.1569433277873, 'timestamp_utc': None}}
2019-09-19 18:13:47,645 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.661097123674367, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 59.02342259861143, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.736424488728133, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1805.2662585464698, 'timestamp_utc': None}}
2019-09-19 18:13:49,656 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.677119096665905, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 59.08293278400855, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.767924016748925, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1804.293509676741, 'timestamp_utc': None}}
2019-09-19 18:13:51,566 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.677119096665905, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 59.15159838254368, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.786950294935195, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1801.2656070873152, 'timestamp_utc': None}}
2019-09-19 18:13:53,590 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.647745479514768, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 59.16838330663005, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.764100090321705, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1797.5095347406987, 'timestamp_utc': None}}
2019-09-19 18:13:55,594 - DEBUG - mycodo.controllers.controller_input_29e1d667 - Adding measurements to InfluxDB with ID 29e1d667-b365-4972-9d7d-b7dfe96ee2ac: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 30.677119096665905, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 59.15159838254368, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 21.786950294935195, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1801.2656070873152, 'timestamp_utc': No [... cut ...]
2019-09-19 18:15:05,491 - ERROR - mycodo.inputs.sht35_29e1d667 - InputModule.get_measurement() method raised IOError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 122, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/custom_inputs/sht35.py", line 106, in get_measurement
    data = self.bus.read_i2c_block_data(0x45, 0x00, 6)
  File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 596, in read_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error

chitters2004, 2019-09-19:
Also, sensor seems to shut down after a while so have run it through a relay as you previously suggested for DH22.

chitters2004, 2019-09-19:
Just rebooted and back with the same issue....

2019-09-19 18:35:06,516 - INFO - mycodo.daemon - Mycodo daemon v7.7.5 starting
2019-09-19 18:35:06,770 - DEBUG - mycodo.daemon - Refreshing camera settings
2019-09-19 18:35:06,820 - DEBUG - mycodo.daemon - Refreshing misc settings
2019-09-19 18:35:06,856 - DEBUG - mycodo.daemon - Anonymous statistics enabled
2019-09-19 18:35:06,879 - INFO - mycodo.daemon - Starting Pyro5 server
2019-09-19 18:35:07,193 - DEBUG - mycodo.daemon - Starting Output Controller
2019-09-19 18:35:07,324 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2019-09-19 18:35:07,423 - INFO - mycodo.controllers.controller_output - Output ab76b6e8 setup on pin 12 and turned OFF (OFF=HIGH)
2019-09-19 18:35:07,425 - DEBUG - mycodo.controllers.controller_output - ab76b6e8 (Channel 8 - Heater) Initialized
2019-09-19 18:35:07,431 - INFO - mycodo.controllers.controller_output - Output a4578f96 setup on pin 16 and turned OFF (OFF=HIGH)
2019-09-19 18:35:07,432 - DEBUG - mycodo.controllers.controller_output - a4578f96 (Channel 2 - Fan Out) Initialized
2019-09-19 18:35:07,433 - INFO - mycodo.controllers.controller_output - Output eb5057d3 setup on pin 20 and turned OFF (OFF=HIGH)
2019-09-19 18:35:07,433 - DEBUG - mycodo.controllers.controller_output - eb5057d3 (Channel 3 - Humidifier) Initialized
2019-09-19 18:35:07,434 - INFO - mycodo.controllers.controller_output - Output 91cb033b setup on pin 21 and turned OFF (OFF=HIGH)
2019-09-19 18:35:07,435 - DEBUG - mycodo.controllers.controller_output - 91cb033b (Channel 4 - CO2 Reg) Initialized
2019-09-19 18:35:07,449 - INFO - mycodo.controllers.controller_output - Output ad9bf24d setup on pin 25 and turned OFF (OFF=HIGH)
2019-09-19 18:35:07,449 - DEBUG - mycodo.controllers.controller_output - ad9bf24d (SHT35) Initialized
2019-09-19 18:35:07,450 - DEBUG - mycodo.controllers.controller_output - output_on_off(ab76b6e8-c60a-40f8-a285-1a6ccd85c0d7, off, 0.0, 0.0, 0.0, False)
2019-09-19 18:35:07,450 - DEBUG - mycodo.controllers.controller_output - Output 1 (Channel 8 - Heater) turned off.
2019-09-19 18:35:07,451 - DEBUG - mycodo.controllers.controller_output - output_on_off(a4578f96-6cc0-4a51-9a80-9485eaa5df26, off, 0.0, 0.0, 0.0, False)
2019-09-19 18:35:07,451 - DEBUG - mycodo.controllers.controller_output - Output 2 (Channel 2 - Fan Out) turned off.
2019-09-19 18:35:07,452 - DEBUG - mycodo.controllers.controller_output - output_on_off(eb5057d3-60aa-451d-b642-b94a0a301859, off, 0.0, 0.0, 0.0, False)
2019-09-19 18:35:07,452 - DEBUG - mycodo.controllers.controller_output - Output 3 (Channel 3 - Humidifier) turned off.
2019-09-19 18:35:07,453 - DEBUG - mycodo.controllers.controller_output - output_on_off(91cb033b-ef04-4b21-8712-dd4be3a1ea87, off, 0.0, 0.0, 0.0, False)
2019-09-19 18:35:07,453 - DEBUG - mycodo.controllers.controller_output - Output 4 (Channel 4 - CO2 Reg) turned off.
2019-09-19 18:35:07,453 - DEBUG - mycodo.controllers.controller_output - output_on_off(ad9bf24d-a2dd-468f-a028-2a732bca95fd, on, 0.0, 0.0, 0.0, True)
2019-09-19 18:35:07,561 - DEBUG - mycodo.controllers.controller_output - Output 5 (SHT35) ON at 2019-09-19 18:35:07.561324.
2019-09-19 18:35:07,674 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2019-09-19 18:35:07,674 - INFO - mycodo.controllers.controller_output - Activated in 478.3 ms
2019-09-19 18:35:08,176 - DEBUG - mycodo.daemon - Output Controller fully started
2019-09-19 18:35:08,176 - DEBUG - mycodo.daemon - Starting all activated Conditional controllers
2019-09-19 18:35:08,177 - INFO - mycodo.daemon - All activated Conditional controllers started
2019-09-19 18:35:08,177 - DEBUG - mycodo.daemon - Starting all activated Trigger controllers
2019-09-19 18:35:08,177 - INFO - mycodo.daemon - All activated Trigger controllers started
2019-09-19 18:35:08,178 - DEBUG - mycodo.daemon - Starting all activated Input controllers
2019-09-19 18:35:08,178 - INFO - mycodo.daemon - All activated Input controllers started
2019-09-19 18:35:08,179 - DEBUG - mycodo.daemon - Starting all activated Math controllers
2019-09-19 18:35:08,179 - INFO - mycodo.daemon - All activated Math controllers started
2019-09-19 18:35:08,179 - DEBUG - mycodo.daemon - Starting all activated PID controllers
2019-09-19 18:35:08,180 - INFO - mycodo.daemon - All activated PID controllers started
2019-09-19 18:35:08,180 - DEBUG - mycodo.daemon - Starting all activated LCD controllers
2019-09-19 18:35:08,180 - INFO - mycodo.daemon - All activated LCD controllers started
2019-09-19 18:35:08,681 - INFO - mycodo.daemon - Mycodo daemon started in 2.151 seconds
2019-09-19 18:35:08,686 - INFO - mycodo.daemon - 36.96 MB RAM in use
2019-09-19 18:35:09,031 - ERROR - mycodo.pyro - ERROR: ComThread: Failed to locate the nameserver
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 280, in connect_and_handshake
    sslContext=sslContext)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/socketutil.py", line 286, in create_socket
    sock.connect(connect)
ConnectionRefusedError: [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/core.py", line 280, in locate_ns
    proxy._pyroBind()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 180, in _pyroBind
[... cut ...]
Pyro5.errors.CommunicationError: cannot connect to ('localhost', 9090): [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 277, in run
    ns = locate_ns()  # find the name server
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/core.py", line 284, in locate_ns
    raise errors.NamingError("Failed to locate the nameserver") from x
Pyro5.errors.NamingError: Failed to locate the nameserver

chitters2004, 2019-09-19:
Would it help to give you access?

kizniche, 2019-09-19:
The daemon may be starting before the nameserver. Try editing the ~/Mycodo/install/mycodo.service file and changing the third line to:

```
After=syslog.target network.target influxd.service mycodopyro.service
```

Save, then issue the following commands:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-mycodo-startup-script
```

Then reboot and see if the issue persists.

chitters2004, 2019-09-19:
Still getting the same i'm afraid

kizniche, 2019-09-19:
What version Raspberry Pi and Raspbian OS?

kizniche, 2019-09-19:
Check if the pyro5-ns is running

```
sudo service mycodopyro status
```

chitters2004, 2019-09-19:
Just doing a fresh flash on new sd atm, will check afterwards if that fails again.

kizniche, 2019-09-19:
You can also try adding ```ExecStartPre=/bin/sleep 10``` under ```[Service]``` of ```~/Mycodo/intsall/mycodo.service```, save, then ```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-mycodo-startup-script```

chitters2004, 2019-09-19:
No luck there either. Again all worked fine until restart and then back to the same issue.  

pyro5-ns is running

Raspi 3B+ on buster lite

kizniche, 2019-09-19:
Have you made both edits from [here](https://github.com/kizniche/Mycodo/issues/692#issuecomment-533245090) and [here](https://github.com/kizniche/Mycodo/issues/692#issuecomment-533269717)?

Please paste the output from these commands after a fresh bootup:

```
sudo service mycodopyro status && sudo service mycodo status
```

chitters2004, 2019-09-19:
I have done both

 mycodopyro.service - Pyro5 NameServer
   Loaded: loaded (/home/pi/Mycodo/install/mycodopyro.service; enabled; vendor p
   Active: active (running) since Thu 2019-09-19 21:26:58 BST; 17min ago
 Main PID: 350 (pyro5-ns)
    Tasks: 6 (limit: 2200)
   Memory: 12.5M
   CGroup: /system.slice/mycodopyro.service
           └─350 /home/pi/Mycodo/env/bin/python3 /var/mycodo-root/env/bin/pyro5-

Sep 19 21:26:58 raspberrypi systemd[1]: Started Pyro5 NameServer.

mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor prese
   Active: active (running) since Thu 2019-09-19 21:27:19 BST; 17min ago
  Process: 584 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 758 ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo
 Main PID: 764 (python)
    Tasks: 4 (limit: 2200)
   Memory: 36.5M
   CGroup: /system.slice/mycodo.service
           └─764 /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_

Sep 19 21:27:05 raspberrypi systemd[1]: Starting Mycodo server...
Sep 19 21:27:19 raspberrypi systemd[1]: Started Mycodo server.
Sep 19 21:27:19 raspberrypi mycodo_daemon[764]: Starting daemon.

kizniche, 2019-09-19:
Can you put the following code into a file ~/test_pyro_ns_lookup.py

```python
import traceback

from Pyro5.api import locate_ns

try:
    ns = locate_ns()  # find the name server
    print("Nameserver: {}".format(ns))
except Exception:
    traceback.print_exc()
```

Then execute with:

```bash
~/Mycodo/env/bin/python ~/test_pyro_ns_lookup.py
```

chitters2004, 2019-09-19:
pi@raspberrypi:~ $ ~/Mycodo/env/bin/python ~/test_pyro_ns_lookup.py
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/client.py", line 2                     80, in connect_and_handshake
    sslContext=sslContext)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/socketutil.py", li                     ne 286, in create_socket
    sock.connect(connect)
ConnectionRefusedError: [Errno 111] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/core.py", line 280                     , in locate_ns
    proxy._pyroBind()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/client.py", line 1                     80, in _pyroBind
    return self.__pyroCreateConnection(True)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/client.py", line 3                     40, in __pyroCreateConnection
    connect_and_handshake(conn)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/client.py", line 3                     02, in connect_and_handshake
    raise errors.CommunicationError(err) from x
Pyro5.errors.CommunicationError: cannot connect to ('localhost', 9090): [Errno 1                     11] Connection refused

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/test_pyro_ns_lookup.py", line 6, in <module>
    ns = locate_ns()  # find the name server
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/core.py", line 284                     , in locate_ns
    raise errors.NamingError("Failed to locate the nameserver") from x
Pyro5.errors.NamingError: Failed to locate the nameserver

chitters2004, 2019-09-19:
pi@raspberrypi:~ $ sudo service mycodopyro stop
pi@raspberrypi:~ $ ~/Mycodo/env/bin/python ~/Mycodo/env/bin/pyro5-ns
Not starting broadcast server for localhost.
NS running on localhost:9090 (127.0.0.1)
URI = PYRO:Pyro.NameServer@localhost:9090

kizniche, 2019-09-19:
Very strange. This is my output:

```
pi@LabDev:~/Mycodo $ ~/Mycodo/env/bin/python ~/Mycodo/env/bin/pyro5-ns
Not starting broadcast server for IPv6.
NS running on localhost:9090 (::1)
URI = PYRO:Pyro.NameServer@localhost:9090
```

I'm not exactly sure what's wrong, but I did just find a way to exclude the pyro5-ns nameserver entirely from the Mycodo system, and the client communicates directly with the daemon. To test this code, you will have to install from master. You can disable your current install and install from master with the following commands:

```
sudo service mycodoflask stop
sudo service mycodo stop
cd
sudo mv Mycodo Mycodo-old
sudo apt install git
git clone https://github.com/kizniche/Mycodo
sudo /bin/bash ~/Mycodo/install/setup.sh
```

chitters2004, 2019-09-19:
Thank you Kyle, running that now.

chitters2004, 2019-09-19:
Have reinstalled and done multiple reboots, seems like that issue is fixed now.  Now i'm seeing no outputs when i go to set up graphs.

kizniche, 2019-09-19:
> Now i'm seeing no outputs when i go to set up graphs

What do you mean by this?

chitters2004, 2019-09-19:
[image]

chitters2004, 2019-09-19:
[image]

kizniche, 2019-09-19:
Just to make sure, after installing Raspbian, you performed an ```sudo apt update``` and ```sudo apt upgrade```?

chitters2004, 2019-09-19:
Pretty certain I did but will run again!

kizniche, 2019-09-19:
I'm performing a fresh Mycodo install right now. I'll add an output when it finishes to see if I can replicate your issue.

chitters2004, 2019-09-19:
Have upgraded but no change. I have had the issue the first time I updated to 7.7.5 and every install since.

kizniche, 2019-09-19:
I replicated the behavior. I'm investigating the issue now.

chitters2004, 2019-09-19:
Thank you, sorry for causing all the extra work!

kizniche, 2019-09-19:
I appreciate the bug reports. I can only test so much and I don't have a lot of test coverage.

kizniche, 2019-09-19:
I just pushed my fixes to master. Can you [install from master](https://github.com/kizniche/Mycodo/issues/692#issuecomment-533312194) again and test if everything is working? If so, I'm going to make a release for everyone else so not to many of this broken release gets out in the wild.

chitters2004, 2019-09-19:
Giving that a go now

chitters2004, 2019-09-19:
Brilliant, seems to be working again. Would it be possible to add the soft and hard reset options for SHTX sensors? As referenced in https://github.com/SeeedDocument/Grove-I2C_High_Accuracy_Temp-Humi_Sensor-SHT35/raw/master/res/Datasheet%20SHT3x-DIS.pdf

kizniche, 2019-09-20:
Where in the datasheet? Does the Adafruit_SHT31 library already have that as an option?

chitters2004, 2019-09-20:
4.9 on p12.   Soft reset command is in the sht31 library : SHT31_SOFTRESET = 0x30A2

chitters2004, 2019-09-20:
def _reset(self):
        """
        Soft reset the device
        The reset command is preceded by a break command as the
        device will not respond to a soft reset when in 'Periodic' mode.
        """
        self._command(_SHT31_PERIODIC_BREAK)
        time.sleep(.001)
        self._command(_SHT31_SOFTRESET)
        time.sleep(.0015)

kizniche, 2019-09-20:
Can you provide a reference for that code?

kizniche, 2019-09-20:
I think I found it. That appears to be a circuit python library, not the one currently in use by Mycodo.

chitters2004, 2019-09-20:
https://github.com/ralf1070/Adafruit_Python_SHT31/blob/master/Adafruit_SHT31.py

Is that not the one referenced by Mycodo?

kizniche, 2019-09-20:
That's the one.

chitters2004, 2019-09-20:
def reset(self):
        self._writeCommand(SHT31_SOFTRESET)
        time.sleep(0.01)  # Wait the required time

kizniche, 2019-09-20:
Ah. Missed that when I looked through last night.

chitters2004, 2019-09-20:
No problem, I referenced the wrong library! 
This is low priority but do you think this light api could be incorporated into Mycodo? 

https://support.heliospectra.com/portal/kb/articles/api-documentation-all-series-models
http://support.heliodevteam.com/API%20documentation%20(all%20series%20&%20models)/tcp_control_socket_140919.pdf

kizniche, 2019-09-20:
Let's take this discussion back to the forum (Ref: https://kylegabriel.com/forum/pid-tuning/custom-inputs-and-pid-for-grow-tent), and close this issue.

closed 2019-09-20 by kizniche