kizniche/Mycodo #693: Unable to restore

chitters2004, 2019-09-19:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.7.5 -> 7.7.4
2019-09-19 17:04:58] #### Creating symlinks to Mycodo executables
[2019-09-19 17:04:58] 
[2019-09-19 17:04:58] #### Creating files and directories
[2019-09-19 17:04:58] 
[2019-09-19 17:04:58] Backing up current Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2019-09-19_17-04-39-7.7.5...Done.
[2019-09-19 17:04:58] 
[2019-09-19 17:04:58] Restoring Mycodo from /var/Mycodo-backups/Mycodo-backup-2019-09-19_12-51-45-7.7.4 to /home/pi/Mycodo...mv: cannot stat '/var/Mycodo-backups/Mycodo-backup-2019-09-19_12-51-45-7.7.4': No such file or directory
[2019-09-19 17:04:58] Failed: Error while trying to restore Mycodo backup from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2019-09-19_17-04-39-7.7.5.
[2019-09-19 17:04:58] /home/pi/Mycodo/mycodo/scripts/mycodo_backup_restore.sh: line 22: /home/pi/Mycodo/.restore: No such file or directory
[2019-09-19 17:04:58] 
[2019-09-19 17:04:58] 
[2019-09-19 17:04:58] Thu 19 Sep 17:04:58 BST 2019
[2019-09-19 17:04:58] #### ERROR ####
[2019-09-19 17:04:58] There was an error detected during the restore. Please review the log at /var/log/mycodo/mycodorestore.log[2019-09-19 17:04:58] /home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh: line 31: cd: /home/pi/Mycodo/mycodo/scripts/../..: No such file or directory
[2019-09-19 17:04:58] 
[2019-09-19 17:04:58] #### Setting permissions
[2019-09-19 17:04:58] chown: missing operand after ‘mycodo.mycodo’
[2019-09-19 17:04:58] Try 'chown --help' for more information.
[2019-09-19 17:04:59] chown: cannot access '/mycodo/scripts/mycodo_wrapper': No such file or directory
[2019-09-19 17:04:59] chmod: cannot access '/mycodo/scripts/mycodo_wrapper': No such file or directory
[2019-09-19 17:05:00] /home/pi/Mycodo/mycodo/scripts/mycodo_backup_restore.sh: line 83: /home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh: No such file or directory
[2019-09-19 17:05:00] Failed: Error while setting permissions.
[2019-09-19 17:05:00] /home/pi/Mycodo/mycodo/scripts/mycodo_backup_restore.sh: line 22: /home/pi/Mycodo/.restore: No such file or directory
[2019-09-19 17:05:00] 
[2019-09-19 17:05:00] 
[2019-09-19 17:05:00] Thu 19 Sep 17:05:00 BST 2019
[2019-09-19 17:05:00] #### ERROR ####
[2019-09-19 17:05:00] There was an error detected during the restore. Please review the log at /var/log/mycodo/mycodorestor

kizniche, 2019-09-19:
Not sure what the cause of those errors are. Looks like missing files/folders. You can move the backup manually to ~/Mycodo and execute ~/Mycodo/mycodo/scripts/post_uograde.sh as root.

chitters2004, 2019-09-19:
Did fresh install and back in business

closed 2019-09-19 by chitters2004