kizniche/Mycodo #694: No Readings BME280

cbaymac, 2019-09-20:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.7.5 Buster Lite
 
#### Problem Description
Please list: No sensor readings but works with 7.7.1 and Stretch Lite

  - what were you trying to do: Set up a PID controler
  - specific setup details that are involved: New SD card with Buster Lite or RPi 3.
 
### Errors
  - List any errors you encountered. No Sensor readings
  - Copy and pasting crash logs, or link to any specific
2019-09-19 19:52:41,715 - INFO - mycodo.daemon - Mycodo daemon v7.7.5 starting
2019-09-19 19:52:42,028 - INFO - mycodo.daemon - Starting Pyro5 server
2019-09-19 19:52:42,743 - INFO - mycodo.controllers.controller_output - Activated in 238.1 ms
2019-09-19 19:52:43,245 - INFO - mycodo.daemon - All activated Conditional controllers started
2019-09-19 19:52:43,245 - INFO - mycodo.daemon - All activated Trigger controllers started
2019-09-19 19:52:43,246 - INFO - mycodo.daemon - All activated Input controllers started
2019-09-19 19:52:43,247 - INFO - mycodo.daemon - All activated Math controllers started
2019-09-19 19:52:43,247 - INFO - mycodo.daemon - All activated PID controllers started
2019-09-19 19:52:43,248 - INFO - mycodo.daemon - All activated LCD controllers started
2019-09-19 19:52:43,749 - INFO - mycodo.daemon - Mycodo daemon started in 2.033 seconds
2019-09-19 19:52:43,753 - INFO - mycodo.daemon - 36.86 MB RAM in use
2019-09-19 19:54:00,645 - ERROR - mycodo.controllers.controller_input_bc0fc416 - Deactivated unexpectedly
2019-09-19 19:57:16,563 - ERROR - mycodo.controllers.controller_input_bc0fc416 - Deactivated unexpectedly
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1 New Buster Lite and Mycodo 7.7.5 installed on RPI 3 with BME280
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2019-09-20:
I'm going to be making a release tonight that includes a bunch of fixes, which may resolve your issue(s). I'd recommend installing Mycodo fresh once 7.7.6 has been released and see if the issues persist. If they do, we can start investigating.

kizniche, 2019-09-20:
7.7.6 Released

cbaymac, 2019-09-20:
Kyle,
 I see the same errors in 7.7.6 with regard to the BME280...no data
2019-09-20 02:57:30,551 - INFO - mycodo.daemon - Mycodo daemon v7.7.6
starting
2019-09-20 02:57:31,519 - INFO - mycodo.controllers.controller_output -
Activated in 240.9 ms
2019-09-20 02:57:32,022 - INFO - mycodo.daemon - All activated Conditional
controllers started
2019-09-20 02:57:32,022 - INFO - mycodo.daemon - All activated Trigger
controllers started
2019-09-20 02:57:32,023 - INFO - mycodo.daemon - All activated Input
controllers started
2019-09-20 02:57:32,024 - INFO - mycodo.daemon - All activated Math
controllers started
2019-09-20 02:57:32,024 - INFO - mycodo.daemon - All activated PID
controllers started
2019-09-20 02:57:32,025 - INFO - mycodo.daemon - All activated LCD
controllers started
2019-09-20 02:57:32,526 - INFO - mycodo.daemon - Mycodo daemon started in
1.974 seconds
2019-09-20 02:57:32,531 - INFO - mycodo.daemon - 36.84 MB RAM in use
2019-09-20 02:58:31,405 - ERROR -
mycodo.controllers.controller_input_40c2eb22 - Deactivated unexpectedly
2019-09-20 03:00:35,800 - INFO - mycodo.daemon - Received command to
terminate daemon
2019-09-20 03:00:36,647 - INFO - mycodo.daemon - All LCD controllers stopped
2019-09-20 03:00:36,648 - INFO - mycodo.daemon - All PID controllers stopped
2019-09-20 03:00:36,655 - INFO - mycodo.daemon - All Math controllers
stopped
2019-09-20 03:00:36,656 - INFO - mycodo.daemon - All Input controllers
stopped
2019-09-20 03:00:36,656 - INFO - mycodo.daemon - All Trigger controllers
stopped
2019-09-20 03:00:36,657 - INFO - mycodo.daemon - All Conditional
controllers stopped
2019-09-20 03:00:36,691 - INFO - mycodo.controllers.controller_output -
Deactivated in 34.1 ms
2019-09-20 03:00:36,692 - INFO - mycodo.daemon - Mycodo terminated in 0.892
seconds

2019-09-20 03:01:06,462 - INFO - mycodo.daemon - Mycodo daemon v7.7.6
starting
2019-09-20 03:01:09,307 - INFO - mycodo.controllers.controller_output -
Activated in 2074.1 ms
2019-09-20 03:01:09,809 - INFO - mycodo.daemon - All activated Conditional
controllers started
2019-09-20 03:01:09,810 - INFO - mycodo.daemon - All activated Trigger
controllers started
2019-09-20 03:01:10,778 - ERROR -
mycodo.controllers.controller_input_40c2eb22 - Deactivated unexpectedly
2019-09-20 03:03:19,371 - ERROR -
mycodo.controllers.controller_input_40c2eb22 - Deactivated unexpectedly

kizniche, 2019-09-20:
Is that the only thing causing you issues in the latest version of Mycodo?

kizniche, 2019-09-20:
Does any other code read measurements from the sensor so we know it's wired properly and communicating with the Pi?

cbaymac, 2019-09-20:
Kyle,   I’m really not sure.  It works with the Stretch OS and 7.7.1

kizniche, 2019-09-20:
Is your sensor detected on the /info page where the i2c-detect output is listed? If not, you may not have i2c enabled.

kizniche, 2019-09-20:
I suspect you might not have enabled the i2c interface after installing Buster. In any case, the Adafruit_BME280 library that's in use has been deprecated and is no longer maintained. Therefore, I have added a new Input Module that uses the RPi.bme280 library that has active development. This new input has not been released yet, but you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to try the new input/library. After upgrading, you should see two different BME280 Inputs to choose from, one that uses the Adafruit_BME280 library, and the other the RPi.bme280 library. Test out the one that uses RPi.bme280 and see if it works with your sensor.

kizniche, 2019-09-20:
Actually, hold off on upgrading to master while I fix an Input issue. I'll update you.

kizniche, 2019-09-20:
Okay. Found and fixed the issue on master. Should be safe to upgrade to.

cbaymac, 2019-09-20:
Kyle,
 I am out of the office until Monday, but will give it a try once I get
back. You are correct however I was not enabling i2C on the RPi
configuration.  Can I back up the current working Stretch and import that
backup into Buster?  Just want an easy way to save my configurations,
outputs, functions, methods, etc? Thank you and have a great weekend. The
mushrooms are going crazy with this cooler weather.

kizniche, 2019-09-20:
Yes. You can copy the database from /var/Mycodo-backups/[]/databases/mycodo.db to ~/Mycodo/databases/ then run:

```
cd ~/Mycodo/databases
sudo ../env/bin/alembic upgrade head
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-alembic-post
sudo service mycodo restart
```

That will upgrade your database from any previous Mycodo version to the one of the system you just copied it to.

cbaymac, 2019-09-25:
Kyle,
  Thank you for the update and tip on updating my configuration. Both
worked perfectly. I am not sure if the whole problem from the start was me
not setting the iC2 in raspi-config from the start. Having said that, when
I copied the config files to the new disk, the BME280 was there but not
working. I deleted it, reinstalled,  and the dependencies were then
installed and everything now works perfect with a fresh install of Buster.
Thank you. My mushrooms are doing great with the PID controller cycling
twice a day on the Sine Wave method. I swing the humidity around 85% up to
95 and down to 75.  The mushrooms seem to like the change in humidity. I am
controlling my humidity with a House of Hydro three ring mister all
controlled by the RPi. I also control the CO2 with a COZIR M19 sensor and a
exhaust fan.

kizniche, 2019-09-26:
Great! Thanks for the update.
You found out if you install fresh and copy a configuration database in that has Inputs that have dependencies, you'll need to install those dependencies before the Inputs will work again. I may develop a system that detects if a new database was installed and check if the dependencies for all the Input are installed, and notifies the user to install them. I've had this issue, but it's not always easy to ensure you've installed them all unless you just add all the Inputs already installed to go through the dependency check.

closed 2019-09-27 by kizniche