kizniche/Mycodo #695: SHT3x Reset

kizniche, 2019-09-20:
Add software reset to SHT3x input code

Ref: https://github.com/kizniche/Mycodo/issues/692#issuecomment-533349295

kizniche, 2019-09-20:
@chitters2004 I added the reset code to the SHT31 Input currently on master. [Upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if the reset fixes the sensor following an error.

kizniche, 2019-09-20:
Code for reference:

```python
        try:
            temperature, humidity = self.sensor.read_temperature_humidity()
        except OSError as e:
            self.logger.debug("OSError: {}".format(e))
            self.logger.debug("Attempting reset of sensor and reread of measurements")
            self.sensor.reset()
            try:
                temperature, humidity = self.sensor.read_temperature_humidity()
            except Exception as e:
                self.logger.exception("Could not read measurements after reset attempt")
                return
```

kizniche, 2019-09-20:
Enable debug logging for the Input to see the debug messages in the new code.

chitters2004, 2019-09-20:
Thank you. Giving it a go now. Will report back.

chitters2004, 2019-09-20:
All appears to be working as intended.

2019-09-20 21:47:36,690 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 28.92004272526131, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 74.4914930952926, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.92595914327346, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1016.8425722507245, 'timestamp_utc': None}}
2019-09-20 21:47:41,626 - DEBUG - mycodo.inputs.sht31_3fbba05e - OSError: [Errno 121] Remote I/O error
2019-09-20 21:47:41,627 - DEBUG - mycodo.inputs.sht31_3fbba05e - Attempting reset of sensor and reread of measurements
2019-09-20 21:47:41,715 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 21:47:46,702 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 21:47:51,685 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 21:47:56,667 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 28.765163653009836, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 75.05607690547036, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.902328931231686, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 985.4665500753744, 'timestamp_utc': None}}
2019-09-20 21:48:01,656 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 21:48:06,638 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 21:48:10,805 - INFO - mycodo.controllers.controller_pid_fb038d11 - PID Settings: Device ID: 3fbba05e-600a-4b34-8473-64becee09059, Measurement ID: a29a2437-25e6-4f34-a857-f41bb4427de6, Direction: both, Period: 20.0, Setpoint: 28.0, Band: 0.5, Kp: 3.0, Ki: 0.1, Kd: 0.0, Integrator Min: -250.0, Integrator Max 250.0, Output Raise: aef5eaa4-ee9f-4a6c-86bd-b9abf45d48bf, Output Raise Min On: 0.0, Output Raise Max On: 0.0, Output Raise Min Off: 0.0, Output Lower: d27327b3-c76a-4640-85b5-0cbf43441812, Output Lower Min On: 2.5, Output Lower Max On: 0.0, Output Lower Min Off: 0.0, Setpoint Tracking: 
2019-09-20 21:48:10,810 - INFO - mycodo.controllers.controller_pid_fb038d11 - Activated in 340.2 ms
2019-09-20 21:48:10,826 - INFO - mycodo.controllers.controller_pid_50e0c5a0 - PID Settings: Device ID: 3fbba05e-600a-4b34-8473-64becee09059, Measurement ID: 733fbed8-6ccc-4283-a97c-b2e8b5b299ca, Direction: both, Period: 20.0, Setpoint: 80.0, Band: 1.0, Kp: 0.5, Ki: 0.05, Kd: 0.0, Integrator Min: -250.0, Integrator Max 250.0, Output Raise: 5591c69f-393f-4ae3-9d05-f11173a27f78, Output Raise Min On: 0.0, Output Raise Max On: 0.0, Output Raise Min Off: 0.0, Output Lower: d27327b3-c76a-4640-85b5-0cbf43441812, Output Lower Min On: 1.5, Output Lower Max On: 0.0, Output Lower Min Off: 0.0, Setpoint Tracking: 
2019-09-20 21:48:10,827 - INFO - mycodo.controllers.controller_pid_50e0c5a0 - Activated in 325.6 ms
[... cut ...]
2019-09-20 21:48:41,648 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}

kizniche, 2019-09-20:
It's strange how often there are measurements that return no values. It's even more common than the errors. Do you happen to have your sensor on long wires or have connections that may be causing instability?

chitters2004, 2019-09-20:
Hmm, I wasn't actually paying much attention and didn't notice that. That's new i think... Will switch input back and check.

kizniche, 2019-09-20:
I just updated the Input to check for NaNs if you want to upgrade to master and try it out.

chitters2004, 2019-09-20:
I realised that I forgot to re enable the sensor power relay after upgrade...

Sensor is about 50cm away from unit with a couple of grove connectors.

Still getting some though. Will upgrade and test.

2019-09-20 22:03:55,152 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.08285648889907, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.13939116502632, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.153406923839974, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 747.2054892756191, 'timestamp_utc': None}}
2019-09-20 22:04:00,071 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 22:04:05,092 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.064164187075605, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.22484168764782, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.153114530089965, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 743.3296206406873, 'timestamp_utc': None}}
2019-09-20 22:04:10,075 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 22:04:15,099 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': nan, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': nan, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': nan, 'timestamp_utc': None}}
2019-09-20 22:04:20,123 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.037460898756393, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.25688563363089, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.133870622227306, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 741.0217840424006, 'timestamp_utc': None}}
2019-09-20 22:04:25,176 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.037460898756393, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.18822003509575, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.119527877209627, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 743.4747765598279, 'timestamp_utc': None}}
2019-09-20 22:04:30,360 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.050812542916006, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.2111085679408, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.137277648810393, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 743.2388517638697, 'timestamp_utc': None}}
2019-09-20 22:04:35,070 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.037460898756393, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.17906462195774, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 23.117614688411837, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 743.8018422288183, 'timestamp_utc': None}}
2019-09-20 22:04:40,088 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 27.037460898756393, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.15770199130236, 'times [... cut ...]
2019-09-20 22:09:05,132 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.457999542229345, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 80.48981460288395, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.825633788527618, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 673.6381499813949, 'timestamp_utc': None}}

chitters2004, 2019-09-20:
`2019-09-20 22:17:28,482 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.694285496299685, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.04165712977799, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.396236498696496, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 592.7026553750214, 'timestamp_utc': None}}
2019-09-20 22:17:33,405 - DEBUG - mycodo.inputs.sht31_3fbba05e - OSError: [Errno 121] Remote I/O error
2019-09-20 22:17:33,406 - DEBUG - mycodo.inputs.sht31_3fbba05e - Attempting reset of sensor and reread of measurements
2019-09-20 22:17:33,480 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.680933852140072, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.04013122758832, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.382904394608378, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 592.2842671724422, 'timestamp_utc': None}}
2019-09-20 22:17:38,473 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.664911879148548, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 81.99588006408788, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.358396248964837, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 593.18013338702, 'timestamp_utc': None}}
2019-09-20 22:17:43,454 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.664911879148548, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.07370107576105, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.3740045683662, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 590.6161714578103, 'timestamp_utc': None}}
2019-09-20 22:17:48,445 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.651560234988935, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.14999618524453, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.37626605270014, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 587.6373075250298, 'timestamp_utc': None}}
2019-09-20 22:17:53,431 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.638208590829322, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.20492866407264, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.37423654734181, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 585.3654591586809, 'timestamp_utc': None}}
2019-09-20 22:17:58,353 - DEBUG - mycodo.inputs.sht31_3fbba05e - OSError: [Errno 121] Remote I/O error
2019-09-20 22:17:58,353 - DEBUG - mycodo.inputs.sht31_3fbba05e - Attempting reset of sensor and reread of measurements
2019-09-20 22:17:58,431 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.638208590829322, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.23392080567636, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.380038831091532, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 584.4117682202472, 'timestamp_utc': None}}
2019-09-20 22:18:03,410 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.651560234988935, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.22781719691768, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.39184719995378, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 585.0753736317338, 'timestamp_utc': None}}
2019-09-20 22:18:08,489 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.651560234988935, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.23849851224537, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.393984780470763, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit' [... cut ...]
2019-09-20 22:19:33,480 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.56343938353551, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.31479362172885, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.323238212087357, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 579.1781821693734, 'timestamp_utc': None}}
`

chitters2004, 2019-09-20:
How do I format the code like that?!

kizniche, 2019-09-20:
It's 3 backtics instead of one

chitters2004, 2019-09-20:
Ah, thanks!

kizniche, 2019-09-20:
Measurements look pretty reliable now.

chitters2004, 2019-09-20:
Much better.

chitters2004, 2019-09-20:
```2019-09-20 22:18:58,461 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.582131685358974, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.26901655603876, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.33233171781736, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 581.32140251271, 'timestamp_utc': None}}
2019-09-20 22:19:03,459 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.595483329518572, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.33920805676357, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.359392909297323, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 579.4787267873418, 'timestamp_utc': None}}
2019-09-20 22:19:08,436 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.595483329518572, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.35141527428092, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.36183193758286, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 579.0781885279415, 'timestamp_utc': None}}
2019-09-20 22:19:13,423 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.582131685358974, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.38498512245366, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.355504729172484, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 577.5193003964479, 'timestamp_utc': None}}
2019-09-20 22:19:18,412 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.582131685358974, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.34683756771192, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.347885183814455, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 578.7699918820605, 'timestamp_utc': None}}
2019-09-20 22:19:23,505 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.56343938353551, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.32089723048753, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.32445779528051, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 578.978293236787, 'timestamp_utc': None}}
2019-09-20 22:19:28,500 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.582131685358974, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.31631952391852, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.341787320747077, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 579.7705450705505, 'timestamp_utc': None}}
2019-09-20 22:19:33,480 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.56343938353551, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.31479362172885, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.323238212087357, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 579.1781821693734, 'timestamp_utc': None}}
2019-09-20 22:19:38,453 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.56343938353551, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.38040741588465, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.336344581694036, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 577.0293761440686, 'timestamp_utc': None}}
2019-09-20 22:19:43,385 - DEBUG - mycodo.inputs.sht31_3fbba05e - At least one measurement is not a number: Temperature: nan, Humidity: nan
2019-09-20 22:19:48,462 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.595483329518572, ' [... cut ...]
2019-09-20 22:22:33,486 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.424582284275573, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 82.41245136186771, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.207193101175385, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 571.2535507571057, 'timestamp_utc': None}}
```

kizniche, 2019-09-20:
I just updated the Input code to make three attempts and improved the logic. Can you test to make sure I didn't break anything?

chitters2004, 2019-09-20:
Will do.

chitters2004, 2019-09-20:
```2019-09-20 22:34:53,718 - DEBUG - mycodo.inputs.sht31_3fbba05e - OSError: [Errno 121] Remote I/O error
2019-09-20 22:34:53,718 - DEBUG - mycodo.inputs.sht31_3fbba05e - Attempting reset of sensor and another measurement
2019-09-20 22:34:53,792 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.316472114137483, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.55138475623713, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.494608613530286, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 700.1749560744382, 'timestamp_utc': None}}
2019-09-20 22:34:58,761 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.244373235675596, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.60173952849622, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.434956830577814, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 695.4866958231826, 'timestamp_utc': None}}
2019-09-20 22:35:03,743 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.273746852826733, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.60021362630656, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.463189827137096, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 696.745073605395, 'timestamp_utc': None}}
2019-09-20 22:35:08,833 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.26039520866712, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.65514610513466, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.46157158480425, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 694.3217904673573, 'timestamp_utc': None}}
2019-09-20 22:35:13,729 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.273746852826733, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.6383611810483, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.471079080469927, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 695.442161881247, 'timestamp_utc': None}}
2019-09-20 22:35:18,815 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.244373235675596, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.65056839856565, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.44505214133731, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 693.8218563358739, 'timestamp_utc': None}}
2019-09-20 22:35:23,802 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.273746852826733, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.6383611810483, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.471079080469927, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 695.442161881247, 'timestamp_utc': None}}
2019-09-20 22:35:28,818 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.244373235675596, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.65056839856565, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.44505214133731, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 693.8218563358739, 'timestamp_utc': None}}
2019-09-20 22:35:33,819 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.231021591515983, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.69634546425574, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.441532982174312, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 691.715910808826, 'timestamp_utc': None}}
[... cut ...]
2019-09-20 22:36:18,734 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.231021591515983, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.61547264820325, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.4248197281413, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 694.4711297531269, 'timestamp_utc': None}}
```

kizniche, 2019-09-20:
Fixed that. Should work perfectly now.

chitters2004, 2019-09-20:
Upgrading now

chitters2004, 2019-09-20:
```2019-09-20 22:44:28,817 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.783779659723805, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.31540398260472, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.690219079330564, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 763.2040971451022, 'timestamp_utc': None}}
2019-09-20 22:44:33,709 - DEBUG - mycodo.inputs.sht31_3fbba05e - At least one measurement is not a number: Temperature: nan, Humidity: nan
2019-09-20 22:44:33,878 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.770428015564207, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.32455939574274, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.679197249162925, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 762.2835442933875, 'timestamp_utc': None}}
2019-09-20 22:44:38,763 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.75440604257267, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.37796597238118, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.674898126481526, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 759.6896588657074, 'timestamp_utc': None}}
2019-09-20 22:44:43,750 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.75440604257267, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.39780270084688, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.679071570733942, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 758.9926958759223, 'timestamp_utc': None}}
2019-09-20 22:44:48,741 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.71168078126192, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.4069581139849, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.63955213600229, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 756.7679667476226, 'timestamp_utc': None}}
2019-09-20 22:44:53,759 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.71168078126192, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.49698634317541, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.658473697985034, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 753.612761459084, 'timestamp_utc': None}}
2019-09-20 22:44:58,741 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.71168078126192, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.64499885557336, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.689540753697795, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 748.4253900525036, 'timestamp_utc': None}}
2019-09-20 22:45:03,826 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.684977492942707, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.61753261615931, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.65786755284134, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 748.2122777641074, 'timestamp_utc': None}}
2019-09-20 22:45:08,806 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.71168078126192, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 78.67246509498742, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.695300140059764, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 747.4627850492208, 'timestamp_utc': None}}
2019-09-20 22:45:13,806 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.698329137102306, 't [... cut ...]
2019-09-20 22:46:13,801 - DEBUG - mycodo.controllers.controller_input_3fbba05e - Adding measurements to InfluxDB with ID 3fbba05e-600a-4b34-8473-64becee09059: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 26.586175326161595, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 79.11650263218128, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 22.666283711197206, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 726.5173833045762, 'timestamp_utc': None}}
```

kizniche, 2019-09-20:
Nice. Looks like the 3-attempt loop gets a measurement every measurement interval. If it fails 3 times ina row, no measurement for that interval will be recorded, but 3 failures in a row seems rare.

kizniche, 2019-09-20:
That's about as robust as I can think to make it.

chitters2004, 2019-09-20:
Working so much better now, great job.

kizniche, 2019-09-20:
You're welcome. We can close this issue. I'll make a release tonight so other SHT3x users can reap the benefits as well.

closed 2019-09-20 by kizniche

chitters2004, 2019-09-20:
Might be worth renaming the input at this point if you haven't already.

kizniche, 2019-09-20:
It's now named "SHT3x (30, 31, 35)" instead of "SHT31"

chitters2004, 2019-09-20:
Perfect

kizniche, 2019-09-20:
I'll need to update the manual, though.

chitters2004, 2019-09-20:
Have tried to do so, no idea what i'm doing though!

kizniche, 2019-09-20:
Your pull request went through. I merged it. Thanks.

kizniche, 2019-09-20:
You're now officially a [contributor](https://github.com/kizniche/Mycodo/graphs/contributors) :wink:

kizniche, 2019-09-20:
I have scripts that do this: https://github.com/kizniche/Mycodo/commit/ed2da82d4a12840ffbe7bce41b1e5fe0b3d84cf5

Otherwise it would be a ridiculous ordeal to update all the manual versions and translations: **Showing with 2,012 additions and 1,951 deletions**

chitters2004, 2019-09-20:
Ah nice, that makes sense. Just need to learn python now and actually contribute!

kizniche, 2019-09-20:
I'd recommend using PyCharm, if you can. It's free if you're a student or educator. a good IDE makes Python development so much more enjoyable. I started using it over Sublime text editor and never looked back.

chitters2004, 2019-09-20:
Thanks for the tip!

chitters2004, 2019-09-20:
Is my incorrect database caused by the manual updates?

kizniche, 2019-09-20:
Yes. That's not an issue. It will correct itself when I make the next release and you upgrade by the normal method.