kizniche/Mycodo #697: Incorrect output duration on dashboard graph

chitters2004, 2019-09-23:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.7.6-8
 
[image]

kizniche, 2019-09-23:
Can you provide steps to replicate the issue?

chitters2004, 2019-09-23:
I realise it's been this way since you fixed the outputs not appearing the other day.

Literally just set up a graph with the above inputs/outputs and this is the result.

kizniche, 2019-09-23:
I can't replicate your issue from those instructions. Delete your current Outputs and graph, add them, and again, and see if your issue persists. If so, please provide detailed steps that can be followed to reproduce the issue.

kizniche, 2019-09-23:
What is the correct duration?

chitters2004, 2019-09-23:
Have just deleted outputs and graph and redone, still getting the same issue. 

At the moment it seems to be sometimes doubling the PID output duration to give the heater output time on the graph. The PID output duration is what is actually happening in real life, the Output duration is incorrect on the graphs.

I'm not really sure how I can provide more detailed instructions i'm afraid....

1. Install Mycodo
2. Configure inputs
3. Configure outputs.
4. Configure PID
5. Create graph with Outputs, Input and PID Duration/%

[image]

kizniche, 2019-09-23:
Can you turn all other controllers off, then turn daemon debug mode on in the general configuration, debug mode on for the PID. Restart the daemon, and collect the first 10 minutes of logs after restarting the daemon and attach it as a txt file (don't paste)?

kizniche, 2019-09-24:
We're you changing PID settings while the PID controller was active?

kizniche, 2019-09-24:
I haven't been able to replicate what you're seeing. Let me know if you can provide the above logging lines. They will be invaluable for investigating the issue.

chitters2004, 2019-09-24:
I had in the past but seems to make no difference whether I make changes while active or not.

[log.txt](https://github.com/kizniche/Mycodo/files/3646434/log.txt)

chitters2004, 2019-09-24:
Is it just combining multiple output times together as graph resolution isn't high enough?

chitters2004, 2019-09-24:
I think it is. Just reduced the x axis down to 60 seconds and now appears to be displaying correctly
[image]
.

kizniche, 2019-09-24:
Try a different browser. I've never seen Highcharts mess up like that before.

kizniche, 2019-09-24:
Actually, it may be the aggregation setting. Though, I don't recall having this issue before, otherwise I would have likely provided the option to enable/disable aggregation in the graph options.

chitters2004, 2019-09-24:
It's doing it on edge too.

kizniche, 2019-09-24:
There will be data aggregation on long duration x-axis spans where data cannot fit. I've experienced it with edge detections/boolean values. I don't recall it happening with Outputs for PIDs as you've shown. Though, perhaps I have and just don't recall.

kizniche, 2019-09-24:
Ref: https://www.highcharts.com/docs/advanced-chart-features/data-grouping
Outputs use the column series type.

kizniche, 2019-09-24:
I can add an option to disable grouping. It will have to wait a few days until I return from traveling for work.

chitters2004, 2019-09-25:
No urgency but thank you.

kizniche, 2019-09-27:
Closed in 64e28686

closed 2019-09-27 by kizniche