kizniche/Mycodo #699: Feature  Request: Slider output control for desktop

SAM26K, 2019-09-29:
## Mycodo  Feature Request:
  - Specific Mycodo Version:7.78
 
Request a slide bar type output  control.
Reason:  Want to use it for a light dimmer.

kizniche, 2019-09-29:
For PWM Outputs?

SAM26K, 2019-09-29:
Yes.      Plan on using GPIO for PWM and a level shifter to boost to 10VDC.

kizniche, 2019-09-29:
And you're suggesting this be a Dashboard widget?

SAM26K, 2019-09-29:
Yes.  Would be more convient to use a slider vs discrete numbers or arrow button.

SAM26K, 2019-09-29:
Would be nice to have slider combined with number input for fine adjustment.

SAM26K, 2019-09-29:
Just started looking at the GPIO PWM signal with a Fluke 199 scope-meter and it looks like there might be some PW jitter in the signal.  Seems to be random but not sure if it's a scope triggering problem or real jitter.  Which library  does Mycodo use for GPIO PWM control?  TIA!

SAM26K, 2019-09-29:
Using a new 3B.

SAM26K, 2019-09-29:
Screen shot of the output control:  
[image]

kizniche, 2019-09-29:
Have you tried a dedicated hardware PWM pin?

kizniche, 2019-09-29:
Also, PWM is controlled by pigpio. You can change the sample rate of pigpio in the configuration menu.

SAM26K, 2019-09-29:
what is the difference between "any pin" and "hardware pin"?

SAM26K, 2019-09-29:
Changing Configure pigpiod from 1msec to 5msec helped a lot.  Now jitter is only around 2%.

kizniche, 2019-09-30:
Browsers have an integrated range selector input. I added a PWM Output widget with this. This is what it looks like in Firefox:

[image]

SAM26K, 2019-09-30:
Do I need to upgrade to master? Upgraded t0 the latest Mycoco rev and get this in Firefox: 
Earlier rev of Mycodo works ok (cert is bypassed not accepted).
[image]

SAM26K, 2019-09-30:
Also it looks like the output control is no longer compatible with chrome.

SAM26K, 2019-09-30:
Guess I have to open a new issue since you closed this one?

kizniche, 2019-09-30:
You can reopen this one if it's about the Output PWM Slider desktop widget. I'm not sure what your exact issue you have or the steps to replicate it. You'll need to provide more information.

SAM26K, 2019-09-30:
I don't see an option to re-open (using chrome).  Need some time to organize the details.  Thanks.

SAM26K, 2019-10-02:
Turned on the certificate error was unique to the new RPi3B and Firefox.  Got that working now but don't see the new slider feature.  Is there a Firefox plugin I need to install?  Have V7.7.9 installed.  TIA.

kizniche, 2019-10-02:
You can check the [changelog](https://github.com/kizniche/Mycodo/blob/master/CHANGELOG.md) to see the release status of changes. To test the code for 7.7.10 prior to release, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

SAM26K, 2019-10-03:
Upgraded to master and  got PWM (any pin) working but still haven't got hardware pin (12) working.
Also getting a db version error:

Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516]
**Database Version: 895ddcdef4ce (Incorrect Version. Should be 0ce53d526f13)
Daemon Status: Running
Daemon Process ID: 4524**
Daemon RAM Usage: 67.564 MB
Daemon Virtualenv: Yes
Frontend RAM Usage: 74.348 MB
Frontend Virtualenv: Yes

SAM26K, 2019-10-03:
Tried pin 18 PWM no luck with Hardware Pin library but works with any pin library.  That is the BCM pin 18 that maps to hardware pin 12 right? TIA,

SAM26K, 2019-10-03:
No errors in daemon log.  Output Control shows green and active but a scope on the pin shows 0V.If I select "any pin" it works but not with "hardware pin" selected.

SAM26K, 2019-10-03:
The new slide control works great but only with "any pin".  Thanks.

kizniche, 2019-10-03:
Working fine for me. Though, there may have been something fixed with the latest code you're not running (from master). Are you able to verify the PWM signal? Can you explain more by what you mean by it not working? It's very hard for me to figure out the cause of an issue without a good amount of information to investigate.

[image]

[image]

[image]

[image]

SAM26K, 2019-10-04:
Any ideas on how to fix this problem?  Complete re-install?

**Database Version: 895ddcdef4ce (Incorrect Version. Should be 0ce53d526f13)**

kizniche, 2019-10-04:
You can just install from master (git clone), then just restore one of your backups when you want to go back to a release.

kizniche, 2019-10-04:
To answer the question, there are a lot of potential reasons why the database version doesn't match the expected version. The upgrade log will give an idea, but if you're on master, it's too much work for me to diagnose such an issue that's likely resolved in the latest commit. Again, I'd recommend a git clone of master and install from that, and after you're done testing, use the restore page to go back to a stable release you have in the lab up list.

SAM26K, 2019-10-04:
How do I git clone from master?  I am hooked up as shown in your  pix above.  By not working I meant there is no square wave on the gpio BCM pin 18 using a scopemeter.

SAM26K, 2019-10-04:
upgrade log says "........successful." After upgrading to master.

SAM26K, 2019-10-04:
Having mixed results.  PWM  "Hardware Pin" 18 is working now and much more stable than "any pin".  Did the following:
1. Restored V7.7.8
2, Upgraded to release V7.7.9
At this; point PWM works fine and database version is green.
3. Upgraded to master.
Slider and PWM works great.  Database shows a dB mismatch.  No errors in the logs and things seem to work so far.
[image]

SAM26K, 2019-10-04:
I don't want to try the rework instructions for db errors in the database before you have a chance to suggest something else. TIA!

SAM26K, 2019-10-04:
upgrade log:
[2019-10-04 02:25:09] Upgrade script executed with the 'force-upgrade-master' argument. Upgrading from github repo master.
[2019-10-04 02:25:09] 
[2019-10-04 02:25:09] #### Upgrade to v7.7.9 initiated 2019-10-04_02-25-03 ####
[2019-10-04 02:25:09] 
[2019-10-04 02:25:09] #### Beginning Upgrade: Stage 1 of 2 ####
[2019-10-04 02:25:18] Stopping the Mycodo daemon...Done.
[2019-10-04 02:25:21] Downloading latest Mycodo version to /home/pi/Mycodo-master.tar.gz...Done.
[2019-10-04 02:25:21] Creating /tmp/Mycodo-7.7.9...Done.
[2019-10-04 02:25:22] Extracting /home/pi/Mycodo-master.tar.gz to /tmp/Mycodo-7.7.9...Done.
[2019-10-04 02:25:22] Removing /home/pi/Mycodo-master.tar.gz...Done.
[2019-10-04 02:25:22] Copying /tmp/Mycodo-7.7.9/.upgrade status file to /tmp/Mycodo-7.7.9...Done.
[2019-10-04 02:25:22] Moving env directory...Done.
[2019-10-04 02:25:22] Copying databases from /home/pi/Mycodo/databases/ to /tmp/Mycodo-7.7.9/databases...Done.
[2019-10-04 02:25:22] Copying flask_secret_key from /home/pi/Mycodo/databases/ to /tmp/Mycodo-7.7.9/databases...Done.
[2019-10-04 02:25:22] Copying statistics ID...Done.
[2019-10-04 02:25:22] Copying SSL certificates...Done.
[2019-10-04 02:25:22] Copying mycodo/inputs/custom_inputs...Done.
[2019-10-04 02:25:22] Copying mycodo/user_python_code...Done.
[2019-10-04 02:25:22] #### Stage 1 of 2 Complete ####
[2019-10-04 02:25:22] 
[2019-10-04 02:25:22] #### Continuing Upgrade: Stage 2 of 2 ####
[2019-10-04 02:25:22] 
[2019-10-04 02:25:22] Moving old Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2019-10-04_02-25-03-7.7.9...Done.
[2019-10-04 02:25:22] Moving new Mycodo from /tmp/Mycodo-7.7.9 to /home/pi/Mycodo...Done.
[2019-10-04 02:25:52] 
[2019-10-04 02:25:52] Running post-upgrade script...
[2019-10-04 02:25:52] 
[2019-10-04 02:25:52] #### Running initialization
[2019-10-04 02:25:52] 
[2019-10-04 02:25:52] #### Creating mycodo user
[2019-10-04 02:25:53] useradd: user 'mycodo' already exists
[2019-10-04 02:25:53] The user `mycodo' is already a member of `adm'.
[2019-10-04 02:25:53] The user `mycodo' is already a member of `dialout'.
[2019-10-04 02:25:53] The user `mycodo' is already a member of `gpio'.
[2019-10-04 02:25:53] The user `mycodo' is already a member of `i2c'.
[2019-10-04 02:25:54] The user `mycodo' is already a member of `kmem'.
[2019-10-04 02:25:54] The user `mycodo' is already a member of `video'.
[2019-10-04 02:25:54] The user `pi' is already a member of `mycodo'.
[2019-10-04 02:25:54] The user `mycodo' is already a member of `pi'.
[2019-10-04 02:25:54] 
[2019-10-04 02:25:54] #### Compiling mycodo_wrapper
[2019-10-04 02:25:57] 
[2019-10-04 02:25:57] #### Creating symlinks to Mycodo executables
[2019-10-04 02:25:57] 
[2019-10-04 02:25:57] #### Creating files and directories
[2019-10-04 02:25:58] 
[2019-10-04 02:25:58] #### Setting permissions
[2019-10-04 02:26:00] 
[2019-10-04 02:26:00] #### Removing statistics files
[2019-10-04 02:26:00] 
[2019-10-04 02:26:00] #### Checking if swap size is 100 MB and needs to be changed to 512 MB
[2019-10-04 02:26:00] #### Swap not currently set to 100 MB. Not changing.
[2019-10-04 02:26:01] 
[2019-10-04 02:26:01] #### Checking python 3 virtualenv
[2019-10-04 02:26:01] #### Virtualenv already exists, skipping creation
[2019-10-04 02:26:01] 
[2019-10-04 02:26:01] 
[2019-10-04 02:26:01] #### Updating apt repositories
[2019-10-04 02:26:01] Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
[2019-10-04 02:26:01] Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
[2019-10-04 02:26:07] Reading package lists...
[2019-10-04 02:26:08] 
[2019-10-04 02:26:08] #### Installing prerequisite apt packages and update pip
[2019-10-04 02:26:08] Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
[2019-10-04 02:26:08] Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
[2019-10-04 02:26:14] Reading package lists...
[2019-10-04 02:26:14] Reading package lists...
[2019-10-04 02:26:15] Building dependency tree...
[2019-10-04 02:26:15] Reading state information...
[... 184 lines cut ...]
[2019-10-04 02:28:45] compiling catalog mycodo_flask/translations/fr/LC_MESSAGES/messages.po to mycodo_flask/translations/fr/LC_MESSAGES/messages.mo
[2019-10-04 02:28:46] compiling catalog mycodo_flask/translations/zh/LC_MESSAGES/messages.po to mycodo_flask/translations/zh/LC_MESSAGES/messages.mo
[2019-10-04 02:28:47] 
[2019-10-04 02:28:47] #### Updating Mycodo restart monitor crontab entry
[2019-10-04 02:28:47] Removing crontab entry ...
[2019-10-04 02:28:47] Appending to crontab:
[2019-10-04 02:28:47] -----------------------------------------------
[2019-10-04 02:28:47] @reboot /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
[2019-10-04 02:28:47] -----------------------------------------------
[2019-10-04 02:28:47] To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove
[2019-10-04 02:28:47] 
[2019-10-04 02:28:47] 
[2019-10-04 02:28:47] #### Setting permissions
[2019-10-04 02:28:48] 
[2019-10-04 02:28:48] #### Restarting the Mycodo daemon
[2019-10-04 02:28:53] Mycodo previously shut down properly
[2019-10-04 02:29:05] 
[2019-10-04 02:29:05] #### Restarting nginx
[2019-10-04 02:29:11] #### Reloading mycodoflask
[2019-10-04 02:29:11] 
[2019-10-04 02:29:11] #### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
[2019-10-04 02:29:20] #### Successfully connected to http://localhost
[2019-10-04 02:29:20] Done.
[2019-10-04 02:29:20] 
[2019-10-04 02:29:20] Upgrade completed successfully without errors.

kizniche, 2019-10-04:
Ah, the mismatch is because I haven't yet updated config.py to reflect the latest DB version. This is expected behavior.

SAM26K, 2019-10-04:
Updated  the config.py file with the expected dB name and mismatch is gone now.

Since this control  will control grow light intensity, I want it to conserve the PWM setting across reboots in case of a power interrupt.  Is there anyway to update the Mycodo code to do that?
Thanks again.

kizniche, 2019-10-04:
What is setting the PWM? If it's a Trigger or Conditional, then it should be set back to whatever duty cycle it was at shortly after the daemon starts back up after a reboot.

SAM26K, 2019-10-05:
First attempt is just a GPIO PWM output control controlled by the new PWM dashboard widget and works well.  Looking at the manual but so far haven't figured out how to preserve PWM after a reboot without writing a python or linux code control.

kizniche, 2019-10-05:
I assume if it's growing something, you're going to want it on a regular, automatic cycle. The Trigger (daily time) can accomplish this. A Conditional can as well, but is more complex (and power) and requires experience writing Python.

SAM26K, 2019-10-05:
Basically I'm using a mechanical timer to turn the lights on and off and using the rpi and Mycodo only for dimming.  Want to preserve what ever the slider value on the dashboard is set to be restored for the PWM output across reboots.

SAM26K, 2019-10-05:
Is it possible to somehow make the PWM control static so the last setting is restored after a reboot?

SAM26K, 2019-10-05:
Is there a get-measurement command to return PWM duty cycle?

SAM26K, 2019-10-06:
Tried this control to return %duty cycle but it always returns 0's and turns the GPIO off after every measurement:
[image]

SAM26K, 2019-10-06:
errors:

2019-10-05 18:57:45,420 - DEBUG - mycodo.controllers.controller_input_516513a3 - Adding measurements to InfluxDB with ID 516513a3-2650-4866-8faa-da072926b386: {0: {'measurement': 'frequency', 'unit': 'Hz', 'timestamp_utc': None, 'value': 0.0}, 1: {'measurement': 'pulse_width', 'unit': 'us', 'timestamp_utc': None, 'value': 0.0}, 2: {'measurement': 'duty_cycle', 'unit': 'percent', 'timestamp_utc': None, 'value': 0.0}}
2019-10-05 18:58:45,259 - ERROR - mycodo.inputs.rpi_signal_pwm_516513a3 - InputModule raised an exception when taking a reading: unsupported operand type(s) for /: 'float' and 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 122, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/rpi_signal_pwm.py", line 87, in get_measurement
    self.value_set(2, read_pwm.duty_cycle())
  File "/home/pi/Mycodo/mycodo/inputs/rpi_signal_pwm.py", line 173, in duty_cycle
    return 100.0 * self._high / self._period
TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'
2019-10-05 18:59:45,327 - DEBUG - mycodo.controllers.controller_input_516513a3 - Adding measurements to InfluxDB with ID 516513a3-2650-4866-8faa-da072926b386: {0: {'measurement': 'frequency', 'unit': 'Hz', 'timestamp_utc': None, 'value': 0.0}, 1: {'measurement': 'pulse_width', 'unit': 'us', 'timestamp_utc': None, 'value': 0.0}, 2: {'measurement': 'duty_cycle', 'unit': 'percent', 'timestamp_utc': None, 'value': 0.0}}

kizniche, 2019-10-06:
The PWM Input is for detecting a PWM signal and writing it to the measurement to the database. You can't both generate and detect a PWM signal on the same pin. You could connect the pin generating the signal to another pin that is measuring the PWM signal, but that's unnecessary because the PWM signal being generated is stored in the database when the duty cycle is changed. So, you're attempting to measure something that's already stored in the database.

For solving your problem of desiring to set the last known duty cycle of a PWM output back to that duty cycle after a reboot, it's not an easy task. It could be done with a Conditional or Trigger, but that would work best if it was also defining the timer for the light itself.

I improved the Output startup/shutdown options, so PWM Outputs will now have the option to have the duty cycle at startup to be set to a user-specified value or the last known value.

SAM26K, 2019-10-07:
Trying to use terminal mycodo daemon commands to turn on GPIO PWM outputs and nothing works.
Tried:
1.  mycodo-client --dutycycle 44.0 73697cab-bfb3-40a8-b630-09dd49a50f2e
(returns error: unrecognized arguments:73697cab-bfb3-40a8-b630-09dd49a50f2e)

2.  mycodo-client --outputon 73697cab-bfb3-40a8-b630-09dd49a50f2e
(returns:  2019-10-06 19:48:09,786 [Remote command] Turn on output with ID '73697cab-bfb3-40a8-b630-09dd49a50f2e': Server returned:)
and output stays off.

kizniche, 2019-10-07:
The command should be mycodo_client.py --outputon ID --dutycycle DUTYCYCLE

SAM26K, 2019-10-07:
Still no preservation of settings after reboot:
[image]

SAM26K, 2019-10-07:
ie the PWM is low  inactive after reboot.

SAM26K, 2019-10-07:
This command does work:
mycodo-client --outputon aab93daf-8be1-494d-a547-c034849e2964 --dutycycle 44

progress :)

SAM26K, 2019-10-07:
PWM setting sticks after a daemon restart  but not after a system restart.

SAM26K, 2019-10-07:
Seems to be working now. :)

closed 2019-10-07 by kizniche

SAM26K, 2019-10-08:
Thanks very much for all your effort getting this working.

SAM26K, 2019-10-09:
There seems to be a problem with PWM values logged to the database.  Signal PWM looks fine on the scope but chart has randomly high duty cycles numbers.  Sometimes it reports the right duty cycle.  Would start a new issue but this is the unreleased master version.

[image]

kizniche, 2019-10-09:
When zoomed out, and data aggregation is not on, column-type series sum the values that are aggregated. To see the true values (at the potential expense of performance), data aggregation must be turned off for that series.

SAM26K, 2019-10-10:
How to "Turn off data aggregation for that series of the graph"?
Also previous PWM values are not always restored on reboot.

kizniche, 2019-10-10:
The graph options.
What does the daemon log show when the daemon starts?

kizniche, 2019-10-10:
The option is called Data Grouping

SAM26K, 2019-10-11:
Having more problems for some reason the TH-16s stopped feeding data to the new RPi.  They work fine with the other 2 RPi's (all 3Bs).  Also the slider settings is not updating when set with the numeric dashboard input.  Doesn't restore setting after reboot.  All this was working before.  Dashboard chart is not updating with PWM data even though the daemon log shows it is receiving good data from the TH16s and the data is not updating.
I did install Domoticz on the new RPi.  Do you think that could cause the issues?

SAM26K, 2019-10-11:
Daemon log:

2019-10-10 22:58:00,090 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.77778, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.8, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.968396014969484, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1725.2184313018543, 'timestamp_utc': None}}
2019-10-10 22:58:14,993 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'svalue': '76.5;44.8;1', 'idx': 74, 'RSSI': 9, 'Battery': 200, 'nvalue': 0, 'StatusSNS': {'AM2301': {'Temperature': 76.5, 'Humidity': 44.8}, 'TempUnit': 'F', 'Time': '2019-10-11T06:58:14'}}
2019-10-10 22:58:15,085 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.72222, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.8, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.918021634245735, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1719.5076844600787, 'timestamp_utc': None}}
2019-10-10 22:58:30,013 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.5, 'Humidity': 44.7}, 'TempUnit': 'F', 'Time': '2019-10-11T06:58:29'}}
2019-10-10 22:58:30,110 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.72222, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.7, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.884095635906446, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1722.6227346130859, 'timestamp_utc': None}}
2019-10-10 22:58:45,041 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.6, 'Humidity': 44.8}, 'TempUnit': 'F', 'Time': '2019-10-11T06:58:44'}}
2019-10-10 22:58:45,134 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.77778, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.8, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.968396014969484, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1725.2184313018543, 'timestamp_utc': None}}
2019-10-10 22:59:00,040 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.3, 'Humidity': 44.7}, 'TempUnit': 'F', 'Time': '2019-10-11T06:58:59'}}
2019-10-10 22:59:00,132 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.61111, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.7, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.783379739474787, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1711.2311009278524, 'timestamp_utc': None}}
2019-10-10 22:59:14,962 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.5, 'Humidity': 44.8}, 'TempUnit': 'F', 'Time': '2019-10-11T06:59:14'}}
2019-10-10 22:59:15,073 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.72222, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.8, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.918021634245735, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1719.5076844600787, 'timestamp_utc': None}}
2019-10-10 22:59:29,960 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.5, 'Humidity': 44.8}, 'TempUnit': 'F', 'Time': '2019-10-11T06:59:29'}}
2019-10-10 22:59:30,051 - DEBUG - mycodo.controllers.controller_input_66bb4168 - Adding measurements to InfluxDB with ID 66bb4168-9b27-43e6-a39b-1d66797c223b: {0: {'unit': 'C', 'measurement': 'temperature', 'value': 24.72222, 'timestamp_utc': None}, 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.8, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.918021634245735, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1719.5076844600787, 'timestamp_utc': None}}
2019-10-10 22:59:45,361 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.5, 'Humidity': 44.8}, 'TempUnit': 'F', 'Time': '2019-10-11T06:59:45'}}
[... cut ...] , 1: {'unit': 'percent', 'measurement': 'humidity', 'value': 44.8, 'timestamp_utc': None}, 2: {'unit': 'C', 'measurement': 'dewpoint', 'value': 11.918021634245735, 'timestamp_utc': None}, 3: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'value': 1719.5076844600787, 'timestamp_utc': None}}
2019-10-10 23:01:59,957 - DEBUG - mycodo.inputs.th1x_am2301_66bb4168 - Returned Data: {'StatusSNS': {'AM2301': {'Temperature': 76.6, 'Humidity': 44.8}, 'TempUnit': 'F', 'Time': '2019-10-11T07:01:59'}}

SAM26K, 2019-10-11:
Domoticz was causing problems.  After removing it things are working again.

[12 more replies]

commit 999c727e (2019-09-30): Add Dashboard Widget: Output PWM Range Slider (closes #699)