kizniche/Mycodo #700: Can only view admin pages after upgrade

chitters2004, 2019-10-16:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.8.1
 
#### Problem Description
Please list:

Raspi4

Updated from 7.7.8
### Errors

 `` -Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1249, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 580, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: dashboard.disable_data_grouping

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 467, in page_dashboard
    dashboard = Dashboard.query.all()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3186, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3342, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3367, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 988, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 287, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1107, in _execute_clauseelement
    distilled_params,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1253, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1473, in _handle_dbapi_exception
    util.raise_from_cause(sqlalchemy_exception, exc_info)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 398, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 152, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1249, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 580, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: dashboard.disable_data_grouping
[SQL: SELECT dashboard.id AS dashboard_id, dashboard.unique_id AS dashboard_unique_id, dashboard.graph_type AS dashboard_graph_type, dashboard.name AS dashboard_name, dashboard.refresh_duration AS dashboard_refresh_duration, dashboard.width AS dashboard_width, dashboard.height AS dashboard_height, dashboard.pid_ids AS dashboard_pid_ids, dashboard.output_ids AS dashboard_output_ids, dashboard.math_ids AS dashboard_math_ids, dashboard.note_tag_ids AS dashboard_note_tag_ids, dashboard.input_ids_measurements AS dashboard_input_ids_measurements, dashboard.x_axis_duration AS dashboard_x_axis_duration, dashboard.use_custom_colors AS dashboard_use_custom_colors, dashboard.enable_navbar AS dashboard_enable_navbar, dashboard.enable_rangeselect AS dashboard_enable_rangeselect, dashboard.enable_export AS dashboard_enable_export, dashboard.enable_title AS dashboard_enable_title, dashboard.enable_auto_refresh AS dashboard_enable_auto_refresh, dashboard.enable_xaxis_reset AS dashboard_enable_xaxis_reset, dashboard.enable_graph_shift AS dashboard_enable_graph_shift, dashboard.enable_manual_y_axis AS dashboard_enable_manual_y_axis, dashboard.enable_start_on_tick AS dashboard_enable_start_on_tick, dashboard.enable_end_on_tick AS dashboard_enable_end_on_tick, dashboard.enable_align_ticks AS dashboard_enable_align_ticks, dashboard.custom_yaxes AS dashboard_custom_yaxes, dashboard.decimal_places AS dashboard_decimal_places, dashboard.custom_colors AS dashboard_custom_colors, dashboard.disable_data_grouping AS dashboard_disable_data_grouping, dashboard.max_measure_age AS dashboard_max_measure_age, dashboard.range_colors AS dashboard_range_colors, dashboard.enable_timestamp AS dashboard_enable_timestamp, dashboard.y_axis_min AS dashboard_y_axis_min, dashboard.y_axis_max AS dashboard_y_axis_max, dashboard.option_invert AS dashboard_option_invert, dashboard.font_em_value AS dashboard_font_em_value, dashboard.font_em_timestamp AS dashboard_font_em_timestamp, dashboard.enable_output_controls AS dashboard_enable_output_controls, dashboard.show_pid_info AS dashboard_show_pid_info, dashboard.show_set_setpoint AS dashboard_show_set_setpoint, dashboard.camera_id AS dashboard_camera_id, dashboard.camera_image_type AS dashboard_camera_image_type, dashboard.camera_max_age AS dashboard_camera_max_age 
FROM dashboard]
(Background on this error at: http://sqlalche.me/e/e3q8)``

kizniche, 2019-10-16:
Looks like your database did not upgrade properly. Perhaps you upgraded to master in the last with this install. You can try restoring a previous version when the database was good, then attempt another upgrade to the latest release.

chitters2004, 2019-10-16:
That worked, thank you

closed 2019-10-16 by chitters2004