kizniche/Mycodo #701: Cannot Modify PID

chitters2004, 2019-10-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

 Trying to change PID settings. Input is a valid decimal. Happens even when no values changed and I save.
 
### Errors

Error: Error in the Max Age (seconds) field - Not a valid decimal value
Error: Error in the Max Age (seconds) field - Number must be between -1000000 and 1000000.
Error: Modify PID: Error

 
### Additional Notes
Nothing obvious in logs.
Also cursor and mouse position are offset on all input fields?!

kizniche, 2019-10-18:
I could replicate the Max Age error, and will make a bug-fix release shortly to fix that.

> Also cursor and mouse position are offset on all input fields?!

I'm not sure what you mean by this. Can you elaborate or take a screenshot if that will help explain?

chitters2004, 2019-10-18:
When I click where the red X is, the cursor appears where it is shown (about character spaces to the right). This is on every field in mycodo.

[Untitled](https://user-images.githubusercontent.com/55252868/67124920-80ade780-f1eb-11e9-89eb-0cc03ddca9d6.png)

chitters2004, 2019-10-18:
Also, would it be possible to add an option to change the white balance of the picam?

kizniche, 2019-10-18:
Please keep the discussion to the central issue. This makes it easier for me to track. I've made a release that fixes the initial issue. Make a new issue(s) for any others.

chitters2004, 2019-10-18:
Have upgraded but no change.

kizniche, 2019-10-18:
Strange. The dev system I was working on it fixed the issue, but my other dev system I just upgraded I could replicate the issue. Investigating further.

commit 2624c662 (2019-10-18): v7.8.4 bugfix release: Actually fix inability to save PID options (fixes #701)
closed 2019-10-18 by kizniche