kizniche/Mycodo #702: Cursor offset from mouse position

chitters2004, 2019-10-18:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.8.2

#### Problem Description

  When I click where the red X is, the cursor appears where it is shown (about character spaces to the right). This is on every field in mycodo.

[image]

kizniche, 2019-10-18:
What operating system, web browser and version?

chitters2004, 2019-10-18:
Windows 10, Version 77.0.3865.120 (Official Build) (64-bit)

kizniche, 2019-10-18:
Browser and version?

chitters2004, 2019-10-18:
Sorry, Chrome Version 77.0.3865.120 (Official Build) (64-bit)

chitters2004, 2019-10-18:
OS Build 18362.418

chitters2004, 2019-10-18:
Unfortunately the last issue fix didn't work but can't reopen the issue.

kizniche, 2019-10-18:
I was able to replicate this, and I traced it back to bootstrap's form-control class, specifically the padding attribute. However, I could not find this issue documented anywhere. I'll dig deeper later.

kizniche, 2020-01-08:
Can you please test the latest version (currently 8.2.2)? I believe updating the latest version of Bootstrap has fixed this issue. I can no longer replicate this issue on the latest version.

closed 2020-01-13 by kizniche