kizniche/Mycodo #703: Feature Request - White Balance for Picam

chitters2004, 2019-10-18:
Option to adjust the white balance of raspberry pi camera

kizniche, 2019-10-18:
Added but untested. If you would like to test, you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

References for options:

https://picamera.readthedocs.io/en/release-1.13/recipes1.html#capturing-consistent-images
https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera.PiCamera.awb_mode
https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera.PiCamera.awb_gains

chitters2004, 2019-10-18:
Thank you Kyle, Will give it a try.

chitters2004, 2019-10-18:
Is there currently any way to change video resolution etc?

kizniche, 2019-10-18:
Should be the width and height options.

chitters2004, 2019-10-18:
It appears to be stuck at default on my end.

[image]

kizniche, 2019-10-18:
I'll look into it. I also just realized I only applied the white balance settings to still photos, not video. I'll push an update shortly to test.

chitters2004, 2019-10-18:
You're a legend mate.

chitters2004, 2019-10-19:
At some point it would be great to be able to adjust ISO, Exposure mode, Shutter speed, Sharpness and Meter modes too....

kizniche, 2019-10-19:
I've had #313 open for a while. I'll push this up the priority list since I started it today.

chitters2004, 2019-10-19:
Oh awesome, I didn't notice that.

kizniche, 2019-10-19:
Added shutter speed, sharpness, iso, exposure mode, meter mode, and image effect

kizniche, 2019-10-19:
Untested, though. If you do test, please report any issues.

chitters2004, 2019-10-19:
Upgrading now for test.

chitters2004, 2019-10-19:
Appears to be working as intended. Will close and come back if anything odd pops up.

closed 2019-10-19 by chitters2004

kizniche, 2019-10-19:
I couldn't replicate your different photo/video resolution issue. Can you tell me what version Pi, version pi camera, and what resolution you were using?

chitters2004, 2019-10-21:
Pi 3B+, cam with 5 Megapixels OV5647 Sensor https://www.amazon.com/d/Networking-Products/MakerFocus-Wide-angle-Automatically-Day-Vision-Night-Vision/B07DPNDP6R, 2592x1944

kizniche, 2019-10-21:
So that is not the Pi camera?

kizniche, 2019-10-21:
If it's not an OEM RPi camera, I'm not sure what to tell you. Have you tried a lower resultion? Can you verify 1024x768 produces the same resulting image for still and stream?

chitters2004, 2019-10-21:
Having difficulties logging in remotely atm but will test as soon as i can. I assumed it would all work the same as it is the same sensor as on the official board.

chitters2004, 2019-10-21:
It seems that they correspond up until 1920x1080 and then the still image resolution increases while the stream zooms in. I wonder if it is possible to have a separate video/still resolution option to provide the best of both worlds?

kizniche, 2019-10-21:
Can you verify the same behavior persists after running the following and rebooting to ensure you have the latest firmware: ```sudo rpi-update```

1296x972 doesn't yield the same resolution with still and stream?