kizniche/Mycodo #704: Feature Request Add system clock to input list.

SAM26K, 2019-10-21:
## Mycodo Feature Request:
  - Specific Mycodo Version:
Mycodo Version: 7.8.4
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516]
Database Version: b6e332156961
Daemon Status: Running
Frontend RAM Usage: 55.104 MB
Frontend Virtualenv: Yes
 
#### Reason:
Want to use conditional to trigger depending on time of day
eg: Ramp a light dimmer from 0-100% for on cycle and ramp  from 100-0% for off cycle.
Needs time of day input to trigger conditional.  Thanks.

kizniche, 2019-10-21:
Why not use a Trigger: Timer (daily point or span)?

kizniche, 2019-10-21:
You can also use [time](https://docs.python.org/3/library/time.html) or [datetime](https://docs.python.org/2/library/datetime.html) in conditional statements.

SAM26K, 2019-10-22:
Trigger Timer Span works but no easy way to ramp the brightness.  Would like to use a linux or python output control to slowly ramp the light  up and down with a for loop.
Tried the following:  The bash file works fine in a terminal session but generates errors with Mycodo. Having similar  problems with python input control.

 Bash file:
`#!/bin/bash
out=$(date +"%T")
echo $out
exit 0`

Daemon error:
`2019-10-22 03:08:01,739 - ERROR - mycodo.controllers.controller_input_58b94ce3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.`

Input control:
[image]

SAM26K, 2019-10-22:
Also tried this python 3 data input control.  Works ok on RPi Python 3 RPi and returns 

> 2019-10-22 03:30:05.632968

but generates errors with Mycodo:
**Save control error:**
`************* Module mycodo.user_python_code.input_python_code_d1bba7fe-8437-4bf3-a7b9-c22a33a4fea9
R: 23, 4: Method could be a function (no-self-use)

**and daemon error:**
`2019-10-22 03:37:44,495 - ERROR - mycodo.controllers.controller_input_d1bba7fe - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.`

`
[image]

SAM26K, 2019-10-22:
Cleaned up python data input but getting a daemon type error:
`2019-10-22 04:42:40,881 - ERROR - mycodo.inputs.python_code_d1bba7fe - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/python_code.py", line 214, in get_measurement
    run.python_code_run()
  File "/home/pi/Mycodo/mycodo/user_python_code/input_python_code_d1bba7fe-8437-4bf3-a7b9-c22a33a4fea9.py", line 26, in python_code_run
    self.store_measurement(channel=0, measurement=time_channel_0)
  File "/home/pi/Mycodo/mycodo/user_python_code/input_python_code_d1bba7fe-8437-4bf3-a7b9-c22a33a4fea9.py", line 21, in store_measurement
    add_measurements_influxdb(self.input_id, measure)
  File "/var/mycodo-root/mycodo/utils/influx.py", line 53, in add_measurements_influxdb
    timestamp=timestamp))
  File "/var/mycodo-root/mycodo/utils/influx.py", line 87, in format_influxdb_data
    checked_value = float(value)
TypeError: float() argument must be a string or a number, not 'datetime.datetime'`
[image]

kizniche, 2019-10-22:
Creating an input is not the way to go, for several reasons. 1. In a time-series database, you can only store numerical values. 2. There is already a timestamp associated with the value in a time-series database (i.e. timestamping the current time with the current time is nonsensical. 3. Python has numerous ways of working with time (e.g. time, datetime).

This is the Python code for calculating the time as epoch in the future for a particular "HH:MM:SS" string given as a parameter:

https://github.com/kizniche/Mycodo/blob/31b0fa3aee66dda09e3fa91dbef445b3e8d9aec6/mycodo/utils/system_pi.py#L199-L212

To use this in your Conditional:

```python
from mycodo.utils.system_pi import epoch_of_next_time

time_str = "10:30:00"  # Desired time (string)
time_epoch = epoch_of_next_time(time_str)  # Desired future time (epoch)

if time_epoch < time.time():
    time_epoch = epoch_of_next_time(time_str)  # Update future epoch to tomorrow
    # code here to do something at 10:30 AM
```

kizniche, 2019-10-22:
If you want to go the Trigger route, I could add a "Ramp PWM" Action that would make the whole process simpler.

kizniche, 2019-10-22:
I just pushed code changes that add a PWM Ramp Action, among other things. You can upgrade to master to test. I would recommend using the Action with a Daily Time Point Trigger. If you do test, let me know how it goes.

SAM26K, 2019-10-23:
Ramp PWM would be great!  Thanks for the help.

kizniche, 2019-10-23:
The code to test has been up. You'll have to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to test.

SAM26K, 2019-10-25:
Seems to work really well if interval is set to 600 secs or more (0-100 or 100-0 %PWM) below that number the duty cycle gets noisy and occasionally jumps up and down.  Maybe if increment could be a user  parameter that could be  set higher than 0.1% it   would smooth out the shorter ramp times?
It's no big deal though and plan on ramp times greater than 900 secs which works fine.
Using single point daily trigger functions, one to ramp up the brightness and a 2nd one to ramp it down.  
[image]

kizniche, 2019-10-25:
I've never experienced the duty cycle jumping up and down, but I did find that short periods cause the PWM to be updated very fast, since it operates on 0.1 % steps. I will likely add the option to select either 1 % or 0.1 % steps, but for the initial implementation to test, I went with a default of 0.1 %. For most applications, this should be sufficient.

kizniche, 2019-10-25:
Just added the ability to select the duty cycle step size.

closed 2019-10-26 by kizniche

kizniche, 2019-10-27:
Minor revision released for upgrade