kizniche/Mycodo #708: No translation possible

grux77, 2019-11-01:
Hallo!
Since some upgrades, I can no longer change Web Interface to any other language. I even reinstalled mycodo without success. I think in 7.7.9 the translation still works. Then I immediately upgraded to 7.8.4 and it did not work anymore. In 7.9.1 neither. Only Chinese, Serbian, Swedish and Norwegian are working.
RPi3 B
Raspbian buster lite

And if it works again you should update the /home/pi/Mycodo/mycodo/mycodo_flask/translations/sv/LC_MESSAGES/messages.po  files. There are now many new terms without translation in the WebUI. Is this also possible with Notepad ++? Then I could update German.

kizniche, 2019-11-01:
For the translation issue, I was able to replicate the behavior.

For the pigpio issue, can you check if pigpiod is running? ```sudo service pigpiod status```

kizniche, 2019-11-01:
You can manually edit the messages.po file with Notepad++. You can make a pull request after committing the edit to your forked Mycodo repository, or just start an issue and attach the file (as txt) and I can add it. Make sure to use the latest po files I just added with the above commit, so you have the latest translated words/phrases.

grux77, 2019-11-02:
> For the pigpio issue, can you check if pigpiod is running? `sudo service pigpiod status`

Have everything again reinstalled - without success. Here the status:

pi@raspberrypi:~ $ sudo service pigpiod status
● pigpiod_low.service - pigpiod service (low latency)
   Loaded: loaded (/home/pi/Mycodo/install/pigpiod_low.service; enabled; vendor
   Active: activating (auto-restart) (Result: exit-code) since Sat 2019-11-02 15
  Process: 5248 ExecStart=/usr/bin/pigpiod -s 1 (code=exited, status=0/SUCCESS)
 Main PID: 5249 (code=exited, status=1/FAILURE)
lines 1-5/5 (END)

_sudo service pigpiod stop_  +  _sudo service pigpiod start_
or
_sudo service pigpiod restart_

Also brings no results

kizniche, 2019-11-02:
Can you send the whole log since the daemon restarted? Also, can you turn on daemon debugging from the general settings and restart the daemon to see debug log lines and paste that?

kizniche, 2019-11-02:
This is what pigpiod status should state:

```
pi@tango:~ $ sudo service pigpiod status
● pigpiod_high.service - pigpiod service (high latency)
   Loaded: loaded (/home/pi/Mycodo/install/pigpiod_high.service; enabled; ven   Active: active (running) since Tue 2019-10-29 18:17:05 EDT; 3 days ago
  Process: 328 ExecStart=/usr/bin/pigpiod -s 5 (code=exited, status=0/SUCCESS Main PID: 348 (pigpiod)
    Tasks: 6 (limit: 2077)
   Memory: 948.0K
   CGroup: /system.slice/pigpiod_high.service
           └─348 /usr/bin/pigpiod -s 5

Oct 29 18:17:04 tango systemd[1]: Starting pigpiod service (high latency)...
Oct 29 18:17:05 tango systemd[1]: Started pigpiod service (high latency).
lines 1-12/12 (END)
```

Notice your status says "activating" while mine says "active".

Can you uninstall pigpiod and reinstall it from the Mycodo configuration (Raspberry Pi) menu?

kizniche, 2019-11-02:
Also notice your log says "exited, failure." so pigpiod isn't running.

kizniche, 2019-11-02:
I will install the latest Raspbian Buster from an image and Mycodo and see if I can find any issues in the installer.

grux77, 2019-11-02:
Here is the log when reinstalling pigpio:

[mycododependency.log](https://github.com/kizniche/Mycodo/files/3800618/mycododependency.log)

And here debug-log:

[mycodo.log](https://github.com/kizniche/Mycodo/files/3800620/mycodo.log)

kizniche, 2019-11-02:
I just flashed Raspbian Buster (2019-09-26), booted, installed mycodo as per the README, set up a PWM output, and it worked.

kizniche, 2019-11-02:
pigpiod is running:

```
pi@raspberrypi:~ $ sudo service pigpiod status
● pigpiod_low.service - pigpiod service (low latency)
   Loaded: loaded (/home/pi/Mycodo/install/pigpiod_low.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-11-02 15:49:44 GMT; 5min ago
  Process: 11589 ExecStart=/usr/bin/pigpiod -s 1 (code=exited, status=0/SUCCESS)
 Main PID: 11590 (pigpiod)
    Tasks: 6 (limit: 2200)
   Memory: 620.0K
   CGroup: /system.slice/pigpiod_low.service
           └─11590 /usr/bin/pigpiod -s 1

Nov 02 15:49:44 raspberrypi systemd[1]: Starting pigpiod service (low latency)...
Nov 02 15:49:44 raspberrypi systemd[1]: Started pigpiod service (low latency).
```

kizniche, 2019-11-02:
These are the only commands I used after flashing Raspbian:

```
sudo apt update
sudo apt upgrade
curl -L https://raw.githubusercontent.com/kizniche/Mycodo/master/install/install | bash
```

after install, I created an admin user, added a PWM output (at which time it prompted to install pigpio, which I did), went back to the Outputs page and installed the PWM output, set duty cycle to 50% and it's on.

[image]

kizniche, 2019-11-02:
Your debug log seems to be missing the appropriate PWM pin initialization:

```
2019-11-02 16:20:29,281 - DEBUG - mycodo.daemon - Starting Output Controller
2019-11-02 16:20:29,388 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2019-11-02 16:20:29,498 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2019-11-02 16:20:29,499 - INFO - mycodo.controllers.controller_output - Activated in 216.6 ms
```

Whereas mine looks like this:

```
2019-11-02 16:06:39,775 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2019-11-02 16:06:39,955 - INFO - mycodo.controllers.controller_output - Output e29eda37 (PWM) setup on pin 10
2019-11-02 16:06:39,956 - DEBUG - mycodo.controllers.controller_output - e29eda37 (PWM (GPIO)) Initialized
2019-11-02 16:06:39,957 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2019-11-02 16:06:39,957 - INFO - mycodo.controllers.controller_output - Activated in 307.8 ms
```

Are you sure you have a pin set for your PWM output?

kizniche, 2019-11-02:
However, whether or not a pin is set, not having pigpiod running will ultimately prevent a PWM signal from being generated.

grux77, 2019-11-02:
> Are you sure you have a pin set for your PWM output?

I deleted it at the time

> These are the only commands I used after flashing Raspbian:
> 
> ```
> sudo apt update
> sudo apt upgrade
> curl -L https://raw.githubusercontent.com/kizniche/Mycodo/master/install/install | bash

I only did this

Tomorrow I will try to install on another RPi2 B. Maybe I have broken hardware.
I will contact you

kizniche, 2019-11-02:
Perhaps an issue with the SD card. Filesystem corruption/bad sectors can cause a host of software issues that may be misidentified as hardware issues. I would also use a different SD card for good measure.

grux77, 2019-11-02:
OK. I do that too

grux77, 2019-11-03:
Your guess was correct. The SD was it.
It was a Chinese 'No name' 16GB. Have now taken Lexar 8GB - no more problems.
Thanks for the hint!

kizniche, 2019-11-03:
So you've used both a different Pi and a different SD card and PWM still doesn't work?

kizniche, 2019-11-03:
Or are you saying it's strange that it didn't work with the old card and everything is working fine with the new card?

Coincidentally, #709 is also an SD card issue.

grux77, 2019-11-03:
> Or are you saying it's strange that it didn't work with the old card and everything is working fine with the new card?

Yes, that's what I mean

grux77, 2019-11-03:
Who buys cheap buys twice

kizniche, 2019-11-03:
> But it is strange that it always causes the same problem.

It likely had a bad sector that affected the same part of the filesystem when Raspbian was flashed.

closed 2019-11-03 by kizniche