kizniche/Mycodo #709: Influxdb IO kills SD card for 3-4 months.

cohe4ko, 2019-11-01:
I already killed 3 SD cards for 1 year period. 

### Versions:

Mycodo Version: 7.9.1
Python Version: 3.7.3 (default, Apr 3 2019, 05:39:12) [GCC 8.2.0]
Database Version: 8f9bf3fe5ec2
 - Raspberry Pi Version: [3B]
 - Raspbian OS Version: [Buster Lite]

### Reproducibility
[image]

I there some way to reduce SD card Usage?

kizniche, 2019-11-01:
Can you provide information about your setup? Specifically, what make/model SD card, what number of inputs and Periods for each, what other controllers (PIDs, Math, etc.) that save measurements to the the database and Periods. I can reply later with what SD cards I use and the study I found them from that tested multiple SD cards for reliability with many read/write events. I've been using Mycodo on multiple systems continually for 7 years and have only had 1 SD card failure.

cohe4ko, 2019-11-01:
Here is my previous db file that was on the system that died.
[mycodo.db.zip](https://github.com/kizniche/Mycodo/files/3798881/mycodo.db.zip)
I used kingston SD card for 16 GB.

kizniche, 2019-11-01:
What's the model and class of the SD cards that failed?

cohe4ko, 2019-11-01:
Exactly the same
[image]

kizniche, 2019-11-01:
The only card I ever had fail was one similar to the one pictured. I would suggest buying a higher quality name-brand card. The tests I mentioned previously actually looked at speed performance, not rewrite performance ([here](https://www.jeffgeerling.com/blog/2018/raspberry-pi-microsd-card-performance-comparison-2018) and [here](https://www.jeffgeerling.com/blog/2019/raspberry-pi-microsd-card-performance-comparison-2019)). The cards I've been buying are the Samsung Evo Plus ([$29 for 5 on Amazon](https://www.amazon.com/gp/product/B07NP96DX5/ref=ppx_yo_dt_b_asin_title_o01_s02?ie=UTF8&psc=1), making them $5.80 each).

SAM26K, 2019-11-11:
cohe4ko, What kind of case are you using?  I have heard of some low profile cases that have over heated the SD card.

cohe4ko, 2019-11-11:
None. My Raspberry pi is installed in a big box that has good ventilation.

SAM26K, 2019-11-11:
Most likely poor quality SD card as Kyle suggested then.

closed 2019-11-18 by kizniche