kizniche/Mycodo #710: Atlas EC calibration

RynFlutsch, 2019-11-06:
Hey,

you can add the calibration of atlas EC probes?

https://www.atlas-scientific.com/_files/_datasheets/_circuit/EC_EZO_Datasheet.pdf

Page 13

kizniche, 2019-11-10:
Sure. I may not get to this right away, but I have the framework for the pH sensor to work with, so it should be relatively smooth when I do start.

kizniche, 2019-11-18:
I just pushed a commit that adds calibration support for the Atlas Scientific EC sensor. If you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test, let me know if there are any issues. Note: You will need to be upgraded to >= 8.0.0 in order to upgrade to master, otherwise you could break your Mycodo. If you're on 7.x, upgrade to 7.10.0, then visit the upgrade page again to attempt an upgrade to 8.x. If you're running Raspbian Buster, you will be able to upgrade to 8.x, however if you're on an earlier version of Raspbian (e.g. Stretch), then you will not be able to upgrade to the latest Mycodo version and you will first need to flash Raspbian Buster and install Mycodo fresh before being able to upgrade to master.

kizniche, 2020-01-08:
Can we close this issue?

RynFlutsch, 2020-01-10:
Yes, if something is wrong i can reopen?

kizniche, 2020-01-10:
Of course. I just can't remember if after I implemented the feature it was able to be tested.

closed 2020-01-24 by kizniche