kizniche/Mycodo #711: Daemon stops running.

SAM26K, 2019-11-08:
Version:Mycodo Version: 7.9.1
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516]
Database Version: 8f9bf3fe5ec2
Daemon Status: Not Running
Frontend RAM Usage: 56.016 MB
Frontend Virtualenv: Yes

Daemon log:

raspberrypi3_3 - 7.9.1
11/07 16:00

    Data
    Setup
    More

Select the number of lines to display from the end of a log
Last 30 lines of /var/log/mycodo/mycodo.log:

2019-11-07 14:10:04,144 - DEBUG - mycodo.controllers.controller_input_2798d315 - Adding measurements to InfluxDB with ID 2798d315-03d0-4838-99b3-2a9ec0ab6c90: {0: {'value': 2.357536, 'timestamp_utc': None, 'unit': 'V', 'measurement': 'electrical_potential'}}
2019-11-07 14:10:05,899 - DEBUG - mycodo.controllers.controller_input_780f2bc5 - Adding measurements to InfluxDB with ID 780f2bc5-c093-43a3-8e12-af6ea46c00ef: {0: {'value': 2.409577, 'timestamp_utc': None, 'unit': 'V', 'measurement': 'electrical_potential'}}
2019-11-07 14:11:07,397 - DEBUG - mycodo.inputs.linux_command_2798d315 - Command being executed: /home/pi/moduser1/getA0.sh
2019-11-07 14:11:08,308 - DEBUG - mycodo.inputs.linux_command_780f2bc5 - Command being executed: /home/pi/moduser1/getA1.sh
2019-11-07 14:11:30,483 - DEBUG - mycodo.inputs.linux_command_2798d315 - Command returned: b'2.357356\n', Status: None, Error: 0
2019-11-07 14:11:31,307 - DEBUG - mycodo.inputs.linux_command_780f2bc5 - Command returned: b'2.409902\n', Status: None, Error: 0
2019-11-07 14:11:31,688 - DEBUG - mycodo.controllers.controller_input_2798d315 - Adding measurements to InfluxDB with ID 2798d315-03d0-4838-99b3-2a9ec0ab6c90: {0: {'value': 2.357356, 'timestamp_utc': None, 'unit': 'V', 'measurement': 'electrical_potential'}}
2019-11-07 14:11:32,368 - DEBUG - mycodo.controllers.controller_input_780f2bc5 - Adding measurements to InfluxDB with ID 780f2bc5-c093-43a3-8e12-af6ea46c00ef: {0: {'value': 2.409902, 'timestamp_utc': None, 'unit': 'V', 'measurement': 'electrical_potential'}}
2019-11-07 14:12:37,280 - DEBUG - mycodo.inputs.linux_command_2798d315 - Command being executed: /home/pi/moduser1/getA0.sh
2019-11-07 14:12:38,319 - DEBUG - mycodo.inputs.linux_command_780f2bc5 - Command being executed: /home/pi/moduser1/getA1.sh
2019-11-07 14:12:58,941 - DEBUG - mycodo.controllers.controller_math_e1d19e6a - Adding measurements to InfluxDB with ID e1d19e6a-8518-4427-bc43-37707bc3edca: {0: {'value': 10.96826662281353, 'unit': 'unitless', 'measurement': 'moisture'}}
2019-11-07 14:13:01,100 - DEBUG - mycodo.controllers.controller_math_2e4ef32c - Adding measurements to InfluxDB with ID 2e4ef32c-fc4c-4448-9e60-ca9692759711: {0: {'value': 149.05965315999998, 'unit': 'unitless', 'measurement': 'moisture'}}
2019-11-07 14:13:37,674 - DEBUG - mycodo.inputs.linux_command_780f2bc5 - Command returned: b'', Status: None, Error: -15
2019-11-07 14:13:37,676 - DEBUG - mycodo.inputs.linux_command_780f2bc5 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'b'''
2019-11-07 14:13:37,712 - DEBUG - mycodo.inputs.linux_command_2798d315 - Command returned: None, Status: None, Error: None
2019-11-07 14:13:37,713 - DEBUG - mycodo.inputs.linux_command_2798d315 - The command returned a non-numerical value. Ensure only one numerical value is returned by the command. Value returned: 'None'
2019-11-07 14:13:37,808 - ERROR - mycodo.inputs.python_code_3f220d5c - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/python_code.py", line 214, in get_measurement
    run.python_code_run()
  File "/home/pi/Mycodo/mycodo/user_python_code/input_python_code_3f220d5c-f6dc-42e3-9039-5687a2c90efa.py", line 53, in python_code_run
    stdin, stdout, stderr = client.exec_command(cmd)
  File "/usr/local/lib/python3.5/dist-packages/paramiko/client.py", line 508, in exec_command
    chan = self._transport.open_session(timeout=timeout)
  File "/usr/local/lib/python3.5/dist-packages/paramiko/transport.py", line 879, in open_session
    timeout=timeout,
  File "/usr/local/lib/python3.5/dist-packages/paramiko/transport.py", line 969, in open_channel
    raise SSHException("SSH session not active")
paramiko.ssh_exception.SSHException: SSH session not active
2019-11-07 14:13:38,151 - ERROR - mycodo.controllers.controller_input_3f220d5c - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2019-11-08:
Have you isolated what's causing your issue?

SAM26K, 2019-11-08:
Only that it happens on 2 out of 3 RPi3B's. The RPi 3B that doesn't have the problem has no software controllers to the other two RPis.

kizniche, 2019-11-08:
I mean have you narrowed down which Input is causing the issue?

SAM26K, 2019-11-08:
Not sure which would terminate the daemon but it always seems to be a software input that's hooked to another RPi that has the actual hardware. Suggestion on how to narrow it down?

SAM26K, 2019-11-08:
Just noticed ADS1256 software inputs were duplicated, hoping that's the problem.

kizniche, 2019-11-08:
How often does it occur? Can you only have one input at a time activated to determine what is causing the issue?

kizniche, 2019-11-08:
Can you apply the above changes and see if there's an exception that's causing issues?

SAM26K, 2019-11-08:
Seems to be working better after deleting duplicate software inputs.  Will try above now.

SAM26K, 2019-11-08:
Have not upgraded to master yet.  Something has changed with the python input control for a ADS1256 on a remote RPi and generates errors when control is saved.
I think one of the Mycodo updates changed it.  Seemed to be working though.

Errors on save:
`************* Module mycodo.user_python_code.input_python_code_3f220d5c-f6dc-42e3-9039-5687a2c90efa
R: 23, 4: Too many local variables (20/15) (too-many-locals)
W: 34, 8: Unused variable 'channel' (unused-variable)
W: 53,16: Unused variable 'stdin' (unused-variable)
W: 58,20: Unused variable 'timestamp' (unused-variable)
W: 53,31: Unused variable 'stderr' (unused-variable)

`
Code block:
`# coding=utf-8
import os
import paramiko

hostname = '192.168.1.17'
port = 22
username = 'pi'
password = 'passw'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'
channel = '0'
command = '/home/pi/Mycodo/env/bin/python ' \
          '/home/pi/Mycodo/mycodo/mycodo_client.py ' \
          '--get_measurement {uuid} {unit}'.format(
              uuid=uuid, unit=unit)
client = paramiko.SSHClient()

try:
    client.load_system_host_keys()
    client.set_missing_host_key_policy(paramiko.WarningPolicy)
    # For the next line to work, you must at least once have connected from the client to the host
    # to add the host to the list of known hosts.
    # To do this, run the following command from the client computer:
    # ssh user@hostIP
    # and when prompted "Are you sure you want to continue connecting?", choose "yes"
    client.load_host_keys(os.path.expanduser('~/.ssh/known_hosts'))
    client.connect(hostname, port=port, username=username, password=password)
    for x in range(0, 8):
        cmd = command + ' ' + str(x)
        stdin, stdout, stderr = client.exec_command(cmd)
        out = stdout.read().decode()
        if ';' in out and len(out.split(';')) == 3:
            status = out.split(';')[0]
            value = out.split(';')[1]
            timestamp = out.split(';')[2]
            if status == 'SUCCESS':
                self.store_measurement(channel=x, measurement=float(value))
finally:
    client.close()`

SAM26K, 2019-11-08:
indentation on the post above is messed up.

kizniche, 2019-11-08:
R is recommended and W is warning, which won't affect the function of the code.

You'll probably like the [new API](https://github.com/kizniche/Mycodo/blob/master/mycodo-api.rst#authentication) ([Manual](https://kizniche.github.io/Mycodo/mycodo-api.html)) I've been working on, which will allow you to more easily communicate between Mycodo instances. The above code can be replaced with the following:

```python
import json
import requests
ip_address = '127.0.0.1'
api_key = 'YOUR_API_KEY'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'
channel = '0'
endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
    unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
response = requests.get(url, headers=headers, verify=False)
response_dict = json.loads(response.text)
print("Time: {}, Value: {}".format(response_dict['time'], response_dict['value']))
```

kizniche, 2019-11-11:
Any daemon issue since removing the duplicate Input?

SAM26K, 2019-11-13:
Still having a problem with mycodo daemon stopping on some Rpis though less frequent.  Upgraded to latest release.

SAM26K, 2019-11-13:
seems to be related to software data inputs.

kizniche, 2019-11-13:
Are you able to upgrade your systems to 8.x to test using the API, instead of your current method, to acquire measurements from remote systems?

SAM26K, 2019-11-15:
Trying that now.  Backed up SD cards just in case.

SAM26K, 2019-11-15:
Upgrade to 8.0.2 seems to have stopped.  Do I really need to start with a newer RPi OS?
 ie start from scratch?

[2019-11-15 13:34:26] Incorrect Python version found. Mycodo requires Python >= 3.6.
[2019-11-15 13:34:26] If you're running Raspbian 9 (Stretch) with Python 3.5, you will need to install at least Raspbian 10 (Buster) with Python 3.7 to upgrade to the latest version of Mycodo.

SAM26K, 2019-11-15:
Trying this [workaround](https://github.com/instabot-py/instabot.py/wiki/Installing-Python-3.7-on-Raspberry-Pi):

kizniche, 2019-11-15:
Mycodo >= 8 won't work without Python >= 3.6

kizniche, 2019-11-15:
If you build from source, you will have to change the link of python3 from python3.5 to python3.7

kizniche, 2019-11-15:
You can export your settings database and measurements database, then import them once you install the new OS/Mycodo. There's currently a bug with importing measurements, but I hope to have that fixed in the next release.

SAM26K, 2019-11-16:
The 'export influxdb" button is causing a 504 timeout and the daemon to stop.  Is there another way to back them up?  Got Python3.7.2 installed and set default to 3.7.  Haven't tried another upgrade yet.

kizniche, 2019-11-16:
I've never received that error before exporting the influxdb. Can you start an issue so we can discuss it there?

As for the python version, there will be an issue with the virtual environment still using 3.5, and that will likely cause an error during the upgrade, if Mycodo permits the upgrade to proceed. In that case, you will need to delete the /env directory in ~/Mycodo after the upgrade errors after starting, then manually execute the upgrade_post.sh script to finish the upgrade and create the new virtualenv.

SAM26K, 2019-11-16:
Is there other database files I should save besides mycodo.db?

kizniche, 2019-11-16:
The database is backed up every time you upgrade to /var/Mycodo-backups, so there's no real need to back it up again unless you are installing Mycocod from scratch. If that's the case, you should use the settings database export/import feature, as this will ensure your settings database is upgraded to the latest version when it's imported.

kizniche, 2019-11-16:
If you aren't wiping the SD card, you don't need to back up the influxdb measurement database. So long as you use the same settings database, all references from Mycodo devices to measurements will be preserved.

SAM26K, 2019-11-16:
So if I delete Mycodo folder and fresh install of 8.x I  can retrieve settings from /var/Mycodo-backups?

kizniche, 2019-11-16:
Yes, they will be located at /var/Mycodo-backups/{find particular backup}/databases/mycodo.db

But, you will need to manually copy and issue the database upgrade command to use those databases. If you use the export/import feature, all of that is done for you through the web interface.

SAM26K, 2019-11-18:
I had to end up upgrading Raspian to 10.0 so now starting over.  Trying to retrieve settings with import settings that I saved to a pc but getting these errors:

    Error: Import Settings: Invalid file name: mycodo.db: mycodo != Mycodo.
    Error: Import Settings: Correct format is: Mycodo_MYCODOVERSION_Settings_DBVERSION_HOST_DATETIME.zip
    Error: An error occurred during the settingsdatabase import.

I''m browsing to .../databases/mycodo.db of version 7.10. to select the import file.

kizniche, 2019-11-18:
That's not how the import works. The import can only import the archive that was exported. Use the file that was exported with the import feature.

SAM26K, 2019-11-19:
Was having p;roblems with export working so I saved some of the backups in /var/Mycodo-backups/ on a pc then transferred the backup files back to the rpi after installing Mycodo 8.x.  The backup shows up in the backup/restore list but the restore wouldn't work and corrupted mycodo and had to re-install.
I noticed the files created by mycodo have "owner and group" change permission whereas the backup folder I copied only has owner with change permissions.  Is there anyway to change the permissions to get it to restore the settings from 7.1?  If I can't salvage what I have I'll be starting over.  For some reason my backed up SD images are not even working.

kizniche, 2019-11-19:
This is what I would do:

1. copy mycodo.db to ~/Mycodo/databases/
2. Run this command: ```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade-master```

This is the easiest way to get your new database in and updated to the latest database version.

SAM26K, 2019-11-19:
Do I need to change the 'change permissions' on mycodo.dbi from owner to owner and group?
Using terminal sudo commands to copy the file.

SAM26K, 2019-11-19:
Your instructions worked to restore the settings but the ADS1256 is not returning data.  The TH16's are working.

Daemon log:

  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_7873c2c3-e6fa-431e-b7c2-6618628a8131.py'
2019-11-18 20:19:48,480 - DEBUG - mycodo.inputs.th1x_am2301_352ca283 - Returned Data: {'StatusSNS': {'Time': '2019-11-19T05:19:48', 'AM2301': {'Temperature': 78.6, 'Humidity': 55.5}, 'TempUnit': 'F'}}
2019-11-18 20:19:48,546 - ERROR - mycodo.controllers.controller_input_7e3dc86d - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2019-11-18 20:19:48,740 - DEBUG - mycodo.controllers.controller_input_352ca283 - Adding measurements to InfluxDB with ID 352ca283-98cb-47d4-a44c-93450d176537: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.88889, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 55.5, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.297157208703727, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1485.7112185913204, 'timestamp_utc': None}}
2019-11-18 20:19:48,897 - ERROR - mycodo.controllers.controller_conditional_4a85c188 - Exception executing check_conditionals() on attempt 2 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_4a85c188-a05d-40b4-ae26-c53a8f2c8a9f.py'
2019-11-18 20:19:54,626 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-18 20:19:57,053 - ERROR - mycodo.controllers.controller_conditional_69a7d632 - Exception executing check_conditionals() on attempt 3 of 3.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_69a7d632-f5b6-4b7b-9e7b-6c5d46337b11.py'
2019-11-18 20:19:57,314 - ERROR - mycodo.controllers.controller_input_066f2eb0 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2019-11-18 20:19:57,698 - ERROR - mycodo.controllers.controller_conditional_7873c2c3 - Exception executing check_conditionals() on attempt 3 of 3.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_7873c2c3-e6fa-431e-b7c2-6618628a8131.py'
2019-11-18 20:19:58,974 - ERROR - mycodo.controllers.controller_conditional_4a85c188 - Exception executing check_conditionals() on attempt 3 of 3.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_4a85c188-a05d-40b4-ae26-c53a8f2c8a9f.py'
2019-11-18 20:20:03,521 - DEBUG - mycodo.inputs.th1x_am2301_352ca283 - Returned Data: {'StatusSNS': {'Time': '2019-11-19T05:20:03', 'AM2301': {'Temperature': 78.6, 'Humidity': 55.5}, 'TempUnit': 'F'}}
2019-11-18 20:20:03,882 - DEBUG - mycodo.controllers.controller_input_352ca283 - Adding measurements to InfluxDB with ID 352ca283-98cb-47d4-a44c-93450d176537: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.88889, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 55.5, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.297157208703727, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1485.7112185913204, 'timestamp_utc': None}}
2019-11-18 20:20:08,063 - ERROR - mycodo.controllers.controller_conditional_8574b62a - Exception executing check_conditionals() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
[... cut ...]
2019-11-18 20:20:33,700 - DEBUG - mycodo.controllers.controller_input_352ca283 - Adding measurements to InfluxDB with ID 352ca283-98cb-47d4-a44c-93450d176537: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 25.88889, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 55.5, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.297157208703727, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1485.7112185913204, 'timestamp_utc': None}

SAM26K, 2019-11-19:
When I click on 'deactivate' on the ADS1256 input controller I get these messages:

    Success: Input Deactivate (SQL)

    Error: Could not deactivate Input controller with ID 64a5f8ec-251a-4146-bd39-e17ae4e88ec2: 'NoneType' object has no attribute 'stop_input'

kizniche, 2019-11-19:
As a result of manually inserting the database, no dependencies are installed for your devices. You will need to add one of every kind of device you have installed to install the dependencies.

kizniche, 2019-11-19:
A benefit of the settings database import is it also will install dependencies for all devices in the database being imported.

SAM26K, 2019-11-19:
Tried adding another ads1256 control and same 0 data problem.  also before restoring the settings,  I tried adding an ads1286  data control (using 8.0.3) and same problem.
Looked like dependecies are OK:
 
Dependencies

Several software dependencies used by Mycodo features may be disabled. This is done to speed up the initial installation process. This page can be used to determine which dependencies are installed and the ability to install currently unmet dependencies.
Device: ADS1256 (ADS1256)

All dependencies have been installed for this device.

kizniche, 2019-11-19:
Can you provide the log excerpts of just the ADS errors after activating it?

Also, your Conditional errors are a result of the generated file for each not existing. You will need to save every conditional for these files to be generated.

SAM26K, 2019-11-19:
Last few lines of Daemon log just after activating ADS1256.  All other data input controllers disab;ed:
(For some reason I always get that first warning when actiivating the ADS1256..)

2019-11-19 11:31:48,603 - WARNING - mycodo.daemon - Cannot activate Input controller with ID 64a5f8ec-251a-4146-bd39-e17ae4e88ec2: It's already active.
2019-11-19 11:31:58,718 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 11:32:01,239 - ERROR - mycodo.controllers.controller_input_905ed4c5 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 11:32:13,718 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 11:32:16,239 - ERROR - mycodo.controllers.controller_input_905ed4c5 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

kizniche, 2019-11-19:
I need it to become active from an inactive state to get any meaningful errors. If you need to, deactivate it, restart the Pi, then activate it.

SAM26K, 2019-11-19:
1. Disabled all input controllers.
2. Restart system.
3. Enabled ADS1256 (ID# 64a5f8ec-251a-4146-bd39-e17ae4e88ec2)

Daemon log:

2019-11-19 13:13:13,362 - INFO - mycodo.daemon - Mycodo daemon started in 64.595 seconds
2019-11-19 13:13:13,368 - INFO - mycodo.daemon - 44.79 MB RAM in use
2019-11-19 13:15:01,854 - ERROR - mycodo.controllers.controller_input_64a5f8ec - initialize_variables() Exception: SPI device /dev/spi* not found. Ensure SPI is enabled and the device is recognized/setup by linux.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 329, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/ads1256.py", line 148, in __init__
    "SPI device /dev/spi* not found. Ensure SPI is enabled "
Exception: SPI device /dev/spi* not found. Ensure SPI is enabled and the device is recognized/setup by linux.
2019-11-19 13:15:01,879 - INFO - mycodo.controllers.controller_input_64a5f8ec - Activated in 754.1 ms
2019-11-19 13:15:01,880 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 13:15:16,528 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 13:15:31,528 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 13:15:46,528 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 13:16:01,528 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2019-11-19 13:16:16,528 - ERROR - mycodo.controllers.controller_input_64a5f8ec - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

kizniche, 2019-11-19:
> 2019-11-19 13:15:01,854 - ERROR - mycodo.controllers.controller_input_64a5f8ec - initialize_variables() Exception: SPI device /dev/spi* not found. Ensure SPI is enabled and the device is recognized/setup by linux.

Is "/dev/spi*" the device you have configured? This is not a valid device name.

kizniche, 2019-11-19:
Did you enable SPI with raspi-config (or Config -> Configure -> Raspberry Pi)?

SAM26K, 2019-11-19:
That was it! It's working now.  Back to monitoring daemon up time.
Thanks very much!

SAM26K, 2019-11-19:
Plan to update other rpi's to stretch 10 now that it's working.    Will try import/export again and start a new issue if it doesn't work.

SAM26K, 2019-11-19:
Still having problems with 8.x (or Stretch 10).  Linux bash files' command lines not working. did 'sshpass' requiirements change? 

command:
err=$((sshpass -p TARA9CAT ssh mycodo@192.168.1.51 mycodo-client --output_state 4128aad4-56ae-4dc0-a059-efb44b786487))"

returns::
bash: sshpass: command; not found

SAM26K, 2019-11-19:
Ah had to install sshpass since its a fresh os.

kizniche, 2019-11-19:
Use the API like I suggested in an earlier comment where I gave example code. You're likely to have the same issue if you don't change the method you access the remote Pi data.

SAM26K, 2019-11-19:
Will try that.  sshpass is not working right with stretch 10 apparently.

SAM26K, 2019-11-20:
Having a hard time understanding the generic examples.  Given the following  parameters how would I turn the relay off or on from a remote  pi?

Local pi (Has relays wired to GPIO lines)
1. IP Address 192.168.1.51
2. Relay Controller ID: c2dd0a27-6e0f-4a4b-abf9-da76dee0199f
3. API Key: <apikey>

Thanks!

SAM26K, 2019-11-21:
Trying to get your example above to work but it's not returning data (python3 data input).  Supposed to return data from an ADS1256 on a remote    pi at 192.168.1.17:
APi Key generated for user Admin generated on remote pi 192.168.1.17. Local ;pi is at 192.168.1.51. 

import json
import requests
ip_address = '127.0.0.1'
api_key = 'aLtqAUiDg5OPCuNmU70vmrDvHpDTtKiGZJwUZ4ifmLf27y'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'
channel = '0'
endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
    unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
response = requests.get(url, headers=headers, verify=False)
response_dict = json.loads(response.text)
print("Time: {}, Value: {}".format(response_dict['time'], response_dict['value']))

Daemon errors:
2019-11-21 13:07:54,804 - INFO - mycodo.controllers.controller_input_7330324b - Activated in 324.5 ms
2019-11-21 13:08:24,944 - ERROR - mycodo.controllers.controller_input_7330324b - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2019-11-21:
You have the IP address set to 127.0.0.1

kizniche, 2019-11-21:
Also, for the linux command input, you will want to format the code to return a single numerical value or nothing, or an error. This can accomplish that, get_measurement.py:

```python
import json
import requests
ip_address = '127.0.0.1'
api_key = 'YOUR_API_KEY'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'
channel = '0'
endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
    unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
try:
    response = requests.get(url, headers=headers, verify=False)
    response_dict = json.loads(response.text)
    print("{}".format(response_dict['value']))
except Exception as err:
    print("Error: {}".format(err))
```

You will want to execute this with the command:

```bash
/home/pi/Mycodo/env/bin/python /home/pi/get_measurement.py
```

kizniche, 2019-11-21:
For the ```Python 3 Code``` Input, this would be the code:

```python
import json
import requests
ip_address = '127.0.0.1'
api_key = 'YOUR_API_KEY'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'
channel = '0'
endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
    unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
try:
    response = requests.get(url, headers=headers, verify=False)
    response_dict = json.loads(response.text)
    self.store_measurement(channel=0, measurement=response_dict['value'])
except Exception as err:
    self.logger.exception("Error")
```

kizniche, 2019-11-21:
Also, for security, you should regenerate your API key. You do not want to give anyone your API key that you wouldn't want controlling your system.

SAM26K, 2019-11-21:
Working now for Ch0 only.  Can I  use a for loop instead of repeating all that code? Up to 8 channels.
Thanks for the help!

SAM26K, 2019-11-22:
The new raspian image is not  user friendly.  It automatically expands the image to the size of the SD card on first use which makes it un-usable on other size cards.  I like to setup a new image with wifi network parameters so I have local headless access I can use with any size SD card before inflating the image.  Just a note of grief with the new os.

kizniche, 2019-11-22:
> I like to setup a new image with wifi network parameters so I have local headless access I can use with any size SD card before inflating the image.

You can do this by creating the file wpa_supplicant.conf on the boot partition first after flashing.

For the ```Python 3 Code Input```:

```python
import json
import requests
ip_address = '127.0.0.1'
api_key = 'YOUR_API_KEY'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'

for channel in range(8):
    endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
        unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
    url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
    headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
    try:
        response = requests.get(url, headers=headers, verify=False)
        response_dict = json.loads(response.text)
        self.store_measurement(channel=channel, measurement=response_dict['value'])
    except Exception:
        self.logger.exception("Error")
```

kizniche, 2019-11-22:
Actually, I see issues with that code. It creates a new timestamp that's different than when the actual measurement was created. I just pushed a commit that enables the use of a user-defined timestamp for measurements being stored with the Python 3 Code Input. You'll need to upgrade to master to try it. Here's the code to preserve the timestamps:

```python
import json
import requests
ip_address = '127.0.0.1'
api_key = 'YOUR_API_KEY'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'

for channel in range(8):
    endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
        unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
    url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
    headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
    try:
        response = requests.get(url, headers=headers, verify=False)
        response_dict = json.loads(response.text)
        self.store_measurement(
            channel=channel,
            measurement=response_dict['value'],
            timestamp="{}Z".format(response_dict['time'].split("+")[0]))
    except Exception:
        self.logger.exception("Error")
```

You can also try this code that will spawn 8 threads to concurrently try to get/save all 8 measurements, which may prevent conditions where measurements with different timestamps are retrieved due to the process taking a while to complete (i.e. after getting the measurements starts, the remote Input starts acquiring new measurements, thereby causing the timestamp of the retrieved measurements to become the newer measurements just acquired). That is difficult to explain, but hopefully that came across. Anyway, I've never tested threads in this Input, but here is the code if you want to test. This will also require upgrading to master to obtain the new timestamp parameter in the store_measurement function.

```python
import json
import requests
import threading

ip_address = '127.0.0.1'
api_key = 'YOUR_API_KEY'
uuid = '64a5f8ec-251a-4146-bd39-e17ae4e88ec2'
unit = 'V'

def store_measurement_for_chan(channel):
    endpoint = 'measurements/last/{unique_id}/{unit}/{channel}/{past_seconds}'.format(
        unique_id=uuid, unit=unit, channel=channel, past_seconds=250)
    url = 'https://{ip}/api/{ep}'.format(ip=ip_address, ep=endpoint)
    headers = {'Accept': 'application/vnd.mycodo.v1+json', 'X-API-KEY': api_key}
    try:
        response = requests.get(url, headers=headers, verify=False)
        response_dict = json.loads(response.text)
        self.store_measurement(
            channel=channel,
            measurement=response_dict['value'],
            timestamp="{}Z".format(response_dict['time'].split("+")[0]))
    except Exception:
        self.logger.exception("Error")

for channel in range(8):
    store_measurement = threading.Thread(
        target=store_measurement_for_chan,
        args=(channel,))
    store_measurement.start()
```

[26 more replies]

closed 2020-02-21 by kizniche