kizniche/Mycodo #712: Error 502: Bad Gateway Error

bndaquino, 2019-11-11:
Hi Kyle,

I'm trying to reinstall Mycodo. The installation says it's successful, but I get an Error 502 when I try to go to my pi's IP address. I've tried rebooting my pi and deleting the Mycodo directory and installing again, but neither helps. I'm pretty new to Mycodo and raspberry pi in general, so I'm at a loss for what else to try.

Prior to trying to reinstall Mycodo today, I was able to get readings from a couple sensors through Mycodo. It's only after trying to reinstall that I'm getting this problem.

Here's my setup.log:

`### Mycodo installation initiated Mon Nov 11 14:37:31 EST 2019
#### Mycodo installation began Mon Nov 11 14:38:21 EST 2019

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Updating apt repositories
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
Package 'python-pip' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libpython-all-dev lxkeymap python-all python-all-dev python-cairo
  python-gobject python-gobject-2 python-gtk2 python-keyring
  python-keyrings.alt python-secretstorage python-wheel python-xklavier
  realpath
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 140 not upgraded.

#### Installing prerequisite apt packages and update pip
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Package 'apache2' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libpython-all-dev lxkeymap python-all python-all-dev python-cairo
  python-gobject python-gobject-2 python-gtk2 python-keyring
  python-keyrings.alt python-secretstorage python-wheel python-xklavier
  realpath
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 140 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
gawk is already the newest version (1:4.1.4+dfsg-1).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u4).
libffi-dev is already the newest version (3.2.1-6).
libi2c-dev is already the newest version (3.1.2-3).
logrotate is already the newest version (3.11.0-0.1).
moreutils is already the newest version (0.60-1).
nginx is already the newest version (1.10.3-1+deb9u3).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
python3-pylint-common is already the newest version (0.2.2-2).
python3-setuptools is already the newest version (33.1.1-1).
python3-smbus is already the newest version (3.1.2-3).
rng-tools is already the newest version (2-unofficial-mt.14-1).
sqlite3 is already the newest version (3.16.2-5+deb9u1).
wget is already the newest version (1.18-5+deb9u3).
The following packages were automatically installed and are no longer required:
  libpython-all-dev lxkeymap python-all python-all-dev python-cairo
  python-gobject python-gobject-2 python-gtk2 python-keyring
  python-keyrings.alt python-secretstorage python-wheel python-xklavier
[... 335 lines cut ...]
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Job for mycodo.service failed because the control process exited with error code.
See "systemctl status mycodo.service" and "journalctl -xe" for details.
#### Mycodo Installer finished Mon Nov 11 14:50:28 EST 2019
#### Total install time: 12 minutes and 7 seconds

************************************
** Mycodo successfully installed! **
************************************`

Thanks in advance for any help.

kizniche, 2019-11-11:
Can you run these commands and copy the output here?

```bash
sudo service mycodoflask stop
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

bndaquino, 2019-11-11:
Thanks for the quick reply:
```
pi@raspberrypi:~ $ sudo service mycodoflask stop
pi@raspberrypi:~ $ cd ~/Mycodo/mycodo
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/run/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2019-11-11 15:26:41 -0500] [5069] [INFO] Starting gunicorn 19.9.0
[2019-11-11 15:26:41 -0500] [5069] [INFO] Listening at: unix:/var/run/mycodoflask.sock (5069)
[2019-11-11 15:26:41 -0500] [5069] [INFO] Using worker: gthread
[2019-11-11 15:26:41 -0500] [5072] [INFO] Booting worker with pid: 5072
[2019-11-11 15:26:43 -0500] [5072] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/gthread.py", line 104, in init_process
    super(ThreadWorker, self).init_process()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py", line 129, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py", line 138, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py", line 52, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py", line 41, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/util.py", line 350, in import_app
    __import__(module)
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
    from mycodo.mycodo_flask.app import create_app
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 23, in <module>
    from mycodo.databases.models import Misc
  File "/home/pi/Mycodo/mycodo/databases/__init__.py", line 7, in <module>
    from mycodo.mycodo_flask.extensions import db
  File "/home/pi/Mycodo/mycodo/mycodo_flask/extensions.py", line 13, in <module>
    from flask_marshmallow import Marshmallow  # must be imported after SQLAlchemy
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_marshmallow/__init__.py", line 24, in <module>
    from . import sqla
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_marshmallow/sqla.py", line 13, in <module>
    import marshmallow_sqlalchemy as msqla
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/marshmallow_sqlalchemy/__init__.py", line 1, in <module>
    from .schema import TableSchemaOpts, ModelSchemaOpts, TableSchema, ModelSchema
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/marshmallow_sqlalchemy/schema.py", line 3, in <module>
    from .convert import ModelConverter
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/marshmallow_sqlalchemy/convert.py", line 80
    ):
    ^
SyntaxError: invalid syntax
[2019-11-11 15:26:43 -0500] [5072] [INFO] Worker exiting (pid: 5072)
[2019-11-11 15:26:44 -0500] [5069] [INFO] Shutting down: Master
[2019-11-11 15:26:44 -0500] [5069] [INFO] Reason: Worker failed to boot.
```

kizniche, 2019-11-11:
It looks like you're running Python <= 3.6, as marshmallow_sqlalchemy is returning an error. I suspect you have Raspbian Stretch installed. Can you perform the following command to check?

```bash
lsb_release -a
```

You will need at least Raspbian Buster to install Mycodo. I thought I included a check for the Python version and returned an error to the user if the version was insufficient, but I must have overlooked that in the updates I've been committing recently.

```bash
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

You can also perform the following command to check your Python version: ```python3 --version```

With Buster, you should have 3.7.3:

```
pi@LabDev:~ $ python3 --version
Python 3.7.3
```

bndaquino, 2019-11-11:
Looks like you were right! I'll try upgrading to Buster. Thanks again for your help.
```
pi@raspberrypi:~/Mycodo/mycodo $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.8 (stretch)
Release:	9.8
Codename:	stretch
```

kizniche, 2019-11-11:
I just pushed a fix for this issue. Until I release 8.0.1, I'll leave this issue open for others to find.

closed 2019-11-11 by kizniche