kizniche/Mycodo #713: Error during update

floriske, 2019-11-13:
When trying to update from 7.9.1. to 7.10.0 I get the following errors during the update proces:

`ERROR: Command errored out with exit status 1:
   command: /home/pi/Mycodo/env/bin/python3.5 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-9uwam0ik/Pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-9uwam0ik/Pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-ug4ef3gp --python-tag cp35
       cwd: /tmp/pip-install-9uwam0ik/Pillow/
  Complete output (172 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.5
  creating build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageShow.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageSequence.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/_binary.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/Image.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageOps.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/MicImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/MspImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/SpiderImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/TarIO.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/DcxImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageQt.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImtImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImagePath.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/features.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageTk.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/JpegImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageMode.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/GimpPaletteFile.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/_version.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageColor.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/FontFile.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/PaletteFile.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/XbmImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImagePalette.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/GifImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/DdsImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/MpegImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/PixarImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageFont.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/BufrStubImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/__init__.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/__main__.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageMorph.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/FtexImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
[... 267 lines cut ...]
      File "/usr/lib/python3.5/distutils/command/build_ext.py", line 339, in run
        self.build_extensions()
      File "/tmp/pip-install-9uwam0ik/Pillow/setup.py", line 687, in build_extensions
        raise RequiredDependencyException(f)
    __main__.RequiredDependencyException: jpeg
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-9uwam0ik/Pillow/setup.py", line 903, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
    
    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html
    
    
    ----------------------------------------
  ERROR: Can't roll back Pillow; was not uninstalled
ERROR: Command errored out with exit status 1: /home/pi/Mycodo/env/bin/python3.5 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-9uwam0ik/Pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-9uwam0ik/Pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-cwz72iyc/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python3.5/Pillow Check the logs for full command output.
`

kizniche, 2019-11-13:
Do you have libjpeg-dev installed? You can find out by running ```dpkg -l libjpeg-dev```

floriske, 2019-11-13:
Found a possible solution in running sudo apt-get install libjpeg-dev zlib1g-dev, update is currently running again

kizniche, 2019-11-13:
libjpeg-dev is a dependency (along with pillow, instructed to install prior to pillow) of the amp8833 and OLED displays. The only way I can think you wouldn't have these dependencies for pillow installed is if you copied a database from a prior installation or uninstalled it. I actually just added in the latest 8.0.1 release to check for any missing dependencies and install them (when the database is imported through the database import page).

floriske, 2019-11-13:
I had some issues updating from 7.7.1 to 7.9.1 so I reverted back to a backup version of 7.7.1 after that the update to 7.9.1 went ok, 

When trying to update to 7.10.0 I got the above errors, could this have something to do with reverting back to 7.7.1 in between?

floriske, 2019-11-13:
Now it seems to be stuck on "Processing triggers for man-db (2.7.6.1-2)" (the actual description is in Dutch but this is the English variant I think it should be).

I'll leave it be for now and hope it continues in a while ...

kizniche, 2019-11-13:
I don't think so... Restoring a backup shouldn't uninstall anything. Did installing the dev packages actually allow Pillow to be installed? Did that error prevent the upgrade from completing properly? I don't see anything on the software side that would cause this, unless these dependencies weren't in a particular version you had when you installed one of these devices that require Pillow, but then I wouldn't expect any upgrade to work if the error prevents it from completing.

https://github.com/kizniche/Mycodo/blob/4ad7c9083127c4e63ebd9227f15aaaae97d2b713/mycodo/inputs/amg8833.py#L41-L47

kizniche, 2019-11-13:
> Now it seems to be stuck on "Processing triggers for man-db (2.7.6.1-2)" 

This can take a while on some systems

floriske, 2019-11-13:
Ok, I'll remain patient ;-) it did actually install Pillow after installing the dev packages

kizniche, 2019-11-13:
More than a few minutes and I would become suspicious.

floriske, 2019-11-13:
In that case ... I'm still waiting

I'll leave it be for now, will have a look later on.

kizniche, 2019-11-13:
man-db is only for the documentation, so shouldn't interfere with the library if it can't be completed. It's also ran at the end of a package install, so the library is likely intact.

floriske, 2019-11-13:
It just started to continue!

floriske, 2019-11-13:
And we're back up and running.

Thanks again for the help!

closed 2019-11-13 by floriske