kizniche/Mycodo #715: CCS811 sensor reporting "No Data"

allstar87, 2019-11-18:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

Using a Sparkfun Environmental Combo Breakout with a BME280 and CCS811 sensor.  The BME280 is functioning, the CCS811 are not.  Two different boards show the same behavior.  Both are connected via a Sparkfun I2C multiplexer, both show on I2C buses 4 and 5 as address 0x5b.  Setting these up as inputs and then setting up a dashboard measurement to display the readings from the CCS811 shows "NO DATA, MAX AGE EXCEEDED."  

### Versions:

 - Mycodo Version: 7.10
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Buster Lite

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2019-11-18:
Can you deactivate the input, check "Log Level: Debug", save, activate the input, then go to Config -> Mycodo Logs and view the daemon log for errors.

kizniche, 2019-11-18:
Have you also tested the sensor without the multiplexer?

allstar87, 2019-11-18:
When I go to activate or deactivate the input, I get the orange banner that says something about "cannot deactivate, has no attribute "stop input" or "is already active."
Daemon log contains the following:
2019-11-17 21:54:17,614 - ERROR - mycodo.controllers.controller_input_c269c859 - initialize_variables() Exception: Device returned an Error! Try removing and reapplying power to the device and running the code again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 329, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/ccs811.py", line 67, in __init__
    busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_CCS811/Adafruit_CCS811.py", line 92, in __init__
    raise Exception("Device returned an Error! Try removing and reapplying power to the device and running the code again.")
Exception: Device returned an Error! Try removing and reapplying power to the device and running the code again.
2019-11-17 21:54:17,622 - INFO - mycodo.controllers.controller_input_c269c859 - Activated in 382.6 ms
2019-11-17 21:54:17,622 - ERROR - mycodo.controllers.controller_input_c269c859 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 399, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

kizniche, 2019-11-18:
Do you have an arduino to follow their [guide](https://learn.sparkfun.com/tutorials/ccs811-air-quality-breakout-hookup-guide) to determine if the sensor works?

allstar87, 2019-11-18:
Still doesn't work when plugged into the hardware I2C pins on GPIO 2 and 3 directly, without MUX (if it helps, the MUX was plugged into GPIO 23 and 24 via Theoi-Meteoroi/GPIO-pca9548 as per mycodo manual.)
I do not have an arduino with me where I am right now, I can fetch one from my workshop on Tuesday.
Adafruit says something about this sensor not working well with the Pi due to I2C clock stretching, have any of these been confirmed to work in Mycodo?

kizniche, 2019-11-18:
#467 #641 are related to the CCS811 and the input/sensor has been verified to work in the past. Checking if an arduino can obtain measurements with their code would be where I would start. Once we verify it's not a hardware issue, I'll be happy to work with you to see if it's a Mycodo issue.

allstar87, 2019-11-21:
The CCS811s work when attached to an Arduino.  I'm breathing on it and making numbers go up.

kizniche, 2019-11-23:
Are you able to provide SSH (user pi) and HTTPS (& Mycodo admin login) access to your Pi so I can test? I don't have this sensor to test myself. If so, you can [contact me privately](https://kylegabriel.com/contact/) with the credentials.

kizniche, 2020-01-26:
I logged in tonight and tested both Adafruit's deprecated CCS811 python library and their newer one that uses circuitpython. Neither worked. Can you try [setting up software I2C](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#am2315) and seeing if that fixes the issue?

allstar87, 2020-01-26:
Done.  Software I2C enabled, SDA on BCM23 and SCL on BCM24, getting data
from the BME280 but not the CCS811.

kizniche, 2020-01-26:
Well, I'm out of ideas. I just tested both libraries again, but no luck. You may have a defective sensor that seems to work with an arduino despite this. I'm not entirely sure. I also don't have this sensor and have relied on other users to confirm it works with the built-in library with mycodo.

kizniche, 2020-01-26:
Let me try another library I just found.

kizniche, 2020-01-26:
https://github.com/sparkfun/CCS811_Air_Quality_Breakout doesn't seem to work either. That was the last python library I could find.

At this point, your options are:

1. Find someone that can diagnose the issue better than I can.
2. Try another of the same sensor.
3. Try another of a different sensor.
4. Interface an ATMega to read the sensor and relay the data to the Raspberry Pi whenever the Raspberry Pi queries the ATMega for sensor data.

allstar87, 2020-01-26:
Well, let's run through those options:
1.  You say you have other users who report success with this sensor, could
you put me in contact with them, perhaps through your forum?
2.   I have two of these boards, both showed the same behavior so I don't
think I have a defective unit.
3.  Other CO2 sensors are pretty expensive; looking at those listed as
options already supported by Mycodo, the cheapest is twice the price of
this one.  I'll consider that a last resort.
4.  adding an ATMEGA or ATTINY chip to the mix would be the most cost
effective solution hardware wise, they're about a buck each. What would you
say is the ideal way of doing that?  Attach the AVR via I2C and have it
pretend to be the sensors, or attach via UART and do some custom Python
code on the Pi?

Quick question, was that sparkfun library looking for the device on I2C1,
or I2C3?  It's still plugged into I2C3.

kizniche, 2020-01-26:
> 1.  You say you have other users who report success with this sensor, could you put me in contact with them, perhaps through your forum?

Take look at the issues in [this comment](https://github.com/kizniche/Mycodo/issues/715#issuecomment-554840237). I'm not sure about the forum, but you can search for mentions of CCS811.

> 2.   I have two of these boards, both showed the same behavior so I don't think I have a defective unit.

Not likely then.

> 3.  Other CO2 sensors are pretty expensive; looking at those listed as
options already supported by Mycodo, the cheapest is twice the price of this one.  I'll consider that a last resort.

The MH-Z19 is as low as $24, and with a $5 UART to USB converter, it's not a bad option. I have one of these and it works great. It's close to the cost of a CCS118 breakout board. But, you already have the sensors, so it's a good idea to get your money's worth out of them.

> 4.  adding an ATMEGA or ATTINY chip to the mix would be the most cost effective solution hardware wise, they're about a buck each. What would you say is the ideal way of doing that?  Attach the AVR [...]

I've never done it, but I know it's possible. The ATMega essentially acts as an I2C slave device, appearing on the Raspberry Pi I2C device tree at a particular address, and responds to queries like a typical I2C sensor would. You could then create a custom input module and import it into Mycodo through the configuration menu. You could also set up a serial connection between an ATMega and the Pi, such as the USB connection many boards use for programming. Then you could write a bash or Python script to write to that USB device (e.g. /dev/USB0), which the microcontroller would receive, query the sensor, then return the measurements. I have used this second method, and it works great.

> Quick question, was that sparkfun library looking for the device on I2C1,
or I2C3?  It's still plugged into I2C3.

Yes, I tested it with bus 3.

allstar87, 2020-01-26:
Looking at the latest commit in issue #641, my ccs811.py file is different than that one.
for example, instead of:
if self.is_enabled(2):
    return_dict[2]['value'] = temp
I have:
if self.is_enabled(2):
    self.value_set(2, temp)

The line numbers also seem to be different.

kizniche, 2020-01-26:
That's merely the framework for inputting the measurements into the time-series database (influxdb). The Python library for acquiring measurements and its implementation within the input module is the same. The framework change was done to make it easier for users to develop their own custom input modules.

allstar87, 2020-01-26:
Trying one more thing.  I'm seeing several mentions around the internet about having to slow the I2C baudrate "way down" to handle the clock stretching (Thanks, Broadcom!) so I'm adding the line
`dtparam=i2c_baudrate=10000`
to /boot/config.txt  Let's see what I break this time.

closed 2020-07-23 by kizniche

djansen1987, 2024-02-03:
any luck finding the solution. Both 2 of the sensors but are unable to get them to work with ESPhome