kizniche/Mycodo #716: ERROR: Could not find conversion for unit C to K

zsole2, 2019-11-21:
### Versions:

 - Mycodo Version: 8.0.3
 - Raspberry Pi Version: Zero
 - Raspbian OS Version: Buster Lite

### Reproducibility

Add Math/Humidity(Wet/Dry)
Calculate only percent humidity
Set appropriate sensors
Did not add any conversion
When activated, there are errors in the log:
```
2019-11-21 05:40:09,836 - INFO - mycodo.controllers.controller_math_e1b0aa60 - Activated in 445.6 ms
2019-11-21 05:40:12,141 - ERROR - mycodo.controllers.controller_math_e1b0aa60 - Could not find conversion for unit C to K
2019-11-21 05:40:12,858 - ERROR - mycodo.controllers.controller_math_e1b0aa60 - Could not find conversion for unit C to K
2019-11-21 05:40:12,872 - ERROR - mycodo.controllers.controller_math_e1b0aa60 - One or more critical errors prevented the humidity from being calculated
```

### Expected behavior

I expected to get humidity measurements without setting any conversions anywhere (the temp sensors were left on defaults).

### Additional context

I just setting up a new system, and decided to go with the latest release. Looks really nice, great work you have done, Kyle!!! Especially surprised how it is more snappier on the Pi Zero.
Still getting the hang of things, looking forward to see how the conditionals work (coming from v5.6.9, it is quite a change!)

kizniche, 2019-11-21:
Thanks for the bug report. I'll fix the conversion. It's necessary for the module that calculates relative humidity, as it needs the units to be Kelvin.

Yes, I was also surprised with the noticeable performance improvement going to Buster/Python 3.7.

kizniche, 2019-11-21:
I just pushed a potential fix. Can you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) or manually edit the controller_math.py file (and restart the daemon) to test the fix?

zsole2, 2019-11-22:
Thanks, I'm travelling but was able to upgrade. 
Humidity values are now presented, but apparently calculated incorrectly: 
Dry: 13.812 C
Wet: 13.875 C
Hum: 38.945%
Should be 100%, and even if the values were reversed, it should also be well above 90%.

kizniche, 2019-11-22:
Wet-bulb should be lower than the dry-bulb. It seems you either have inaccuracy with your sensor(s) or they're improperly placed/setup.

kizniche, 2019-11-22:
The wet being above the dry may be causing an issue with the software that calculates the humidity. I'll run an experiment and see if I need to include a check for wet being above the dry and returning 100% if true.

kizniche, 2019-11-22:
My example program is calculating 100% humidity from your temperature values, using the code used by Mycodo:

```python
import mycodo.utils.psypy as SI

temp_c_wet = 13.875
temp_c_dry = 13.812
pressure_pa = 101325

temp_k_wet = temp_c_wet + 273.15
temp_k_dry = temp_c_dry + 273.15

psypi = SI.state("DBT", temp_k_dry, "WBT", temp_k_wet, pressure_pa)

if psypi:
    percent_relative_humidity = psypi[2] * 100

    # Ensure percent humidity stays within 0 - 100 % range
    if percent_relative_humidity > 100:
        percent_relative_humidity = 100
    elif percent_relative_humidity < 0:
        percent_relative_humidity = 0

    specific_enthalpy = float(psypi[1])
    humidity = float(percent_relative_humidity)
    specific_volume = float(psypi[3])
    humidity_ratio = float(psypi[4])

    print("Humidity: {}".format(humidity))
```

zsole2, 2019-11-22:
This is a system under construction, both sensors run dry. Unable to access them, I simply switched them in the math controller. Now the readings are:
12.125
12
34.254
With a difference of about 0.1C, the humidity should be around 98-99%.
My production system (v5.6.9) currently has:
7.94
7.19
90.414
This is about correct...

kizniche, 2019-11-22:
I added a debug line in the controller to show the value sbeing use to calculate the humidity, in the daemon log. Can you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and turn on ```Log Level: Debug``` for your Math controller, and see what the values are in the daemon log?

zsole2, 2019-11-22:
I think this will be helpful...
```
2019-11-22 21:46:40,421 - DEBUG - mycodo.controllers.controller_math_e1b0aa60 - Dry Temp: (C: 12.687, K: 285.837), Wet Temp: (C: 12.562, K: 285.712), Pressure: 101325,Humidity: 98.62522544863097
2019-11-22 21:46:40,986 - DEBUG - mycodo.controllers.controller_math_e1b0aa60 - Adding measurements to InfluxDB with ID e1b0aa60-fd15-4921-88c5-34b71cbf7ae6: {0: {'measurement': 'humidity', 'unit': 'percent', 'value': 35.514934384947296}, 1: {'measurement': 'humidity_ratio', 'unit': 'kg_kg', 'value': 98.62522544863097}, 2: {'measurement': 'specific_enthalpy', 'unit': 'kJ_kg', 'value': 0.8214764067999315}, 3: {'measurement': 'specific_volume', 'unit': 'm3_kg', 'value': 0.00901205419917034}}
```
Seems like the order of the values is mixed up.

kizniche, 2019-11-22:
Yes! Very helpful. I just pushed a potential fix. Can you upgrade to master to test?

zsole2, 2019-11-22:
Will do soon...
If I left all 4 measurements in place, this would have been evident, though.

BTW, how can I add those other 3 back? Without deleting and re-adding the control. Or maybe only my phone interface makes it impossible?

kizniche, 2019-11-22:
From your log, it appears all 4 are enabled, as they're all being stored in influxdb. You can view them all on the live page and they should be available for selection on other pages (Dashboard, PID, etc.).

zsole2, 2019-11-22:
Yeah, I've seen it in the log. 
But believe me, only humidity was set up originally, and still is the only measurement reported:
[image]

[image]

I do not think that it is worth the trouble to write extra code for the ability to add back the missing measurements, but on the other hand, if really 4 items are  stored instead of 1 at every time point, it could be unnecessary overhead, especially on the Pi Zero, where influxdb is the biggest load.

kizniche, 2019-11-22:
Add a new Math controller and see if the new one has all the measurements.

zsole2, 2019-11-23:
I did that earlier, and it did have all the measurement. So my question was rather theoretic. 
I consider this issue resolved, thanks for your quick action!

kizniche, 2019-11-23:
Well, if you're install isn't behaving as expected, then there's still an issue. I just created a Dry-bulb Humidity Math Controller and these are the default measurements:

[image]

zsole2, 2019-11-23:
Yes, I had the 4 measurements available, and I deliberately deleted everything except humidity. 
Then during debugging the issue, I realized that there is no apparent way to re-add the disabled measurements. 
I can delete the old controller, and add a new one with any of the default measurements, and that was the reason to state that this is a non-issue.

kizniche, 2019-11-23:
Ah, sorry, I read what you wrote as "I did that earlier and didn't have all the measurements".

kizniche, 2019-11-23:
It appears I don't have code in the Math Controller to not add measurements to influxdb if they're disabled (not enabled by deselecting them in the selection box). I'll need to add that. Did you delete your measurements from mycodo.db with an SQLite editor?

zsole2, 2019-11-23:
No, íi did not. But this is a brand new system, so it should not matter, I think.

kizniche, 2019-11-23:
I don't know what you mean by "I deliberately deleted everything except humidity".

zsole2, 2019-11-23:
I mean when setting up the controller, I deselected three out of the four available measurements, leaving only humidity (CH0) to be calculated. I do not use the other three values. Sorry for the confusion.

kizniche, 2019-11-23:
None should ever disappear from the Measurements Enabled box if they're deselected.

zsole2, 2019-11-23:
I see your point now. But the three deselected measurement disappeared, see the screenshot above:
 [image]

kizniche, 2019-11-23:
I just replicated the issue. I'll be able to make a fix later.

closed 2019-11-26 by kizniche