kizniche/Mycodo #718: Dashboard PID control not working

zsole2, 2019-11-25:
### Versions:

 - Mycodo Version: 8.0.3
 - Raspberry Pi Version: Zero
 - Raspbian OS Version: Buster Lite

### Reproducibility

PID was set up normally, and works.
Set up PID Control as in the screen-shot

### Screenshots
Basic setup:
[image]

What seems strange to me is the content of the PID field:

[image]

### Additional context

Log  indicates several errors:
WEB:
```
Nov 25 15:49:00 chz-2 gunicorn[17515]: 2019-11-25 15:49:00,173 URL for 'last_pid' raised and error: 'NoneType' object has no attribute 'measurement'
Nov 25 15:49:00 chz-2 gunicorn[17515]: Traceback (most recent call last):
Nov 25 15:49:00 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 1049, in last_data_pid
Nov 25 15:49:00 chz-2 gunicorn[17515]:     if len(pid.measurement.split(',')) == 2:
Nov 25 15:49:00 chz-2 gunicorn[17515]: AttributeError: 'NoneType' object has no attribute 'measurement'
```
... this is repeated continuously.

When I click on the buttons in the control, there are also errors, like this example:
```
Nov 25 15:56:08 chz-2 gunicorn[17515]: 2019-11-25 15:56:08,462 Exception on /pid_mod_unique_id/9c2bbb62-dd4e-4ad1-bb05-08803b230df3,c7deead2-899f-4dd9-95a0-888d5e8d8f44/pause_pid [GET]
Nov 25 15:56:08 chz-2 gunicorn[17515]: Traceback (most recent call last):
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
Nov 25 15:56:08 chz-2 gunicorn[17515]:     response = self.full_dispatch_request()
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
Nov 25 15:56:08 chz-2 gunicorn[17515]:     rv = self.handle_user_exception(e)
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
Nov 25 15:56:08 chz-2 gunicorn[17515]:     return original_handler(e)
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
Nov 25 15:56:08 chz-2 gunicorn[17515]:     reraise(exc_type, exc_value, tb)
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
Nov 25 15:56:08 chz-2 gunicorn[17515]:     raise value
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
Nov 25 15:56:08 chz-2 gunicorn[17515]:     rv = self.dispatch_request()
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
Nov 25 15:56:08 chz-2 gunicorn[17515]:     return self.view_functions[rule.endpoint](**req.view_args)
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
Nov 25 15:56:08 chz-2 gunicorn[17515]:     return func(*args, **kwargs)
Nov 25 15:56:08 chz-2 gunicorn[17515]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 1151, in pid_mod_unique_id
Nov 25 15:56:08 chz-2 gunicorn[17515]:     pid.is_paused = True
Nov 25 15:56:08 chz-2 gunicorn[17515]: AttributeError: 'NoneType' object has no attribute 'is_paused'
```

kizniche, 2019-11-25:
The PID Control Widget should have PID Devices to select from, not PID Measurements. I'll fix that shortly.

kizniche, 2019-11-25:
Just pushed a fix that was tested to work, which will be live in the next release. Thanks for the bug report.

closed 2019-11-25 by kizniche