kizniche/Mycodo #719: Kankun Smart Wifi Plug  Output support

smguth, 2019-12-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
No.  Wired relays and cable management hastles could be alleviated. 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.  I would like integration with Mycodo outputs.  
This is a wifi plug that can recieve commands via SSH.   I was considering using gpio wired relays but this seems like a perfect solution to avoid all the wires.  I just purchased a unit on amazon, waiting for it to arrive.  https://www.amazon.com/gp/product/B00MWR6ZY4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
My thought is I can use output commands with Mycodo and it will work just fine but also want the Mycodo community to know about this smart plug because I think they would like it and come up with ideas for it that I possibly would not
.  
Future enhancement suggestion.   Integration with Mycodo

**Additional context**
Add any other context or screenshots about the feature request here.
Please see this link it tells about the wifi plug further
http://www.anites.com/2015/01/hacking-kankun-smart-wifi-plug.html

kizniche, 2019-12-05:
I love the wireless outlets. The Command Output that's currently available in Mycodo would be the most appropriate for this. Essentially, a script needs to be developed that turns it on and off, then that script can be executed by the Mycodo Output Controller to control the plug. When you get the plug, we can start developing a working script to test.

kizniche, 2019-12-05:
It looks like the bash script from your second link should work well:

```bash
#!/bin/sh
echo "Content-Type: text/plain"
echo "Cache-Control: no-cache, must-revalidate"
echo "Expires: Sat, 26 Jul 1997 05:00:00 GMT"
echo

RELAY_CTRL=/sys/class/leds/tp-link:blue:relay/brightness

case "$QUERY_STRING" in
 state) 
  case "`cat $RELAY_CTRL`" in
   0) echo "OFF"
   ;;
   1) echo "ON"
   ;;
  esac
 ;;
 on) 
  echo 1 > $RELAY_CTRL
  echo OK
 ;;
 off) 
  echo 0 > $RELAY_CTRL
  echo OK
 ;;
esac
```

smguth, 2019-12-05:
Yeah I was wondering if it was worth bothing to add a check to ensure that
the relay state changed after the control was sent.   If i did bother with
it then do i send the control again and check the state again.  After so
many failed attempts trigger an alert or email.  Probably just start with
having it send the command and work from there though.  I have a feeling it
will always work as long as it is in range

SAM26K, 2019-12-09:
That AC plug looks awesome as it has ssh options.

SAM26K, 2019-12-11:
@smguth do you have a sales link to the US version?  Unable to locate so  far. Thanks.

smguth, 2019-12-11:
@SAM23K I only found it on amazon with the link from above https://www.amazon.com/gp/product/B00MWR6ZY4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1.   With the holidays I am still waiting for it to arrive too.  There was mention of finding it on  AliExpress for about $20 in the other link but I didn't go down that route.  I am planning on verifying that everything works smoothly when the one I ordered from Amazon comes in then probably try to source them cheaper as $30 per plug is a bit more than I ideally want to spend.  If anyone does find a better source for these please let us know!

kizniche, 2019-12-11:
There seem to be a lot of wifi plugs on the market, I'm sure there are many that have already been hacked or have an open API of sorts out-of-the-box.

https://www.amazon.com/s?k=wifi+plug

kizniche, 2019-12-11:
Where were these things a few years ago!? Some interesting finds:

https://www.amazon.com/AHRISE-Protector-6-Outlet-Compatible-Assistant/dp/B07VV2MGMC
https://www.amazon.com/POWRUI-4-Outlet-Weatherproof-Assistant-Network，Black/dp/B07KK62GB7
https://www.amazon.com/Compatible-Wireless-Smartphone-Waterproof-certified/dp/B07S915JZ7
https://www.amazon.com/Kasa-Smart-Power-Strip-TP-Link/dp/B07G95FFN3
https://www.amazon.com/IMILLET-Protector-Outlets-Compatible-Assistant/dp/B074CJYPZV

kizniche, 2019-12-11:
I found my holy grail of power strips, with the rare use of solid state relays, however it looks like this product has been discontinued: https://github.com/arendst/Tasmota/wiki/LeFun-SK2-Smart-Power-Strip-(Model-ZLD-44USA-W)

SAM26K, 2019-12-12:
@smguth How long ago did you order it? Says 10 days for delivery on Amazon.

SAM26K, 2019-12-12:
A lot of the wireless  plugs can be hacked by burning Tasmota firmware but often it's very difficult to connect to the data ports.  The Kankun is the first one I've heard of that has an open interface and reflashing isn't necessary.

kizniche, 2019-12-12:
I bought the LeFun power strip with hope I'll be able to put Tasmota on it. The SSRs are really appealing for my applications.

SAM26K, 2019-12-12:
That one looks good and on the Tasmota list of deviices.

smguth, 2019-12-12:
wow, now I want a smart power strip lol.  Looks like they can monitor power usage too!

smguth, 2019-12-12:
Wow I hadn't heard of Tasmota before but now I need to learn about it.  Do some of these smart plugs monitor power usage too?

smguth, 2019-12-12:
https://blakadder.github.io/templates/kmc-4.html  found this guy says it a energy monitoring smart plug might order this next after my kankun comes in

smguth, 2019-12-12:
https://blakadder.github.io/templates/ce_power_stake.html seems like this would be good for humid environments

SAM26K, 2019-12-13:
If you pick a device for Tasmota flashing google it  fist. I look for detailed instructions with pictures.

SAM26K, 2019-12-13:
Most of the wireless  AC adapters  use the ESP8266 chip which is the target for Tasmota.  Some devices are easy to connect to and others are very difficult to connect for reflashing.

SAM26K, 2019-12-13:
BTW the energy monitoring software is stock with many of the device's  original firmware but not sure if it's  included with Tasmota.  You can use Mycodo to set current draw  for a  device and have a calculated value for power logged to the database.

smguth, 2019-12-13:
Thanks SAM26K I think i'm starting to get a feel for some of this stuff now :D .   I'm still very new to Mycodo so might be off base with my understanding of it.  Trying to get my project off the ground yet but having fun learning as I go.  So if my device is Tasmota compatible does that mean it is compatible with Mycodo or might there still be configuration/integration yet?

kizniche, 2019-12-13:
If you can get Tasmota on it, which will allow for switching on/off via HTTP or SSH, then Mycodo can be configured to use it as an output.

SAM26K, 2019-12-16:
@smguth please flollow up when you finally get your  Kankun device.   May work with Mycodo  out of the  box without  Tasmota.

smguth, 2019-12-16:
Okay I'll keep you updated Amazon says it will arrive the 16-19 so possibly today.

smguth, 2019-12-22:
Got this guy in over the weekend had a few hurdles to overcome with the set up.  For what ever reason the android app wouldn't work for me so I had to configure it with ssh and get it on my wifi network.  The instructions worked but I had to set my password key to psk2 instead of what their guide said.  Also, to control the relay I had to use
root@koven:~# echo 0 > /sys/devices/platform/leds-gpio/leds/i-konke\:red\:relay/
brightness
root@koven:~# echo 1 > /sys/devices/platform/leds-gpio/leds/i-konke\:red\:relay/
brightness

My model must have been slightly different than the one in the guide

smguth, 2019-12-22:
my /etc/config/network file looks as follow: 

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd54:f988:6027::/48'

config interface 'lan'
	option ifname 'eth0'
	option force_link '1'
	option type 'bridge'
	option proto 'static'
	option ipaddr '192.168.145.253'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option proto 'dhcp'
	option ifname 'eth1'

config interface 'wwan'
	option proto 'dhcp'

and my /etc/config/wireless:

config wifi-device 'radio0'
	option type 'mac80211'
	option channel '11'
	option hwmode '11ng'
	option path 'platform/ar933x_wmac'
	list ht_capab 'SHORT-GI-20'
	list ht_capab 'SHORT-GI-40'
	list ht_capab 'RX-STBC1'
	list ht_capab 'DSSS_CCK-40'
	option htmode 'HT20'
	option disabled '0'
	option country 'CN'

config wifi-iface
	option device 'radio0'
	option network 'wwan'
	option ssid 'Network Name'
	option encryption 'psk2'
	option mode 'sta'
	option key 'password'

smguth, 2019-12-22:
[image]
Hi guys,  I created a script on the Pi that ssh's into the kankun and runs a script that turns on the relay.  If I run the script from my pi it turns it on and off with both sets of scripts.  When I try to run the script from mycodo it does not work.  I'm wondering if its a user or permissions thing.

SAM26K, 2019-12-24:
Looks promising.  Can you disable the password requirement?  You are probably right about permissions. Also try using Python code instead of ssh.

SAM26K, 2019-12-24:
Also is there errors in the daemon log?

smguth, 2019-12-24:
[image]
I didn't see any errors in the log.  It looks like mycodo thinks its just executing the commands succesfully from my understanding of it.  I'll look into writing this in python gonna have to do the holiday thing here shortly though.

SAM26K, 2019-12-25:
Having problems with my Linux controls since  installing the new Raspbian o/s and Mycodo 8.1.x. In the process of converting them to Python.

SAM26K, 2019-12-25:
Do you know what's causing the GPIO error?

smguth, 2019-12-25:
I have a Dht11 connected to pin 17 and it's giving me data so I was a little surprised to see the error.  I never put a resistor in series with the the data line now that I think about it so maybe I'll do that and see if it fixes it

kizniche, 2019-12-26:
I can add the option to execute the commands as the user pi or root. The default user commands are executed as is currently pi.

SAM26K, 2019-12-26:
That would be nice.

SAM26K, 2019-12-29:
@smguth any progress? Advise not to update with the vendors firmware.  You may be able to install Tasmota OTA (over the air)) without the need to wire directly into the device.

smguth, 2019-12-29:
Hey @SAM26K I havnt gotten much further.  I was hoping to try replace my scripts with python code but I've never coded in python before so that's been a hurdle.  I can try getting tasmota on it.  That will protect the manufacturer from locking me out?

kizniche, 2020-05-06:
> I can add the option to execute the commands as user pi or root.

I had forgotten about this since it's not really related to this issue, but I was going through issues today and found it. Better late than never. I just committed an update that adds the ability to select which user (pi, mycodo, or root) executes Linux Command Outputs.

I also forgot to give an update on the power strip I ordered. The seller cancelled the order and that was the only supplier I could find for that particular version.

smguth, 2020-05-14:
Well I got this working.  I reinstalled my operating system and started over with a clean mycodo install.  There were alot of steps but I followed the notes I had in this issue.  I also had to run
`ssh root@wifiplug "tee -a /etc/dropbear/authorized_keys" < ~/.ssh/id_rsa.pub`
From my raspberrypi in order for the pi to send the commands to the kankun smart plug without a password.   Also, I added wifiplug to hosts file so I didn't have to keep using the ip address.   No when I click 
[image]
 On and off buttons the relay in the kankun plug turns on and off

kizniche, 2020-07-23:
I'm reviving this thread since I just ordered 4 individual wifi outlets that include power monitoring.

https://templates.blakadder.com/efun_SH331W.html
https://www.amazon.com/gp/product/B07DCJ7TDR

Hopefully I won't have a reoccurrence of what happened last time and the seller cancels my order. I'll let you kow how flashing of Tasmota and building the output module go after I receive the product.

kizniche, 2020-08-11:
All 4 plugs were quick to flash Tasmota and I can now control them via MQTT publish commands from an MQTT server (mosquito) hosted on the same Raspberry Pi Mycodo is installed on. Each plug had the MQTT server set to localhost, as well as the host for the MQTT outputs. The Mycodo output topic was set to "cmnd/tasmota1/Power" with tasmota1 being your device name, and the ON Payload to "on" and OFF payload to "off".

It's a very nice system and I haven't had any issues in the past week or so they've been running. I'm closing this issue since we've figured out connectivity with Tasmota.

closed 2020-08-11 by kizniche

aTanguay, 2022-01-20:
Hi-

Is there any way you can elaborate on setting this up? Very new to MQTT and I am puzzled. I have Tasmoto installed on my plugs, and mosquito server installed running on the same Pi as Mycodo, but I can't seem to get these to talk to each other.

Here's my MQTT setup in the plug,

[image]

and here's my server output config..

[image]

From here, I'm just trying to use the On/Off buttons on the output manually. But I get nothing. 

Sorry for the newbie question, but this is definitely some new ground for me.

aTanguay, 2022-01-21:
I was close. Apparently, I had not installed my MQTT broker properly, so it wasn't able to accept outside connections. Reinstalled it and now it's good. For others trying this, I followed this to get MQTT going...

https://randomnerdtutorials.com/how-to-install-mosquitto-broker-on-raspberry-pi/

From there I got the info from my info screen on the switch...

[image]

Its configured this way...

[image]

Very exciting, works great, over the air.