kizniche/Mycodo #720: Error 500 upgrading from 8.2.0

SAM26K, 2019-12-09:
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 539, in admin_upgrade
    is_internet=is_internet)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/asyncsupport.py", line 76, in render
    return original_render(self, *args, **kwargs)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 2, in top-level template code
    {% set active_page = "upgrade" %}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1005, in render
    return concat(self.root_render_func(self.new_context(vars)))
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 18, in root
    objDiv.scrollTop = objDiv.scrollHeight;
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 335, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 119, in block_body
    {{_('No upgrade is available. You are running the latest release, version')}} <a href="https://github.com/kizniche/Mycodo/releases/tag/v{{current_release}}" target="_blank">{{ current_release }}</a>
TypeError: '>' not supported between instances of 'NoneType' and 'int'

kizniche, 2019-12-09:
8.2.0 was never a release, as it was a placeholder for those who upgraded to master. I'm not sure how that error came about, but try upgrading to master again.

kizniche, 2019-12-09:
Was fixed yesterday with 139d3b37dcb0886432052cdea82a9066033527a3

closed 2019-12-09 by kizniche