kizniche/Mycodo #721: Flask Loading Issue

not5, 2019-12-10:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

Gunicorn worker throwing an exception while loading flask

### Versions:

 - Mycodo Version: 8.1.1
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Desktop

### Reproducibility

Not really sure. I think it has to do with my enabling VNC through raspi-config.

Tried stopping the service and manually running mycodoflask with gunicorn using the following:
`sudo service mycodoflask stop
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app`

### Expected behavior

Compress not able to be imported due to null bytes:
`[2019-12-09 22:51:26 -0500] [1298] [INFO] Starting gunicorn 20.0.0
[2019-12-09 22:51:26 -0500] [1298] [INFO] Listening at: unix:/var/run/mycodoflas                                                   k.sock (1298)
[2019-12-09 22:51:26 -0500] [1298] [INFO] Using worker: gthread
[2019-12-09 22:51:26 -0500] [1304] [INFO] Booting worker with pid: 1304
[2019-12-09 22:51:27 -0500] [1304] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", li                                                   ne 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread                                                   .py", line 92, in init_process
    super().init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py                                                   ", line 133, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py                                                   ", line 142, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", l                                                   ine 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py"                                                   , line 49, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py"                                                   , line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line                                                    331, in import_app
    mod = importlib.import_module(module)
  File "/home/pi/Mycodo/env/lib/python3.7/importlib/__init__.py", line 127, in i                                                   mport_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
    from mycodo.mycodo_flask.app import create_app
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 16, in <module>
    from flask_compress import Compress
ValueError: source code string cannot contain null bytes
[2019-12-09 22:51:27 -0500] [1304] [INFO] Worker exiting (pid: 1304)
[2019-12-09 22:51:27 -0500] [1298] [INFO] Shutting down: Master
[2019-12-09 22:51:27 -0500] [1298] [INFO] Reason: Worker failed to boot.`

kizniche, 2019-12-10:
I've never come across this error. I'd suggest trying a new SD card and redownloading and reflashing Raspbian to rule out a bad image or OS write, or bad SD card.

not5, 2019-12-11:
Not sure what the issue was, but reflashed buster & upgraded, now everything works. Must have been a bad flash.

closed 2019-12-11 by not5