kizniche/Mycodo #722: Philips Hue Smart Light Integration

smguth, 2019-12-10:
**Is your feature request related to a problem? Please describe.**
Light automation support/Ideas with Philips Hue

**Describe the solution you'd like**
I'm pretty excited about this one...
I found this URL  https://github.com/bahamas10/hueadm
So I ran:
curl -sL https://deb.nodesource.com/setup_13.x | sudo -E bash -
sudo apt-get install -y nodejs
sudo npm install -g hueadm
hueadm search
hueadm "IPADDRESS_OF_HUE" register user
nano ~/.hueadm.json  
pasted my ipaddress and user string in the file **follow the url's guide**
hueadm lights
hueadm light 3
The commands all work on my raspberry pi that is running mycodo.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I would like to use mycodo for light schedules with the philips hue lights.  Also, alerts could send an email and set a light bulb a notification color.  Example: temp high alarm - light turns red
**Additional context**
Add any other context or screenshots about the feature request here.
https://github.com/bahamas10/hueadm

kizniche, 2019-12-14:
Conditionals are pretty powerful and can be programmed to do all you mentioned. Conditionals are programmed in Python 3. You can use ```subsystem``` to execute ```hueadm```, unless there's a python library that would make it easier/more secure.

Found some libraries:

 -  https://github.com/studioimaginaire/phue
 -  https://github.com/quentinsf/qhue
 -  https://github.com/issackelly/python-hue
 -  https://github.com/adamcharnock/python-hue-client

kizniche, 2019-12-14:
Looks like phue is the most feature-rich and up-to-date library that's actively maintained. I'd suggest testing this library for use in Conditionals, as it's in the native language used by Mycodo.

kizniche, 2020-05-14:
I've been working on a new feature for the 8.5.0 release that allows custom output modules to be imported, which will allow user-created, single-file output modules to be used. This will enable rapid-prototyping and deployment of new output modules for use in Mycodo. This issue is a perfect candidate for using the new output system and I'd like to help develop an output module for these lights. There will likely be some tweaking that needs to go on behind the scenes to allow color to be passed to an output module, but the heavy lifting has mostly been done.

ebo, 2020-12-17:
Out of curiosity, has anyone looked at the Hue's security (see: https://www.usatoday.com/story/tech/2020/02/05/how-to-avoid-smart-lights-getting-hacked/4660430002/ ).  Personally I would love to use them but am concerned about their security.

closed 2021-09-20 by kizniche